mavlink/mavros #301: Fix orientation matching util

vooon, 2015-05-24:
https://github.com/mavlink/mavros/pull/296#issuecomment-105034994

vooon, 2015-05-24:
Moving to UAS should fix that.

TSC21, 2015-05-24:
Is it ongoing or do I check it and merge?

vooon, 2015-05-24:
Wait a moment please.

vooon, 2015-05-24:
Now it should work. C++ linking sometime strange :)

```
vovan@robert:~/ros$ objdump devel/lib/libmavros_plugins.so -x | grep orientation
0000000000000000       F *UND*	0000000000000000              _ZN6mavros3UAS24get_attitude_orientationEv
vovan@robert:~/ros$ objdump devel/lib/libmavros_extras.so -x | grep orientation 
0000000000000000       F *UND*	0000000000000000              _ZN6mavros3UAS27sensor_orientation_matchingEh
0000000000000000       F *UND*	0000000000000000              _ZN6mavros3UAS24get_attitude_orientationEv
vovan@robert:~/ros$ objdump devel/lib/libmavros.so -x | grep orientation        
0000000000000000 l    df *ABS*	0000000000000000              sensor_orientation.cpp
00000000003cbbc0 l     O .bss	00000000000004e0              _ZL18sensor_orientation
000000000017a79b l     F .text	0000000000000015              _GLOBAL__sub_I_sensor_orientation.cpp
0000000000179c3c g     F .text	000000000000004f              _ZN6mavros3UAS27sensor_orientation_matchingEh
000000000011a4bc g     F .text	0000000000000093              _ZN6mavros3UAS24get_attitude_orientationEv
```

closed 2015-05-24 by vooon

TSC21, 2015-05-24:
I confirm it works! :D Can close now

commit 0ff3726d (2015-05-24): mavros fix #301: move sensor orientation util to UAS