mavlink/mavros #304: mavros px4.launch can not get CON: Got HEARTBEAT, connected.

congleetea, 2015-06-02:
I use odroid U3 connect to TELEM2, after I run roslaunch mavros px4.launch. It should get message:  
 CON: Got HEARTBEAT, connected.
however, I can not get it and can not switch to offboard switch . anyone can tell me the reason. thank in advance.  following is the message in the terminal after running roslaunch mavros px4.launch:
~$ roslaunch mavros px4.launch 
... logging to /home/congleetea/.ros/log/c87ec48a-092e-11e5-9413-68f728225dee/roslaunch-congleetea-Lenovo-5220.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://congleetea-Lenovo:52409/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat: 5.0
 * /mavros/conn/system_time: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync: 0.1
 * /mavros/fcu_url: /dev/ttyUSB0:57600
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: fcu
 * /mavros/global_position/frame_id: local_origin
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/send_tf: False
 * /mavros/gps/frame_id: gps
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/child_frame_id: fcu
 * /mavros/local_position/frame_id: local_origin
 * /mavros/local_position/send_tf: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
 * /mavros/mocap/use_tf: False
 * /mavros/plugin_blacklist: ['image_pub', 'gp...
 * /mavros/plugin_whitelist: []
 * /mavros/px4flow/frame_id: px4flow
 * /mavros/px4flow/ranger_fov: 0.0
 * /mavros/px4flow/ranger_max_range: 5.0
 * /mavros/px4flow/ranger_min_range: 0.3
 * /mavros/safety_area/p1/x: 1.0
 * /mavros/safety_area/p1/y: 1.0
 * /mavros/safety_area/p1/z: 1.0
 * /mavros/safety_area/p2/x: -1.0
 * /mavros/safety_area/p2/y: -1.0
 * /mavros/safety_area/p2/z: -1.0
 * /mavros/setpoint_accel/send_force: False
 * /mavros/setpoint_attitude/child_frame_id: attitude
 * /mavros/setpoint_attitude/frame_id: local_origin
 * /mavros/setpoint_attitude/listen_tf: False
 * /mavros/setpoint_attitude/listen_twist: False
 * /mavros/setpoint_attitude/pose_with_covariance: False
 * /mavros/setpoint_attitude/reverse_throttle: False
 * /mavros/setpoint_attitude/tf_rate_limit: 10.0
 * /mavros/setpoint_position/child_frame_id: setpoint
 * /mavros/setpoint_position/frame_id: local_origin
 * /mavros/setpoint_position/listen_tf: False
 * /mavros/setpoint_position/tf_rate_limit: 50.0
 * /mavros/startup_px4_usb_quirk: True
 * /mavros/sys/disable_diag: False
 * /mavros/sys/min_voltage: 10.0
 * /mavros/target_component_id: 50
 * /mavros/target_system_id: 1
 * /mavros/tdr_radio/low_rssi: 40
 * /mavros/time/time_ref_source: fcu
 * /mavros/time/timesync_avg_alpha: 0.6
 * /mavros/vision_pose/child_frame_id: vision
 * /mavros/vision_pose/frame_id: local_origin
 * /mavros/vision_pose/listen_tf: False
 * /mavros/vision_pose/pose_with_covariance: False
 * /mavros/vision_pose/tf_rate_limit: 10.0
 * /mavros/vision_speed/listen_twist: False
 * /mavros/visualization/child_frame_id: fcu
 * /mavros/visualization/fixed_frame_id: local_origin
 * /mavros/visualization/marker_scale: 2.0
 * /rosdistro: indigo
 * /rosversion: 1.11.9

NODES
  /
    mavros (mavros/mavros_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[mavros-1]: started with pid [5239]
[ INFO] [1433253326.787763464]: FCU URL: /dev/ttyUSB0:57600
[ INFO] [1433253326.788234434]: serial0: device: /dev/ttyUSB0 @ 57600 bps
[ INFO] [1433253326.789099904]: GCS bridge disabled
[ INFO] [1433253326.879212620]: Plugin 3dr_radio loaded and initialized
[ INFO] [1433253326.885226785]: Plugin actuator_control loaded and initialized
[ INFO] [1433253326.894795820]: Plugin command loaded and initialized
[ INFO] [1433253326.910303384]: Plugin ftp loaded and initialized
[ INFO] [1433253326.926012964]: Plugin global_position loaded and initialized
[ INFO] [1433253326.926239507]: Plugin gps blacklisted
[ INFO] [1433253326.926286133]: Plugin image_pub blacklisted
[ INFO] [1433253326.944903295]: Plugin imu_pub loaded and initialized
[ INFO] [1433253326.952314554]: Plugin local_position loaded and initialized
[ INFO] [1433253326.952765165]: Plugin mocap_pose_estimate blacklisted
[ INFO] [1433253326.957620186]: Plugin param loaded and initialized
[ INFO] [1433253328.385011478]: Plugin px4flow loaded and initialized
[ INFO] [1433253328.392542040]: Plugin rc_io loaded and initialized
[ INFO] [1433253328.397487539]: SA: Set safty area: P1(1.000000 1.000000 1.000000) P2(-1.000000 -1.000000 -1.000000)
[ INFO] [1433253328.402860872]: Plugin safety_area loaded and initialized
[ INFO] [1433253328.411927749]: Plugin setpoint_accel loaded and initialized
[ INFO] [1433253328.412025767]: Plugin setpoint_attitude blacklisted
[ INFO] [1433253328.427087690]: Plugin setpoint_position loaded and initialized
[ INFO] [1433253328.427155353]: Plugin setpoint_velocity blacklisted
[ INFO] [1433253328.438508307]: Plugin sys_status loaded and initialized
[ INFO] [1433253328.445707853]: Plugin sys_time loaded and initialized
[ INFO] [1433253328.448817017]: Plugin vfr_hud loaded and initialized
[ INFO] [1433253328.463745704]: Plugin vision_pose_estimate loaded and initialized
[ INFO] [1433253328.471174749]: Plugin vision_speed_estimate loaded and initialized
[ INFO] [1433253328.479640878]: Plugin visualization loaded and initialized
[ INFO] [1433253328.492799509]: Plugin waypoint loaded and initialized
[ INFO] [1433253328.493059549]: Autostarting mavlink via USB on PX4
[ INFO] [1433253328.493176221]: Built-in mavlink dialect: ardupilotmega
[ INFO] [1433253328.493256417]: MAVROS started. MY ID [1, 240], TARGET ID [1, 50]

Royalcheese, 2015-06-02:
Why are many plugin blacklisted ? your setting ?

congleetea, 2015-06-02:
@Royalcheese  yes, I set them . I use mocap indoor, I think they are not used.

Royalcheese, 2015-06-02:
What does it say when you try switch offboard ?

congleetea, 2015-06-02:
@Royalcheese  REJECT OFFBOARD.  but there is nothing in the terminal.

congleetea, 2015-06-02:
whether the TELEM2 is damaged?  How can i check it ?

vooon, 2015-06-02:
Not connected message - usualy means that you try to connect to wrong port.
E.g. android phone also may register ACM.

Also seems that your default configuration is changed from default one: https://github.com/mavlink/mavros/blob/master/mavros/launch/px4_pluginlists.yaml
Please make your own package with your launch and config. Do not edit mavros ones because its hard to support when user changes default scripts.

vooon, 2015-06-02:
Usually those packages called `xyz_bringup` or similar.

vooon, 2015-06-02:
Also i edited first message to format log as code block. Please look https://guides.github.com/features/mastering-markdown/ syntax highlighting.

congleetea, 2015-06-02:
@vooon the port is right , last week I used another px4 board to achieve offboard.

vooon, 2015-06-02:
What `/diagnostics` say? Do you used hardware UART on your OBC?

vooon, 2015-06-02:
Ok, i see FTDI-like USB module. They can change numbering, see `dmesg`.
Other thing did you configured `SYS_COMPANION`?

congleetea, 2015-06-02:
@vooon /diagnostics says 
message: not connected
hardware_id:/dev/ttyUSB0:57600

type dmesg in terminal : there has a message: usb 1-3.2:FTDI USB Serial device converter now attached to ttyUSB0.
i did not configure SYS_COMPANION, Do i need configure it ?

vooon, 2015-06-02:
Yes, it starts mavlink app on FCU side.

congleetea, 2015-06-02:
@vooon NOW I set SYS_COMPANION to 921600 using QGC, but also nothing works.

Royalcheese, 2015-06-03:
Is there hardware difference between your old board and the new one ?

vooon, 2015-06-03:
@congleetea and did you set baud rate to 921600 at mavros url?

vooon, 2015-06-03:
And without connection most interesting part of diagnostics is data rate.

congleetea, 2015-06-03:
thank you a lot,  my boards are same,  the hardware is ok, just two differents: I did not set  SYS_COMPANION . another is that I add a io-shield on odroid u3. yesterday i set SYS_COMPANION to 921600 but not reboot the board, it can not work. today ,  I start the board , it works. whether  I need reboot board  after change parameter  SYS_COMPANION ?

Royalcheese, 2015-06-03:
Sys_companion act like config.txt in the SD card of the fcu , so if the fcu do not restart, the starting config can't be set up I guess

vooon, 2015-06-03:
@Royalcheese are right, SYS_COMPANION handled in RC scripts, so reboot are necessary.

congleetea, 2015-06-04:
@vooon  yes , now i connected it. thanks you again.

closed 2015-06-04 by vooon

alvikag, 2018-03-27:
Hi,
 I am using mavros 0.23.3 and ros kinetic setup on an odroid, I am connecting a pixracer via USB. Upon launching the mavros node, I am not able to get the hearbeat connection. I even tried setting the SYS_COMPANION parameter to 921600 as recommended and also did a reboot, but it doesnt seem to be working.