mavlink/mavros #305: Official method to configure message rate

y22ma, 2015-06-07:
Hey guys,

Just wondering what is currently the offical method to configure message rate? I was hoping to configure IMU message rate to around 200Hz on the Telem port 2 on the Pixhawk.

I did see this [ticket](https://github.com/mavlink/mavros/issues/178), but I could not find the documentation on QGC for rate configuration as mentioned by @LorenzMeier.

DonLakeFlyer, 2015-06-07:
You can’t configure message rates from QGC.

y22ma, 2015-06-07:
@DonLakeFlyer thanks for the response. So currently the only way to do it is to modify the rc.usb and reflash? I'm assuming that the stream API is still deprecated.

Cheers,

Yan

DonLakeFlyer, 2015-06-07:
I don’ t know the answer as to how to do it from the firmware side.

vooon, 2015-06-07:
@y22ma you don't need reflash, use `extras.txt` on SD card.

y22ma, 2015-06-08:
Got it. I just need to add, for example: 

mavlink stream -d /dev/ttyACM0 -s HIGHRES_IMU -r 100 

to increase the rate for IMU messages to 100Hz?

y22ma, 2015-06-08:
So I added a extras.txt on /etc/ on the SD card, and here's what I have:
```
#!nsh
#
# USB MAVLink start
#

mavlink start -r 80000 -d /dev/ttyS6 -x
# Enable a number of interesting streams we want via USB
mavlink stream -d /dev/ttyS6 -s PARAM_VALUE -r 300
mavlink stream -d /dev/ttyS6 -s MISSION_ITEM -r 50
mavlink stream -d /dev/ttyS6 -s NAMED_VALUE_FLOAT -r 10
mavlink stream -d /dev/ttyS6 -s OPTICAL_FLOW_RAD -r 10
mavlink stream -d /dev/ttyS6 -s VFR_HUD -r 20
mavlink stream -d /dev/ttyS6 -s ATTITUDE -r 20
mavlink stream -d /dev/ttyS6 -s ACTUATOR_CONTROL_TARGET0 -r 30
mavlink stream -d /dev/ttyS6 -s RC_CHANNELS_RAW -r 5
mavlink stream -d /dev/ttyS6 -s SERVO_OUTPUT_RAW_0 -r 20
mavlink stream -d /dev/ttyS6 -s POSITION_TARGET_GLOBAL_INT -r 10
mavlink stream -d /dev/ttyS6 -s LOCAL_POSITION_NED -r 30
mavlink stream -d /dev/ttyS6 -s MANUAL_CONTROL -r 5
mavlink stream -d /dev/ttyS6 -s HIGHRES_IMU -r 100
mavlink stream -d /dev/ttyS6 -s GPS_RAW_INT -r 20

# Exit shell to make it available to MAVLink
exit
```

I changed the HIGHRES_IMU rate to 100, and changed the port to ttyS6 to publish data through the SERIAL4 port. I don't know what the default baud rate is so I assumed 57600, and I don't see data coming out of it. Do I have to enable data on SERIAL4 somehow, and how do I configure a fast baudrate?

Cheers,

Yan

closed 2015-06-22 by mhkabir

491734045, 2016-04-16:
@y22ma 
Does it mean we have to add the extras.txt on /etc/? The PX4 firmware will read the extras.txt on the /etc/ getting configuration parameters?

491734045, 2016-04-16:
@y22ma 
I want to change the message rate with PX4 firmware. Does it mean we have to add the extras.txt on /etc/? The PX4 firmware will read the extras.txt on the /etc/ getting configuration parameters?  Anything else we need to do except adding the extras.txt?
Thanks!