mavlink/mavros #307: MoCap testers needed (VICON, Optitrack, ...)

TSC21, 2015-06-12:
Testers to `mocap_pose_estimate.cpp` are welcomed! The purpose is also to test https://github.com/PX4/Firmware/pull/2317 implementation, which fuses mocap data into position and attitude estimators (the simpler ones for now).

TSC21, 2015-06-12:
@congleetea, @blackcoder, @tonybaltovski ping!

Royalcheese, 2015-06-22:
Just to know, now we can use mocap topic instead of vision topic for optitrack ? I will surely make some test when I find some time. If you could tell me what kind of data you want :)

TSC21, 2015-06-22:
@Royalcheese tests to this branch: https://github.com/PX4/Firmware/pull/2361. We are in a process of finding out what's the correct conversions to apply in the coordintates right now. Feel free to test the branch, do a video and send some logs with log muncher :)

TSC21, 2015-06-28:
@Royalcheese any news on testing PX4/Firmware#2361?

Royalcheese, 2015-06-29:
No I'm sorry , my drone is out of flight for now, I borrow propeller to team to build another one waiting for new propeller to arrive.
I'll test test as soon as possible :/ ..

LorenzMeier, 2015-06-29:
@TSC21: I think it's time to fundamentally change the development workflow: all changes should be validated in simulation, and the maintainers should have simulations or require the parallel contribution of those for Gazebo. The current process is intransparent, error-prone and slow. This is how the autopilot software development sustains itself - it would be hopeless without unit tests and simulation.

blackcoder, 2015-06-29:
@LorenzMeier  totally agree with you. I've been experimenting recently with the gazebo simulation support inside px4.  You guys have done absolutely fantastic work on getting the simulation basics up and running. 

Speaking of which, not all the launch files in the https://github.com/PX4/rotors_simulator work, some of them require slight modifications and I was wondering if I can just use https://github.com/ethz-asl/rotors_simulator instead (seems or up to date with more generic structure)? 

On the long run, generating unit tests, and system integration tests with known use cases is the way to go. I can assist with this once I get my head around all the functionalities of the simulation.

wangsen1312, 2015-07-02:
VICON works very well when using  remap form mocap/tf to vicon topic.

TSC21, 2015-07-02:
@wangsen1312 thanks for the feedback. Have you flight tested latest master? It now integrates mocap data into the estimators, so you should be able even do POSCTL.

wangsen1312, 2015-07-06:
@TSC21  Thank you for your information, I have just tested the latest master firmware, it works well. What you mean POSCTL, the same as OFFBOARD with SETPOSITION? Because we can get the local position accurately.

arushk1, 2015-07-06:
Okay, so I have an optitrack system. PX4 Flightcontrol, PX4FLOW. Anything I can contribute?

TSC21, 2015-07-07:
@wangsen1312: https://pixhawk.org/firmware/apps/mc_pos_control.
@arushk1 You can try use latest master of PX4 Firmware together with [mocap_pose_estimate](https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/mocap_pose_estimate.cpp) plugin of mavros. Test if you can get good pose estimation on QGS plots of `local_position` (or checking `/mavros/position/local` on rqt_plot) through bench tests. If yes, then you can proceed to flight tests, first in manual mode, then ou can try POSCTL. You can also test OFFBOARD mode if you have some sort of position controller running on a OBC or on you computer.

TSC21, 2015-07-07:
Anyhow, @blackcoder already tested mocap plugin with the mocap support on the Firmware with success, so you should also get good results with it.

TSC21, 2015-07-21:
@arushk1 any updates?

mzahana, 2015-10-02:
Hello guys. I am trying to control the position of pixhawk by sending setpoint from my PC to pixhawk via 3DR telemetry. I am working with optitrack mocap, and I'd like to test the latest development in the PX4 firmware. I already have the position and the orientation from mocap streamed to my PC. I know that there set_vision mavlink message, which sends positions and pose to pixhawk. Is this the right mavlink message to use, or is there something else better to use?
Another question, should I send both the position and orientation, if i only want to control the position, or sending mocap position is enough?

Axes convention in optitrack is North(Z+), West(X+), Up(Y+)
Pitch(around X), Yaw (around Y), Roll(around Z)

How to convert this to acceptable convention of the pixhawk?

What I have in mind after getting this done is to:
- Arm the quad by sending the appropriate mavlink message (done and tested)
- activate the OFFBOARD mode (tested)
- send set points using the setposition local ned mavlink message (is this the right message to use?)

I apreciate if you can point me to the right direction to get this work.

NOTE: I am controlling my commands from MATLAB, and I have a way to send mavlink messages from there. So, I am working directly with mavlink messages not mavros.

Thanks

BrownKyle, 2015-10-06:
Hey, similar to @mzahana I would like to test this but will need clarification on the coordinate system. mobcap_pose_estimate.cpp requires NED, even though ROS is usually ENU.

Does the NED transform have to include North pointing to magnetic north for magnetometer sensors.

I can get position on ROS from either vrpn_client_ros or UDP from Simulink. We have an optitrack system with the older tracking tools not motive.

We have a vehicle with ROS indigo and Pixhawk with either Raspberry Pi 2 running ubuntu or Odroid XU4 also running ubuntu.

mzahana, 2015-10-08:
Hi all,

I finished setting up my mocap system, Optitrack, and did the frame conversion and now getting NED position and orientation. This  is all done in MATLAB. Now I am sending the quaternion and xyz using mocap_att_pos mavlink message, and I'd like to check that Pixhawk, running the latest px4 frimware, is receiving the correct data, and in adequate rate. How is this possible? Is there some logs that I can plot in the analysis tab of Qgroundcontrol?

My second concern, after successfully transmitting the mocap data to pixhawk, how can I use the position control mode to hold the quad in a particular position, what is the procedure?

Thanks

kaustubha-jagtap, 2015-10-15:
Hi all, 
I am trying to use optitrack system to fly PX4 flight controller indoors. I am having trouble checking if the FCU is receiving the mocap data. I published the mocap data on /mavros/mocap/pose( z axis up) but I can not find a way to check if FCU is receiving it.
Thanks

mzahana, 2015-10-16:
1- First make sure that you are using the latest PX4 firmware
2- You can use QgroundControl-->Analyize tab, to monitor localNED.x, localNED.y, localNED.z

mzahana, 2015-10-18:
Hi All,

I would like to give some updates regarding my indoor experiments.

1- I am finally able to successfully transmit OptiTrack mocap data (position/attitude) via MATLAB, and I can check in QgroundControl that it's being received correctly by plotting logs: mocap.x, mocap.y, mocap.z ...etc

2- I am able to test **altitude control** mode successfully [link to Video](https://www.youtube.com/watch?v=CZyfx7if7VU)
3- I am currently trying to test **position control** mode, but the control parameters need to be tuned. I am using Micro PCB 250 quad [link here](http://hobbyking.com/hobbyking/store/__45467__Turnigy_Integrated_PCB_Micro_Quad_V2_Black_KIT_.html). So, if you already have some suggestions on the PID parameters of the position controller, kindly let me know.

Everything (i.e. monitoring/commanding) is done in Matlab using a tool that I developed called **MatMav**.

I will post my updates, possibly with some videos, once I get this mode working fine.

My next step is to test the **offboard** mode afterwards.

Cheers,
MZahana

jespestana, 2015-10-20:
Hi all,

We, isowerale and me, have been working with the PX4FMU using a custom drone for a while. We are using the PX4FMU with the PX4IOAR and an AR Drone frame. We have a small indoors flying space equipped with an OptiTrack motion tracking system (mocap system), and we have been able to fly autonomously (offboard mode) with the pixhawk. Until now, we were working with an "old" version of the firmware and mavros. In September we made an update to the latest version of the Firmware, mavros and Qgroundcontrol; and after some interface updates we could fly autonomously again. I would like to explain our current configuration, so that hopefully we can start a fruitful discussion and collaboration on this topic.

Current software versions, as of 20 October 2015: 
* Firmware stable branch, utilized commit from Aug 30, 2015 or Sep 24, 2015 (they are equivalent): https://github.com/PX4/Firmware/commit/8580ac0155215ae0d0ad7b7303797fe1c8cce4a7
* mavros modified from original commit: tag '0.15.0' from Sep 17 21:44:10 2015 +0300: https://github.com/mavlink/mavros/commit/e1656ed8bc4d4580f52ec1b98d7063ad5d3e00f3
* Qgroundcontrol:
  * qgroundcontrol v2.7:   setting parameter values online, PX4FMU sensor calibration, PX4FMU board 1.7.2 firmware update.
  * qgroundcontrol v2.5.2: setting parameter values online, PX4FMU sensor calibration, PX4FMU board 1.7 firmware update.

Control architecture main characteristics:
* There is a computer onboard the drone running ubuntu that gets the mocap data through a wifi link. To summarize, our setup is the following: windows groundstation >LAN> linux groundstation >[ROS]WLAN> onboard computer >[MAVROS]serial> PX4FMU
* The mocap system pose measurements are used as input to the INAV EKF inside the PX4FMU, mavros topic /mavros/vision_pose/pose
* The INAV EKF ouput estimate can be checked at the mavros topic /mavros/local_position/local
* The position setpoint is set using the mavros topic /mavros/setpoint_position/local
  
Main parameter tuning, configuration and development that was required for our setup: 
* Set the INAV EKF to ignore the barometer measurements (set their weight to 0).
* Align the room based world frame with the world frame that is used by the pixhawk internally. Basically you need to know where the north is pointing in your room, and also perform the necessary reference changes in: the pose measurements obtained from the movap system for the INAV EKF, the INAV EKF ouput estimate and the position setpoints for the FMU commands.  
* Work around some problems with the magnetometer in our indoors laboratory (which affects the heading estimate in the Attitude EKF inside the FMU).

What can we do using offboard mode and position setpoints?
* Just for completeness: the position setpoints consist of a point in 3D space and a heading
* The drone can:
  * take-off and land,
  * hover at any position inside our tracking volume, and
  * fly along paths specified with position setpoints. The speed of these flights is not fast, as expected from close range position hovering commands.

What we would like to discuss?
1. @TSC21 What kind of tests need to be done to test the PX4 with the mocap system? Should we use the /mavros/mocap/pose topic instead of the vision_pose?
2. As mentioned above, the /mavros/vision_pose/pose topic is used to provide measurements to the INAV EKF. Can we expect more accuracy by using the /mavros/mocap/pose topic? As far as we could tell the firmware only uses the position estimates from these topics, the attitude estimates are not used.
3. Is there a firmware branch that additionally uses the attitude provided by the mocap system in the data fusion performed by the INAV EKF?

mzahana, 2015-10-20:
Hi,

Thanks for sharing your experience. I have reached a stage where I can test altitude hold and position control give position and attitude feedback from Optitrack.  All the interface is done in Matlab (receive Micah data from Motive, and sending it to Pixhawk using mailing interface). I am using the Micah mailing message not the vision one, and it'seems working pretty well. My next step is to test off board mode by sending position and heading set points. So, I was wondering if there are some parameters that should be set/modified before going activating this mode. Next, I am going to try controlling multiple Pixhawk at the same time.

Thanks.

mhkabir, 2015-10-21:
@jespestana 
> 1. Should we use the /mavros/mocap/pose topic instead of the vision_pose?

Yes. The Inav estimator supports the MOCAP message as well.

> As far as we could tell the firmware only uses the position estimates from these topics, the attitude estimates are not used.

See below.

> Is there a firmware branch that additionally uses the attitude provided by the mocap system in the data fusion performed by the INAV EKF?

Yes - https://github.com/PX4/Firmware/pull/2667

jespestana, 2015-10-21:
@mzahana ,
My advice prior to offboard mode flight tests is to make sure that the INAV estimated pose is of good quality. Usually you should be able to receive the pose estimates from the INAV EKF. In our case, this is done directly by the mavros node, but you can check the code in mavros to see which mavlink message is received and published in the /mavros/local_position/local ROS topic. This way you could take a look at the data in real-time in Matlab. Currently, we check the INAV EKF estimated pose in ros rviz as a tf frame. It really helps, because you can compare the estimated pose with the groundtruth (from the optitrack). 

I think that it will pay off to play around with the main sensor weights in the INAV configuration. With mavros you can do this in real-time by configuring the UDP connection with the IP of your groundstation, and setting the parameters from qgroundcontrol. Once you are happy with the INAV estimated pose, then you can start testing the offboard mode and the position setpoints.

@mhkabir ,
Thanks for the information, we will test the firmware that you are referencing ( PX4/Firmware#2667 ) soon. Currently we are a bit busy configuring a new drone, but we will try it in the next few weeks.

kaustubha-jagtap, 2015-10-21:
Hi @jespestana : I am relatively new to this and I was wondering how to use INAV EKF. I did not find any parameter to enable this. or am I suppose to install another package for this? It would of great help if you can point me towards any resources for reference. 
Thanks,

jespestana, 2015-10-21:
@kaustubha-jagtap
The PX4 stack has been changing recently (at least, since I started using it). So, it is not easy to extrapolate things from old posts.

I already give various hints as to how to use the INAV EKF, read my two comments on these threads for more information than the one I provide here. In our PX4FMU autopilot (or our drone for that matter), the INAV EKF/estimator is started once it receives the first pose measurement. Using a recent qgroundcontrol (for instance 2.5 or 2.7) it is easy to find its parameters, just go to parameters > INAV. The parameters have short explanations, so you can figure out approximately what they do. You can also understand from these parameters what sensor data is fused by this estimator.

To test the INAV EKF:
[0. Start your roscore, but I am assuming that you have experience with ROS]
1. Connect to your drone using mavros
2. Publish the pose estimate of where your drone is in the /mavros/vision_pose/pose ROS topic
3. Subscribe to the /mavros/local_position/local ROS topic. The messages in this topic are the pose estimates from the INAV EKF

Note: it also helps to have the code of the firmware and the mavros version that you are using. This way you can always read the code to figure out specific things during your debugging sessions.

BrownKyle, 2015-11-27:
Hey, I have the flight controller receiving the motion capture data that has been converted to NED. I have tested and verified with both the macros topic local position and local position in ground control.

I have a set point position script running on my companion computer and when it runs the flight controller accepts off board mode.

The issue I am having is that when I use the master build the multi rotor oscillates badly under manual control. This isn't an issue with the stable build.

Will the oscillations be an issue if I run the set point control script? If yes, are there ways to stop this will still using the master build. If not will the motion capture integration work on the older stable builds of PX4?

mhkabir, 2015-11-27:
Please report the oscillations issue on the PX4 Firmware bugtracker, along with requisite logs demonstrating the issue.

BrownKyle, 2016-02-09:
Successfully tested publishing data to /mavros/mocap/pose. We have a Optitrack system using the older software tracking tools. I used VRPN to get the position data and transformed it with a python node on the on-board raspberry pi 2.

I tested it using way-point commands and velocity vector commands. The velocity vector commands I used are just the error between position and desired position.

I have uploaded the video to you tube: https://www.youtube.com/watch?v=k-wj-quYrPk

aitjellal, 2016-04-13:
@jespestana  thank you for the detailed procedure

I have the problems on how to align the pixhawk with the tracking system (Optitrack).
Could you please further details 
_"Align the room based world frame with the world frame that is used by the pixhawk internally. Basically you need to know where the north is pointing in your room, and also perform the necessary reference changes in: the pose measurements obtained from the movap system for the INAV EKF, the INAV EKF ouput estimate and the position setpoints for the FMU commands."_

I am using mavros , and  I have remapped the "optitrack" topic to "vision"
I set ATT_EXT_HDG_M = 1 (I have remapped "mocap" topic to "vision" for mavros); and ATT_W_EXT_HDG=0.95
and "tried" to align the pixhawk with the Tracking system.
For yaw= 0 on the Tracking system (Optitrack)
I expect to get in the FCU  mavros/local_position/pose"
Orientation quaternion=
x: 0.0
y: 0.0
z: 0.0
w: 1.0

But I get
orientation:
x: 0.00305893994527
y: 0.00357438427461
z: -0.999928040518
w: 0.0110354179502
(x =0 , y=0, z= -1, w=0.01)

is this normal ?
How should I proceed to set the Heading correctly? And how to align Pixhawk with Optitrack correclty?
Thanks

jespestana, 2016-04-14:
Hi aitjellal,

If I interpret your quaternion correctly, you mostly have a yaw=-180deg.
Meaning that the "x-direction" of your mocap "world" reference frame is
pointing in the opposite direction.

The x and y components of the quaternion are small, about 0.2deg each.
Which is normal.

Just try to visualize the reference frames in rviz (I think that there is
an option to publish them as "tf" messages, otherwise you just need to make
a node to republish them as "tf" messages). Make sure that you understand
your reference frames before proceeding with any flights!

Jesús

aitjellal, 2016-04-14:
I think there is a bug somewhere in (Mavros ,Mavlink, Px4) I am using the newest Mavros (gitlab) and newest stable px4 firmware.
when I align the my Optitrack coordinate frame with Pixhawk 
and use external heading (no compass heading, ATT_W_EXT_HDG=1.0)
The positionn x,y,z  is transmitted corretly
but the rotation (Quaternion x,y,z,w) is not sent correctly.
so when I compare 
mavros/local_position/pose
and 
vrpn_client/pose_world

the positions argees but the rotation do not agree ( , ATT_W_EXT_HDG=1.0)

when I use only the compass  heading , it works.
best regards

jespestana, 2016-04-15:
@aitjellal
It is difficult to assess you problem without more precise information
about the tests that you have conducted. Please, be more descriptive about
those experiments.

My preferred solution is to use the compass and the mocap system. When you
use the compass the PX4 will use the north direction to orient the ground
frame with it. Because you will be flying indoors, the north direction
estimate (based on the magnetometer measurements) will be variable. It is
important to determine the PX4 north direction estimate in order to be able
to send position commands according to it. This is no problem, because you
can get an estimate by comparing the yaw estimate coming from the PX4 and
the mocap system.

I don't keep the firmware of my drones up-to-date, so there might be a
better solution with the recent firmware. I don't think that I ever tuned
the ATT_W_EXT_HDG configuration parameter.

aitjellal, 2016-04-15:
@jespestana  Thank you for your help.

I think I should read the code carefully to understand it.
because I thought setting ATT_W_EXT_HDG=1.0   will  make the heading only from the external mocap.  but it does not. it is still depending on the compass.

I would use the default value for ATT_W_EXT_HDG.
now it is working so I will keep it like that.

jespestana, 2016-04-15:
@aitjellal
Back in the days, we found that there is a section of the PX4 autopilot
code which is supposed to create fictitious magnetometer readings based on
the mocap/vision data. We never managed to make it work properly though. As
far as we could tell (based on multiple experiments), these sections of the
code were not reachable.

Jesús

princeward, 2017-02-11:
@BrownKyle Do you have to specify angular velocity when you do velocity control?

Also, which MAVROS topic did you use? 
``
setpoint_velocity/cmd_vel
`` or
``
setpoint_attitude/cmd_vel
``?

closed 2017-02-11 by mhkabir

siddhu45, 2017-04-12:
Hi @TSC21 @mhkabir @vooon 
What test results did you guys obtain? I am working with OptiTrack system to do position control on the UAV with Pixhawk and Raspberry Pi onboard. I have achieved the following:

1. I can get position and attitude feedback from mocap system and publish it to mocap/pose
2. I can perform commanded trajectories in OFFBOARD mode via MAVROS

Apart from this, I have achieved quite a lot of stuff. But what I need is:

1. The coordinate frame Pixhawk uses (local and global) in the indoor GPS-void flight in the mocap room.
2. Also, as said, I can perform the trajectories,  but my main purpose is to perform the System Identification on my UAV. For that I need input motor RPMs and output measurements, the position, linear velocity, euler angles and angular rates. I collect all the measurements from the local/pose topic. I know the Pixhawk works on NED frame and ROS works in ENU frame, so I kept all my post calculations also in ENU frame. But still I am not able to get identification results. After checking everything, I know that the coordinate conversion (transformation) I'm doing is wrong for sure.
3. When I start to publish the mocap data to Pixhawk, the local/pose, imu/data and odom topics start showing the data in ENU frame from the previous local NED frame. Why? shouldn't ROS always use ENU. I have verified it many times.
4. My identification results will only be feasible if my coordinate transformation is correct. Even when I am able to use the mocap data, still the transformation is wrong somewhere I'm sure. Can anyone help me to proceed in correct direction or point me to the relevant discussion so I can find answers.
5. What is the local and global frames of Pixhawk in indoor environment or GPS-void space. I verfied #473 as @mhkabir suggested and is true.
6. I need the data in a known frame so that I can perform relevant transformation and get results. As default, everyone uses in mathematical modelling of a UAV, the general conventions are: position and Euler angles in Global, linear velocity and rotational rates in Body frames. So, can anyone give me (if there is any other way) to collect this data? Or maybe to get all the values directly from mocap system if anyone else is doing that?
7. I recorded velocity from the local/pose and also calculated velocity by integrating the linear accelerations from imu/data, but they don't match. The velocity in z just keeps on increasing. So, I know the data I am getting is definitely in wrong frame. 

I hope I was clear, I can help anyone who is stuck till this point in all the way but further, I need assistance of you good guys. I have a lot of small doubts and if I can solve them, maybe I can get identification results.

Cheers,
Sid