mavlink/mavros #308: Add support for `VIBRATION` msg

TSC21, 2015-06-12:
It's usage on firmware will be applied on the next weeks, according to: https://github.com/mavlink/mavlink/pull/398. In the meantime, I think I can add a decoder to it.

TSC21, 2015-06-12:
@vooon can you update mavlink release so this can compile?

vooon, 2015-06-12:
What FCU will support that? I can release, but there only 6 days from last one...

TSC21, 2015-06-12:
Both!  Please send it so I can test compilation

vooon, 2015-06-12:
Releasing....

TSC21, 2015-06-12:
>Releasing....

How can one update it now?

vooon, 2015-06-12:
- https://github.com/ros/rosdistro/pull/8664
- https://github.com/ros/rosdistro/pull/8665
- https://github.com/ros/rosdistro/pull/8667

TSC21, 2015-06-12:
Ok I'll be waiting for those to get merged then :)

TSC21, 2015-06-13:
@vooon changes were merged but how can I update mavlink version now?

TSC21, 2015-06-15:
Closed via https://github.com/mavlink/mavros/pull/309

closed 2015-06-15 by TSC21