mavlink/mavros #310: Plugin sending raw MAVlink message

tuloski, 2015-06-15:
Hi all,
I was trying to build a plugin to send this MAVlink message "MISSION_ACK (#47)" but I don't understand how to encode the MAVlink.
I was taking a look at some plugins as rc_io.cpp, and for decoding there is the function decode as in "mavlink_msg_rc_channels_raw_decode(msg, &port);" while to encode I don't see anything.
The only thing I noticed is this:

mavlink_msg_rc_channels_override_pack_chan(UAS_PACK_CHAN(uas), &msg,
				UAS_PACK_TGT(uas),
				channels[0],
				channels[1],
				channels[2],
				channels[3],
				channels[4],
				channels[5],
				channels[6],
				channels[7]
				);
		UAS_FCU(uas)->send_message(&msg);

that seems to encode and send the message but I don't understand where that function comes from and the way to use it, since in another plugin it is different.
Any help?

Thanks

vooon, 2015-06-15:
Why you need `MISSION_ACK`? That definitely will interferer with waypoint plugin, because it implements mission protocol.

There exists `mavlink_msg_*_encode()`, but `mavlink_msg_*_pack_chan()` is more usable
because you don't have to fill message structure (this has some downsides at field ordering).

All `mavlink_*` comes from mavlink library, for ROS i do monthly release (usually day == month). You may find sources in `/opt/ros/jade/include/mavlink/v1.0` (or indigo if you used previous ros release).

vooon, 2015-06-15:
Also you may find common message set documentation here: https://pixhawk.ethz.ch/mavlink/

tuloski, 2015-06-16:
Thanks for the answer, but how is the usage of mavlink_msg_*_pack_chan()? Because I can't find it anywhere (I'll take a look at mavlink sources). Do I need to define first a custom .msg? In which order do I need to put UAS_PACK_CHAN(uas), &msg, UAS_PACK_TGT(uas) and the payload (fields) of the message?
Thanks

vooon, 2015-06-16:
See [WaypointPlugin::mission_ack()](https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/waypoint.cpp#L691-L700).

All mavlink encoders/decoders placed in header only library in include folder of ros installation space.

tuloski, 2015-06-17:
It worked thanks.

vooon, 2015-06-17:
Ok. Now i think may close issue.

closed 2015-06-17 by vooon