mavlink/mavros #313: Change in remote UDP port

tuloski, 2015-06-19:
Hi all,
I'm trying to connect to the "FCU" via UDP, setting in the .launch file this string:
<arg name"fcu_url" default=udp://192.168.0.106:5000@192.168.0.102:6654" />

And when I launch the .launch everything goes fine but mavros changes automatically the remote port and it writes:

-udp0: Bind address: 192.160.0.106:5000   (this is fine)
-udp0: Remote address: 192.160.0.102:6654   (this is fine)

and after a few seconds (1-2 seconds):

-udp0: Remote address: 192.160.0.102:60236   (this is not fine it changed the port).

Why does it changes the port I wrote in the launch? Is it because it cannot connect to the port I specified?

PS: the "FCU" is sending MAVlink messages to mavros to the remote port 5000 (as requested) but maybe using a random bind port (it could be that 60236).

vooon, 2015-06-19:
It is because your program sends data from 192.160.0.102:60236 to bind port (192.160.0.106:5000).

In that case you may leave remote url part default. Also i don't recommend bind to interface address, better use IPv4 ANY = 0.0.0.0 (default).
So URL: `udp://:5000@`, and then, when FCU will send heartbeats mavros automatically acquires address.

tuloski, 2015-06-19:
Yeah, that is definitely better, but why the fcu_url gets overrided even if explicitely written?
Is this working as intended?

vooon, 2015-06-20:
It is intentional. Url just gives remote address at startup, but it **can change** in any time, so libmavconn udp simply switch to that new address. 
That feature requires you establish only point-to-point connections, many-to-point will break logic.
Here in issue list exists idea for multicast udp, but currently i don't need it.

closed 2015-06-28 by vooon