mavlink/mavros #314: "Time out" or " Mode changed" when use offboard mode

wangsen1312, 2015-06-21:
I send a stream of setpoints and tried offboard mode using command ''rosrun mavros mavsys mode -c OFFBOARD".
only shows "Time out" or "Mode changed". The UAV did not move at all.
Can someone give me suggestion, Thank you very much!

vooon, 2015-06-22:
`/diagnostics` please!

wangsen1312, 2015-06-22:
@vooon  @TSC21 Thank for your help.
My main purpose is control the UAV fly to the position which offered by mocap.
I use new stable PX4 firmware installed yesterday, 3DRradio connect to UAV directly without any onboard computer.

1) The first stage seems OK now. I use Vicon_bridge get the mocap data and publish the data to vision_pose/pose, and I change the INAV's parameters, only left the CBRK_no_vision = 0, vision weight max value 10, others are zero.
Now I can get the /mavros/local_position/local data as Vicon system offers.
 
2) Now I want to set local postition for my UAV.  I do following steps:
a. arm the UAV first (it works well) rosrun mavros mavsafety arm
b. sending a stream of setpoints (I use the the demo set point https://raw.githubusercontent.com/kuri-kustar/indoor_uav_control/master/scripts/setpoint_demo_2.py), which seems OK.

c. request offboard    rosrun mavros mavsys mode -c OFFBOARD 
This is the problem I meet, the quad is not taking off as I wish and not stablized and sometimes show "Time out". I only use demo to set positions, should I change some other parameters to stablize the UAV?

The /diagnostics file is below:

header: 
  seq: 78
  stamp: 
    secs: 1434983924
    nsecs: 567774546
  frame_id: ''
status: 
  - 
    level: 0
    name: mavros: FCU connection
    message: connected
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: Received packets:
        value: 1781
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 25
      - 
        key: Tx sequence number:
        value: 91
      - 
        key: Rx total bytes:
        value: 65234
      - 
        key: Tx total bytes:
        value: 1814
      - 
        key: Rx speed:
        value: 607.000000
      - 
        key: Tx speed:
        value: 20.000000
  - 
    level: 2
    name: mavros: GPS
    message: No satellites
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: Satellites visible
        value: 0
      - 
        key: Fix type
        value: 0
      - 
        key: EPH (m)
        value: Unknown
      - 
        key: EPV (m)
        value: Unknown
  - 
    level: 0
    name: mavros: Heartbeat
    message: Normal
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: Heartbeats since startup
        value: 72
      - 
        key: Frequency (Hz)
        value: 1.099231
      - 
        key: Vehicle type
        value: Quadrotor
      - 
        key: Autopilot type
        value: PX4
      - 
        key: Mode
        value: ALTCTL
      - 
        key: System status
        value: Standby
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: Sensor present
        value: 0x00000000
      - 
        key: Sensor enabled
        value: 0x00000000
      - 
        key: Sensor helth
        value: 0x00000000
      - 
        key: CPU Load (%)
        value: 32.8
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 0
    name: mavros: Battery
    message: Normal
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: Voltage
        value: 11.76
      - 
        key: Current
        value: 0.3
      - 
        key: Remaining
        value: 65.0
  - 
    level: 0
    name: mavros: 3DR Radio
    message: Normal
    hardware_id: /dev/ttyUSB0:57600
    values: 
      - 
        key: RSSI
        value: 165
      - 
        key: RSSI (dBm)
        value: -40.2
      - 
        key: Remote RSSI
        value: 168
      - 
        key: Remote RSSI (dBm)
        value: -38.6
      - 
        key: Tx buffer (%)
        value: 100
      - 
        key: Noice level
        value: 22
      - 
        key: Remote noice level
        value: 25
      - 
        key: Rx errors
        value: 0
      - 
        key: Fixed
        value: 1

By the way, after radio connect to UAV, some error information occurs, such as
[ERROR] [1434915513.490270708]: FCU: REJ. WP CMD: partner id mismatch
I don't know it will cause something wrong or not? 

Can you give me some suggestion? Thank you very much.

mhkabir, 2015-06-22:
Check if your local position estimates (LOCAL_POSITION_NED message) match your mocap data.
I have a strong suspicion that setting weights to zero in INAV is causing a bad estimate.

Royalcheese, 2015-06-23:
Or use ODROID onboard + WIFI + ROS network :)

wangsen1312, 2015-06-24:
@mhkabir  @Royalcheese  @vooon Thank you for your suggestion.
I have check the local position, which is matched with my mocap data.
I have reduce the frequency to 20HZ and the timeout cannot be seen, but the UAV cannot take off either. The propeller changed their speed but cannot take off!
Have you ever tried use mavros to take off? or I should do something else before take off?
How can you try the offboard Mode?
Thank you very much!

Royalcheese, 2015-06-25:
I think, it is because PID parameters have to be set following the weight of your drone , as i told you by mail. If you have a look on equations which are the basis of mc_pos_control, you'll see that there is the weight taken in count. but in the controller, you only control the Thrust, without taking care of the inertial characteristics of your uav. 
I guess you have to set proportional parameters the proper way in order to compensate weight, and this is that why I've tried to make a scilab simulation of the controller , to find the good parameters including the inertial characteristics.
But you can also do it step by step, increasing proportional part of controller incrementally until it takes off. but it can be "dangerous" cause of unstability induced by proportional controller ( I think.. ).

vooon, 2015-06-28:
Updates?

wangsen1312, 2015-06-29:
@vooon  I don't know why TImeout see again. The UAV cannot takeoff with mavros? Have you ever try to use the mavros to make the UAV takeoff first without GPS?

vooon, 2015-06-29:
FCU armed?

wangsen1312, 2015-06-29:
@vooon  arm works well. Now it cannot follow the set local position command.

vooon, 2015-06-29:
Hmm, strange. Please don't use master now. Seems that transform are broken...

Royalcheese, 2015-06-30:
@vooon  You mean the tf of ROS ?

TSC21, 2015-06-30:
You have to limit the tf publishing rate or it will result in timeouts

wangsen1312, 2015-07-01:
@vooon @mhkabir @TSC21 
I find it's hard to takeoff from floor so have tested another method: 
1) use MANUAL mode to take off and stabilize. 
2) Send local position stream 
3) change to the OFFBOARD mode using RC
I find something dangerous happen, UAV fly to some position far away from setting position.
Do you think the set_local_position function works well when I use VICON replace INAV vision parameters?

Royalcheese, 2015-07-02:
Is the problem issued is the code of mavros/plugins/setpoint position.cpp (issue 273) is solved yet ?

			 * Current PX4 has bug: it cuts throttle if there no velocity sp
			 * Issue #273.

Maybe is it your problem, but it implies that you have to send a trajectory and not just a setpoint.

vooon, 2015-07-02:
Bug #273 still opened, but there workaround in mavros.

LorenzMeier, 2015-07-02:
It was fixed on the Firmware side and I consequently closed it.

LorenzMeier, 2015-07-02:
@Royalcheese That comment should be removed consequently.

TSC21, 2015-07-02:
@LorenzMeier @Royalcheese we will remove comment soon then.

closed 2015-07-09 by vooon