mavlink/mavros #316: Installation in Indigo

Tomofoye, 2015-06-23:
I installed using sudo apt-get install ros-indigo-mavros ros-indigo-mavros-extras. When I try to build it using catkin_build, I get an error at 83%:

[ 83%] Building CXX object mavros/mavros/CMakeFiles/mavros.dir/src/lib/mavlink_diag.cpp.o
make[2]: *** No rule to make target `/usr/lib/libboost_signals-mt.so', needed by `/home/dukerama/catkin_ws/devel/lib/libmavros.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[ 85%] Building CXX object mavros/mavros/CMakeFiles/mavros.dir/src/lib/rosconsole_bridge.cpp.o
make[1]: *** [mavros/mavros/CMakeFiles/mavros.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Does this have something to do with the boost libraries?

TSC21, 2015-06-23:
Yes it is. What versiom of boost are you using and in what Ubuntu version? (If using Ubuntu of course)

Tomofoye, 2015-06-23:
I am using Boost version 1.54.0 on ubuntu 14.04,

vooon, 2015-06-23:
Why you want source installation?

Installed mavros debs should pull almost all dependencies, but not sure aboud `*-dev` packages.
You should do rosdep step.

Tomofoye, 2015-06-23:
I used source because it was the only method working so far. When I tried using the wstool method and got to this step: rosdep install --from-paths src --ignore-src --rosdistro indigo -y

I got an error 

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
mavros_extras: No definition of [python-argparse] for OS version [trusty]
libmavconn: Cannot locate rosdep definition for [libconsole-bridge-dev]
mavros: No definition of [boost] for OS version [trusty]

vooon, 2015-06-23:
Did you inited rosdep? See ros installation steps: a. init as root; b. update as user

Tomofoye, 2015-06-24:
After updating rosdep, I was able to perform the step: rosdep install --from-paths src --ignore-src --rosdistro indigo -y

However when I build the package with catkin_make, I still get the same initial error, here was the output from catkin_make: 

[  0%] [  0%] [  0%] Built target diagnostic_msgs_generate_messages_cpp
Built target std_msgs_generate_messages_cpp
Built target sensor_msgs_generate_messages_cpp
[  0%] [  0%] Built target sensor_msgs_generate_messages_py
Built target std_msgs_generate_messages_py
[  0%] Built target diagnostic_msgs_generate_messages_py
[  0%] Built target diagnostic_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_lisp
[  3%] Built target mavconn
[ 28%] Built target mavros_generate_messages_cpp
[ 53%] Built target mavros_generate_messages_lisp
[ 80%] Built target mavros_generate_messages_py
[ 80%] Built target mavros_generate_messages
make[2]: *** No rule to make target `/usr/lib/libboost_signals-mt.so', needed by `/home/dukerama/catkin_ws/devel/lib/libmavros.so'.  Stop.
make[1]: *** [mavros/mavros/CMakeFiles/mavros.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

vooon, 2015-06-24:
Try to make symlink `libboost_signals-mt.so` to `libboost_signals.so` in `/usr/lib`.

Tomofoye, 2015-06-24:
I just tried that, unfortunately I still get the same error :(

vooon, 2015-06-28:
Updates?

Tomofoye, 2015-06-29:
For now I just down graded to hydro and I was able to get mavros to work on that. I couldn't get it to work on indigo, let me know if you have anymore suggestions or ideas of what I could try.

vooon, 2015-06-29:
You can try Jade, Hydro EOLed.

closed 2015-07-09 by vooon