mavlink/mavros #319: Transfer math to Eigen

vooon, 2015-06-28:
We already have some matrix math done by hand and TF bullet.
Also other dev's use Eigen with TF2, so seems that it required for #71.

TSC21, 2015-06-28:
:+1: finally! Just let me know what's needed.

vooon, 2015-07-09:
Ok, all code without tf listener are moved.
Exception is visualization because #336.

vooon, 2015-07-10:
Ok, #71 now closed. So most of tf::Vector3 and tf::Quaternion are gone.

But i find strange code at distance_sensor, @TSC21 how RPY-Q may work if you give degrees to that function?
It wants radians. I think to refactor orientation enum.
I don't see where really may be needed euler angles (in degrees), while quaternions may be useful.
(even in your code - *single usage of method* - converts to q).

closed 2015-07-23 by vooon