mavlink/mavros #321: Write unit tests for transform_frame utilities

vooon, 2015-06-29:


vooon, 2015-06-29:
Wrote part, got error:

```
-- run_tests.py: execute commands
  /home/vovan/ros/devel/lib/mavros/libmavros-frame-test --gtest_output=xml:/home/vovan/ros/build/mavros/test_results/mavros/gtest-libmavros-frame-test.xml
[==========] Running 9 tests from 2 test cases.
[----------] Global test environment set-up.
[----------] 8 tests from VECTOR
[ RUN      ] VECTOR.transform_frame_xyz_111
[       OK ] VECTOR.transform_frame_xyz_111 (0 ms)
[ RUN      ] VECTOR.transform_frame_xyz_pi00
[       OK ] VECTOR.transform_frame_xyz_pi00 (0 ms)
[ RUN      ] VECTOR.transform_frame_xyz_0pi0
[       OK ] VECTOR.transform_frame_xyz_0pi0 (0 ms)
[ RUN      ] VECTOR.transform_frame_xyz_00pi
[       OK ] VECTOR.transform_frame_xyz_00pi (0 ms)
[ RUN      ] VECTOR.transform_frame_attitude_rpy_111
/home/vovan/ros/src/mavros/mavros/test/test_frame_conv.cpp:60: Failure
Value of: out
  Actual: 32-byte object <8C-16 22-AA FD-90 10-40 00-00 00-00 00-00 F0-3F 00-00 00-00 00-00 F0-3F 00-00 00-00 00-00 00-00>
Expected: expected_out
Which is: 32-byte object <00-00 00-00 00-00 00-00 00-00 00-00 00-00 F0-BF 00-00 00-00 00-00 F0-BF 00-00 00-00 00-00 00-00>
[  FAILED  ] VECTOR.transform_frame_attitude_rpy_111 (1 ms)
[ RUN      ] VECTOR.transform_frame_attitude_rpy_pi00
/home/vovan/ros/src/mavros/mavros/test/test_frame_conv.cpp:69: Failure
Value of: out
  Actual: 32-byte object <18-2D 44-54 FB-21 19-40 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00>
Expected: expected_out
Which is: 32-byte object <18-2D 44-54 FB-21 09-40 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00>
[  FAILED  ] VECTOR.transform_frame_attitude_rpy_pi00 (0 ms)
[ RUN      ] VECTOR.transform_frame_attitude_rpy_0pi0
/home/vovan/ros/src/mavros/mavros/test/test_frame_conv.cpp:78: Failure
Value of: out
  Actual: 32-byte object <18-2D 44-54 FB-21 09-40 18-2D 44-54 FB-21 09-40 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00>
Expected: expected_out
Which is: 32-byte object <00-00 00-00 00-00 00-00 18-2D 44-54 FB-21 09-40 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00>
[  FAILED  ] VECTOR.transform_frame_attitude_rpy_0pi0 (0 ms)
[ RUN      ] VECTOR.transform_frame_attitude_rpy_00pi
/home/vovan/ros/src/mavros/mavros/test/test_frame_conv.cpp:87: Failure
Value of: out
  Actual: 32-byte object <18-2D 44-54 FB-21 09-40 00-00 00-00 00-00 00-00 18-2D 44-54 FB-21 09-40 00-00 00-00 00-00 00-00>
Expected: expected_out
Which is: 32-byte object <00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 18-2D 44-54 FB-21 09-40 00-00 00-00 00-00 00-00>
[  FAILED  ] VECTOR.transform_frame_attitude_rpy_00pi (0 ms)
[----------] 8 tests from VECTOR (1 ms total)

[----------] 1 test from QUATERNION
[ RUN      ] QUATERNION.transform_frame_attitude_q_111
[ INFO] [1435592908.513114350]: Input: 0.167519 0.570941 0.167519 0.786067
[ INFO] [1435592908.513205891]: Output: 0.786067 0.167519 -0.570941 -0.167519
[ INFO] [1435592908.513255020]: Exp. expected_output: 0.167519 -0.570941 -0.167519 0.786067
/home/vovan/ros/src/mavros/mavros/test/test_frame_conv.cpp:104: Failure
Value of: out
  Actual: 32-byte object <A0-D4 13-34 75-27 E9-3F 6A-32 EF-78 41-71 C5-3F A8-2B 6D-0C 27-45 E2-BF 67-32 EF-78 41-71 C5-BF>
Expected: expected_out
Which is: 32-byte object <69-32 EF-78 41-71 C5-3F A8-2B 6D-0C 27-45 E2-BF 69-32 EF-78 41-71 C5-BF A0-D4 13-34 75-27 E9-3F>
[  FAILED  ] QUATERNION.transform_frame_attitude_q_111 (1 ms)
[----------] 1 test from QUATERNION (1 ms total)

[----------] Global test environment tear-down
[==========] 9 tests from 2 test cases ran. (2 ms total)
[  PASSED  ] 4 tests.
[  FAILED  ] 5 tests, listed below:
[  FAILED  ] VECTOR.transform_frame_attitude_rpy_111
[  FAILED  ] VECTOR.transform_frame_attitude_rpy_pi00
[  FAILED  ] VECTOR.transform_frame_attitude_rpy_0pi0
[  FAILED  ] VECTOR.transform_frame_attitude_rpy_00pi
[  FAILED  ] QUATERNION.transform_frame_attitude_q_111

 5 FAILED TESTS
```

mhkabir, 2015-06-29:
@TSC21 Can you please have a check and verify the quaternion order for tranforms?

vooon, 2015-06-29:
Test from https://github.com/mavlink/mavros/commit/c91e39590dfcf056590a1c0a20b1129889643fc8 dont work! @TSC21 what wrong?
```
-- run_tests.py: execute commands
  /home/vovan/ros/devel/lib/mavros/libmavros-frame-test --gtest_output=xml:/home/vovan/ros/build/mavros/test_results/mavros/gtest-libmavros-frame-test.xml
[==========] Running 12 tests from 2 test cases.
[----------] Global test environment set-up.
[----------] 8 tests from VECTOR
[ RUN      ] VECTOR.transform_frame_xyz_111
[ INFO] [1435595758.726338027]: In (x y z): 1.000000 1.000000 1.000000
[ INFO] [1435595758.726378237]: Expected:   1.000000 -1.000000 -1.000000
[ INFO] [1435595758.726400802]: Out:        1.000000 -1.000000 -1.000000
[       OK ] VECTOR.transform_frame_xyz_111 (0 ms)
[ RUN      ] VECTOR.transform_frame_xyz_pi00
[ INFO] [1435595758.726458002]: In (x y z): 3.141593 0.000000 0.000000
[ INFO] [1435595758.726477923]: Expected:   3.141593 0.000000 0.000000
[ INFO] [1435595758.726496865]: Out:        3.141593 -0.000000 -0.000000
[       OK ] VECTOR.transform_frame_xyz_pi00 (0 ms)
[ RUN      ] VECTOR.transform_frame_xyz_0pi0
[ INFO] [1435595758.726540654]: In (x y z): 0.000000 3.141593 0.000000
[ INFO] [1435595758.726560370]: Expected:   0.000000 -3.141593 0.000000
[ INFO] [1435595758.726579140]: Out:        0.000000 -3.141593 -0.000000
[       OK ] VECTOR.transform_frame_xyz_0pi0 (0 ms)
[ RUN      ] VECTOR.transform_frame_xyz_00pi
[ INFO] [1435595758.726621050]: In (x y z): 0.000000 0.000000 3.141593
[ INFO] [1435595758.726640458]: Expected:   0.000000 0.000000 -3.141593
[ INFO] [1435595758.726658737]: Out:        0.000000 -0.000000 -3.141593
[       OK ] VECTOR.transform_frame_xyz_00pi (0 ms)
[ RUN      ] VECTOR.transform_frame_attitude_rpy_111
[ INFO] [1435595758.726701352]: In (x y z): 1.000000 1.000000 1.000000
[ INFO] [1435595758.726720770]: Expected:   0.000000 -1.000000 -1.000000
[ INFO] [1435595758.726739243]: Out:        4.141593 1.000000 1.000000
/home/vovan/ros/src/mavros/mavros/test/test_frame_conv.cpp:85: Failure
Value of: out
  Actual: 32-byte object <8C-16 22-AA FD-90 10-40 00-00 00-00 00-00 F0-3F 00-00 00-00 00-00 F0-3F 00-00 00-00 00-00 00-00>
Expected: expected_out
Which is: 32-byte object <00-00 00-00 00-00 00-00 00-00 00-00 00-00 F0-BF 00-00 00-00 00-00 F0-BF 00-00 00-00 00-00 00-00>
[  FAILED  ] VECTOR.transform_frame_attitude_rpy_111 (0 ms)
[ RUN      ] VECTOR.transform_frame_attitude_rpy_pi00
[ INFO] [1435595758.726987526]: In (x y z): 3.141593 0.000000 0.000000
[ INFO] [1435595758.727021779]: Expected:   3.141593 0.000000 0.000000
[ INFO] [1435595758.727040340]: Out:        6.283185 0.000000 0.000000
/home/vovan/ros/src/mavros/mavros/test/test_frame_conv.cpp:96: Failure
Value of: out
  Actual: 32-byte object <18-2D 44-54 FB-21 19-40 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00>
Expected: expected_out
Which is: 32-byte object <18-2D 44-54 FB-21 09-40 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00>
[  FAILED  ] VECTOR.transform_frame_attitude_rpy_pi00 (1 ms)
[ RUN      ] VECTOR.transform_frame_attitude_rpy_0pi0
[ INFO] [1435595758.727162224]: In (x y z): 0.000000 3.141593 0.000000
[ INFO] [1435595758.727182810]: Expected:   0.000000 -3.141593 0.000000
[ INFO] [1435595758.727201497]: Out:        3.141593 3.141593 0.000000
/home/vovan/ros/src/mavros/mavros/test/test_frame_conv.cpp:107: Failure
Value of: out
  Actual: 32-byte object <18-2D 44-54 FB-21 09-40 18-2D 44-54 FB-21 09-40 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00>
Expected: expected_out
Which is: 32-byte object <00-00 00-00 00-00 00-00 18-2D 44-54 FB-21 09-C0 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00>
[  FAILED  ] VECTOR.transform_frame_attitude_rpy_0pi0 (0 ms)
[ RUN      ] VECTOR.transform_frame_attitude_rpy_00pi
[ INFO] [1435595758.727323345]: In (x y z): 0.000000 0.000000 3.141593
[ INFO] [1435595758.727343568]: Expected:   0.000000 0.000000 -3.141593
[ INFO] [1435595758.727361578]: Out:        3.141593 0.000000 3.141593
/home/vovan/ros/src/mavros/mavros/test/test_frame_conv.cpp:118: Failure
Value of: out
  Actual: 32-byte object <18-2D 44-54 FB-21 09-40 00-00 00-00 00-00 00-00 18-2D 44-54 FB-21 09-40 00-00 00-00 00-00 00-00>
Expected: expected_out
Which is: 32-byte object <00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 18-2D 44-54 FB-21 09-C0 00-00 00-00 00-00 00-00>
[  FAILED  ] VECTOR.transform_frame_attitude_rpy_00pi (0 ms)
[----------] 8 tests from VECTOR (1 ms total)

[----------] 4 tests from QUATERNION
[... 32 lines cut ...]
[ INFO] [1435595758.728180721]: Expected:     0.000000 0.000000 -1.000000 0.000000
[ INFO] [1435595758.728199710]: Out:          0.000000 1.000000 0.000000 0.000000
/home/vovan/ros/src/mavros/mavros/test/test_frame_conv.cpp:165: Failure
Value of: out
  Actual: 32-byte object <07-5C 14-33 26-A6 91-3C 00-00 00-00 00-00 F0-3F 07-5C 14-33 26-A6 91-3C 48-5D 8A-85 CE-77 33-39>
Expected: expected_out
Which is: 32-byte object <00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00 F0-BF 07-5C 14-33 26-A6 91-3C>
[  FAILED  ] QUATERNION.transform_frame_attitude_q_00pi (0 ms)
[----------] 4 tests from QUATERNION (1 ms total)

[----------] Global test environment tear-down
[==========] 12 tests from 2 test cases ran. (2 ms total)
[  PASSED  ] 4 tests.
[  FAILED  ] 8 tests, listed below:
[  FAILED  ] VECTOR.transform_frame_attitude_rpy_111
[  FAILED  ] VECTOR.transform_frame_attitude_rpy_pi00
[  FAILED  ] VECTOR.transform_frame_attitude_rpy_0pi0
[  FAILED  ] VECTOR.transform_frame_attitude_rpy_00pi
[  FAILED  ] QUATERNION.transform_frame_attitude_q_111
[  FAILED  ] QUATERNION.transform_frame_attitude_q_pi00
[  FAILED  ] QUATERNION.transform_frame_attitude_q_0pi0
[  FAILED  ] QUATERNION.transform_frame_attitude_q_00pi

 8 FAILED TESTS
```

TSC21, 2015-06-29:
GTests are failing cause the expected outputs are wrong. For example in https://github.com/mavlink/mavros/blob/master/mavros/test/test_frame_conv.cpp#L160, 
the result should be `tf::createQuaternionFromRPY(M_PI, 0, M_PI);`
and not `tf::createQuaternionFromRPY(0, 0, -M_PI);`. Don't mess rotation matrices with resulting RPY values.

TSC21, 2015-06-29:
Also you are feeding `1 rad` values here: https://github.com/mavlink/mavros/blob/master/mavros/test/test_frame_conv.cpp#L127. Not a problem but values like these should be applied to positions, and not angles.

TSC21, 2015-06-29:
Please change outputs in https://github.com/mavlink/mavros/blob/master/mavros/test/test_frame_conv.cpp#L91, https://github.com/mavlink/mavros/blob/master/mavros/test/test_frame_conv.cpp#L102, https://github.com/mavlink/mavros/blob/master/mavros/test/test_frame_conv.cpp#L113, https://github.com/mavlink/mavros/blob/master/mavros/test/test_frame_conv.cpp#L127, https://github.com/mavlink/mavros/blob/master/mavros/test/test_frame_conv.cpp#L138, https://github.com/mavlink/mavros/blob/master/mavros/test/test_frame_conv.cpp#L149 and https://github.com/mavlink/mavros/blob/master/mavros/test/test_frame_conv.cpp#L160.

TSC21, 2015-06-29:
Relatively to `M_PI` vs `boost::math::constants::pi<double>()`: https://meetingcpp.com/index.php/br/items/cpp-and-pi.html.
> Thats why M_PI or boost can offer an alternative, especially if you don't know pi from your head, so no need to look it up. Boost has another advantage: It works with types of multiple precision, as stated on reddit, and as you can see in the example of boost.multiprecision.

So I think that boost wins in these case, since we already use boost lib in our code. But again, if "like" or "not like" is a valid metric, seems I don't have much of an opinion here.

TSC21, 2015-06-29:
Solved with https://github.com/mavlink/mavros/pull/322

vooon, 2015-07-01:
https://youtu.be/Q1_wX4SEb1A

vooon, 2015-07-01:
<iframe width="560" height="315" src="https://www.youtube.com/embed/Q1_wX4SEb1A" frameborder="0" allowfullscreen></iframe>

vooon, 2015-07-01:
Attitude transform are wrong: https://youtu.be/Q1_wX4SEb1A

TSC21, 2015-07-01:
why you have the axis upside-down?

TSC21, 2015-07-01:
The same rotation is both applied on rpy and quaternion transformation, and both return exactly the same result. So transform is right, and I have confirmed it with math. Also @blackcoder was able to flight test this transformation (even with POSCTL), so I don't see where's this is wrong. The previous conversion you add was `roll, -pitch, -yaw` and adding PI to Roll results exactly in the same rotation cause (+PI,0,0) results in a rotation matrix with the following form {{1,0,0},{0,-1,0},{0,0,-1}}.

vooon, 2015-07-01:
> why you have the axis upside-down?

Good question to You! I replaced attitude math with xyz and got normal position of axes.
With your math pitch and yaw has wrong sign, you may notice that in GCS nose upper horizon, while in rviz down.

vooon, 2015-07-01:
So simple add 180° to roll DO NOT WORK. If you want do rotation, do rotation matrix and apply it.

vooon, 2015-07-01:
And i'm going to release in current state, without that math.
While transforms on "local" (previous y x -z) changed to common x -y -z, i think that it is better, than completely broken system.

TSC21, 2015-07-01:
Well that's unexpected... I wonder how @blackcoder got it to work then

TSC21, 2015-07-01:
So the current transforms are then x,-y,-z,roll,-pitch,-yaw?

TSC21, 2015-07-01:
Ok now i know the reason why it worked with Tarek: mocap plugin uses quaternions and the rotation yaw roll is transformed into a quaternion which acts as a rotation to the other one. Multiplying both results on the correct transformation. So my mistake on this, as I thougth summing the rotation was similar to applying the rotation matrix which is obviously wrong. Thanks for finding that out @vooon.

TSC21, 2015-07-01:
Why you changed quaternions transform too @vooon? Those are right cause quaternions act as a form of rotation... So th PI on Roll is converted to quaternion, and that math is right!

vooon, 2015-07-01:
Here covariance transform with eigen: https://github.com/ros-perception/imu_pipeline/blob/indigo-devel/imu_transformer/include/imu_transformer/tf2_sensor_msgs.h#L69-L75

@TSC21 math looks exact what you statement :)

TSC21, 2015-07-01:
> Here covariance transform with eigen: https://github.com/ros-perception/imu_pipeline/blob/indigo-devel/imu_transformer/include/imu_transformer/tf2_sensor_msgs.h#L69-L75

That's exactly what I need :D thanks for pointing that @vooon. Now we need Eigen on business...

vooon, 2015-07-01:
Can you please run imu test that i commited in test_package branch?

TSC21, 2015-07-01:
> Can you please run imu test that i commited in test_package branch?

Hi will, but I'm out of bandwidth now :S

vooon, 2015-07-01:
APM SITL + mavros: https://youtu.be/mUIptiNbmS4

Here tested base plugins.

vooon, 2015-07-01:
@TSC21 please check. I wrote simple eigen-based transforms.
Vector's one seems ok, while quaternion require check.

There is no function similar to `tf::quaternionFromRPY()`, and use `Eigen::AxisAngle()` in different order gives different result.
We should check that order.

vooon, 2015-07-01:
Added a test, and seems that right order is same as in [MIRA quaternionFromYawPitchRoll()][1]

[1]: http://www.mira-project.org/MIRA-doc/YawPitchRoll_8h_source.html#l00153

vooon, 2015-07-15:
Almost all test implemented, except `UAS::transform_frame(Covariance6d)` (not implemented).

vooon, 2015-07-23:
All used transforms have tests. Closing.

closed 2015-07-23 by vooon