mavlink/mavros #324: Interface PX4 SITLto MAVROS test suite

LorenzMeier, 2015-07-01:
@vooon @TSC21 I would appreciate your support in interfacing the PX4 SITL to the MAVROS test suite. We actually have two versions:

1) Minimal ROS-only SITL: It only simulates the controllers, but *everything* is running as native ROS nodes. This is what we use for our jenkins server:
https://pixhawk.org/dev/ros/sitl_testing

The jenkins server is here and tests every PR against PX4/Firmware:
http://sitl01.dronetest.io

The *important* part is that it tests *everything* and *before* it enters the main codebase. So its a very effective gate, and this is what mavros will need as well.

2) Full system SITL: This simulates the complete autopilot and essentially works like the APM SITL: 
https://github.com/PX4/Firmware/tree/master/posix-configs/SITL

It will start a mavlink instance on UDP port 14556 on localhost, to which you should be able to connect mavros to.

TSC21, 2015-07-01:
@LorenzMeier which one do you think we should face first? The 2nd one is the one which uses Rotors_Simulator?

vooon, 2015-07-01:
Actually i wanted to do testing stuff in after 0.12, because there already some things that i want to release:
- `--wait` flag is very useful for launch scripts
- param changes: period to rate.

But that change in frame transform too deep and stopping me. I agree that CI testing is great thing.

TSC21, 2015-07-01:
Well this will help: https://pixhawk.org/dev/ros/sitl_testing

TSC21, 2017-04-04:
Implementation on "test_mavros". Forward work on it will be considered. Closing it

closed 2017-04-04 by TSC21