mavlink/mavros #325: Merge gps and global_position plugins

vooon, 2015-07-02:
I think we may merge that plugins, because their output needed for one purpose: provide global position data.

TSC21, 2015-07-02:
Seems good. Assign that task to me, if you may

vooon, 2015-07-02:
First we need to discus proposal.

TSC21, 2015-07-02:
What do you have in mind?

vooon, 2015-07-02:
Past versions of APM don't send fused global position, so for that old firmwares blacklisting global_position may save little ram.
But those plugins are coupled, because gp can't properly work without gps.

So in one hand i have splitted raw and fused data and possibility to disable fused part
and in other e.g. apm antenna tracker uses only fused coordinates.

TSC21, 2015-07-02:
You can use `#if #endif` clauses depending on the autopilot. So then you can keep the plugin active. What do you think?

vooon, 2015-07-02:
Compile time option not useful, ros buildfarm cant produce several packages with different compile time options.

Currently ros binaries uses APM's dialect, but because almost all used messages exists in common PX4 don't require special binary.

TSC21, 2015-07-02:
So we can add a parameter `is_raw`, which when set publishes on raw gps, but when not set publishes filtered position.

vooon, 2015-07-02:
No, stream raw and fused, just like imu_pub.

TSC21, 2015-07-02:
At the same time but for different topics, right? Seems legit then.

vooon, 2015-07-02:
Merge is done. GPS raw topics now in `~global_position/raw`, and we moved to TF2.
Also parameters changed, tf things moved to `tf` subnamespace.

closed 2015-07-02 by vooon