mavlink/mavros #326: [setpoint_attitude] Reduce message type

vooon, 2015-07-02:
Do **WE REALLY NEED** all this message type variations?

I mean that Twist seems ok, (but should be angular + linear.z or x as thrust?),
but PoseStamped and PoseWithCovarianceStamped?

TSC21, 2015-07-02:
Here's the reason: https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/setpoint_attitude.cpp#L53

TSC21, 2015-07-02:
Well you can always delete `PoseWithCovarianceStamped` if you want, as covariance values are not used on setpoints.

TSC21, 2015-07-02:
> but should be angular + linear.z or x as thrust

Well thrust is not the same as linear velocity in Z, because thrust is a force, so it's measured in Newtons, and linear.Z is in meters/sec.

vooon, 2015-07-02:
I see unneeded generalization. In theory there may be exist FCU with support
for other features other than APM or PX4.

But really i want to do one good product for existing and known autopilots, and support all or at least most used functions of that AP.
You know, i tried to do bridge for OpenPilot, but their too complex protocol and **small market propagation** stopped me (i mean revolution, not CC).

vooon, 2015-07-02:
> Well thrust is not the same as linear velocity in Z

Got it.

TSC21, 2015-07-02:
> I see unneeded generalization. In theory there may be exist FCU with support
for other features other than APM or PX4.

In this case it's not a generelization for FCU's, it's a control mode and it's required when you just send attitude setpoint to the controller. Check https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/setpoint_attitude.cpp#L124 (https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/setpoint_attitude.cpp#L128).

> But really i want to do one good product for existing and known autopilots, and support all or at least most used functions of that AP.

That's what we all want, right? :)

> You know, i tried to do bridge for OpenPilot, but their too complex protocol and small market propagation stopped me (i mean revolution, not CC).

Yes I'm aware of OPGateway, which I'm possibly interested to continue when I have some time (we had some conversation regarding it's architecture few months ago, remember?).

vooon, 2015-07-10:
I removed PoseWithCovarianceStamped.

closed 2015-07-10 by vooon