mavlink/mavros #327: offboard reject

dreamkeep, 2015-07-02:
@vooon  when  i switch to offboard ,it report that:

[ WARN] [1435839397.770402629]: GPS: no fix
[ERROR] [1435839407.447141970]: FCU: REJECT OFFBOARD
[ERROR] [1435839417.539181644]: FCU: REJECT OFFBOARD

when I run  rosrun mavros mavsys mode -c OFFBOARD

it show :
Timed out!

this is my comand list:

roslaunch mavros px4.launch &

px4@ubilinux:~/ros_catkin_ws$ roslaunch mavros px4.launch &
[1] 2021
px4@ubilinux:~/ros_catkin_ws$ ... logging to /home/px4/.ros/log/b54007a4-20bb-11e5-b2f7-784b879ee7ab/roslaunch-ubilinux-2021.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubilinux:57735/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat: 5.0
 * /mavros/conn/system_time: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync: 0.1
 * /mavros/fcu_url: /dev/ttyMFD1:57600
 * /mavros/gcs_url: udp://:14556@192....
 * /mavros/global_position/child_frame_id: fcu
 * /mavros/global_position/frame_id: local_origin
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/send_tf: False
 * /mavros/gps/frame_id: gps
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/child_frame_id: fcu
 * /mavros/local_position/frame_id: local_origin
 * /mavros/local_position/send_tf: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
 * /mavros/mocap/use_tf: False
 * /mavros/plugin_blacklist: ['image_pub', '3d...
 * /mavros/plugin_whitelist: []
 * /mavros/px4flow/frame_id: px4flow
 * /mavros/px4flow/ranger_fov: 0.0
 * /mavros/px4flow/ranger_max_range: 5.0
 * /mavros/px4flow/ranger_min_range: 0.3
 * /mavros/safety_area/p1/x: 1.0
 * /mavros/safety_area/p1/y: 1.0
 * /mavros/safety_area/p1/z: 1.0
 * /mavros/safety_area/p2/x: -1.0
 * /mavros/safety_area/p2/y: -1.0
 * /mavros/safety_area/p2/z: -1.0
 * /mavros/setpoint/accel/send_force: False
 * /mavros/setpoint_accel/send_force: False
 * /mavros/setpoint_attitude/child_frame_id: attitude
[... 274 lines cut ...]
    name: mavros: Time Sync
    message: No events recorded.
    hardware_id: /dev/ttyMFD1:57600
    values: 
      - 
        key: Timesyncs since startup
        value: 0
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Last dt (ms)
        value: 0.000000
      - 
        key: Mean dt (ms)
        value: 0.000000
      - 
        key: Last system time (s)
        value: 0.000000000
      - 
        key: Time offset (s)
        value: 0.000000000
---

what should i do?

TSC21, 2015-07-02:
@dreamkeep are you sending any kind of setepoint stream to the FCU? (using setpoint plugins for example). Cause you have to. Otherwise FCU will reject changing to `OFFBOARD` mode 'cause it doesn't detect any external setpoint.

wangsen1312, 2015-07-02:
I have meet this. Now I use RC change to the Manual Mode and Offboard Mode, which seem works well.

dreamkeep, 2015-07-02:
I use RC change to the Manual mode is ok.

I use this setpoint example:

#!/usr/bin/env python
# vim:set ts=4 sw=4 et:
#
# Copyright 2015 UAVenture AG.
#
# This program is free software; you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation; either version 3 of the License, or
# (at your option) any later version.
#
# This program is distributed in the hope that it will be useful, but
# WITHOUT ANY WARRANTY; without even the implied warranty of MERCHANTABILITY
# or FITNESS FOR A PARTICULAR PURPOSE. See the GNU General Public License
# for more details.
#
# You should have received a copy of the GNU General Public License along
# with this program; if not, write to the Free Software Foundation, Inc.,
# 59 Temple Place, Suite 330, Boston, MA 02111-1307 USA
# Updated: Tarek Taha : tarek.taha@kustar.ac.ae
#    - Changed topic names after re-factoring : https://github.com/mavlink/mavros/issues/233
 
import rospy
import thread
import threading
import time
 
from geometry_msgs.msg import PoseStamped, Quaternion
from math import *
from mavros.srv import CommandBool
from mavros.utils import *
from std_msgs.msg import Header
from std_msgs.msg import String
from tf.transformations import quaternion_from_euler
 
class Setpoint:
 
    def __init__(self, pub, rospy):
        self.pub = pub
        self.rospy = rospy
 
        self.x = 0.0
        self.y = 0.0
        self.z = 0.0
 
        try:
            thread.start_new_thread( self.navigate, () )
        except:
            print "Error: Unable to start thread"
 
        # TODO(simon): Clean this up.
        self.done = False
        self.done_evt = threading.Event()
        sub = rospy.Subscriber('/mavros/local_position/local', PoseStamped, self.reached)
 
    def navigate(self):
        rate = self.rospy.Rate(10) # 10hz
 
        msg = PoseStamped()
        msg.header = Header() 
        msg.header.frame_id = "base_footprint"
        msg.header.stamp = rospy.Time.now()
 
        while 1:
            msg.pose.position.x = self.x
            msg.pose.position.y = self.y
            msg.pose.position.z = self.z
 
            # For demo purposes we will lock yaw/heading to north.
            yaw_degrees = 0  # North
            yaw = radians(yaw_degrees)
            quaternion = quaternion_from_euler(0, 0, yaw)
            msg.pose.orientation = Quaternion(*quaternion)
 
            self.pub.publish(msg)
 
            rate.sleep()
 
    def set(self, x, y, z, delay=0, wait=True):
        self.done = False
        self.x = x
        self.y = y
        self.z = z
 
        if wait:
            rate = rospy.Rate(5)
            while not self.done:
                rate.sleep()
 
        time.sleep(delay)
 
 
    def reached(self, topic):
            #print topic.pose.position.z, self.z, abs(topic.pose.position.z - self.z)
            if abs(topic.pose.position.x - self.x) < 0.5 and abs(topic.pose.position.y - self.y) < 0.5 and abs(topic.pose.position.z - self.z) < 0.5:
                self.done = True
 
            self.done_evt.set()
 
def setpoint_demo():
    pub = rospy.Publisher('/mavros/setpoint_position/local_position', PoseStamped, queue_size=10)
 
    rospy.init_node('pose', anonymous=True)
    rate = rospy.Rate(10) 
 
    setpoint = Setpoint(pub, rospy)
 
    print "Climb"
    setpoint.set(0.0, 0.0, 3.0, 0)
    setpoint.set(0.0, 0.0, 10.0, 5)
 
    print "Sink"
    setpoint.set(0.0, 0.0, 8.0, 5)
 
    print "Fly to the right"
    setpoint.set(10.0, 4.0, 8.0, 5)
 
    print "Fly to the left"
    setpoint.set(0.0, 0.0, 8.0, 5)
 
    offset_x = 0.0
    offset_y = 0.0
    offset_z = 10.0
    sides = 360
    radius = 20
 
    print "Fly in a circle"
    setpoint.set(0.0, 0.0, 10.0, 3)   # Climb to the starting height first
    i = 0
    while not rospy.is_shutdown():
        x = radius * cos(i*2*pi/sides) + offset_x   
        y = radius * sin(i*2*pi/sides) + offset_y
        z = offset_z
 
        wait = False
        delay = 0
        if (i == 0 or i == sides):
            # Let it reach the setpoint.
            wait = True
            delay = 5
 
        setpoint.set(x, y, z, delay, wait)
 
        i = i + 1 
        rate.sleep()
 
        if (i > sides):
            print "Fly home"
            setpoint.set(0.0, 0.0, 10.0, 5)
            break
 
    # Simulate a slow landing.
    setpoint.set(0.0, 0.0,  8.0, 5)
    setpoint.set(0.0, 0.0,  3.0, 5)
    setpoint.set(0.0, 0.0,  2.0, 2)
    setpoint.set(0.0, 0.0,  1.0, 2)
    setpoint.set(0.0, 0.0,  0.0, 2)
    setpoint.set(0.0, 0.0, -0.2, 2)
 
    print "Bye!"
 
 
if __name__ == '__main__':
    try:
        setpoint_demo()
    except rospy.ROSInterruptException:
        pass

wangsen1312, 2015-07-02:
In the def __init__(self, pub, rospy): part   sub = rospy.Subscriber('/mavros/local_position/local', PoseStamped, self.reached)
may should change to /mavros/setpoint_position/local_position'

TSC21, 2015-07-02:
> In the def init(self, pub, rospy): part sub = rospy.Subscriber('/mavros/local_position/local', PoseStamped, self.reached)
may should change to /mavros/setpoint_position/local_position'

No, that's a subscription. Doesn't have nothing to do with the setpoints. Those are being published through `pub = rospy.Publisher('/mavros/setpoint_position/local_position', PoseStamped, queue_size=10)`

vooon, 2015-07-02:
1. please use code block tags for listings
2. no gps fix, wp & pr timeout cause me that your have bad link to fcu.
3. don't use & (foreground) executing or roslaunch, that not error, but else you can miss log reports

TSC21, 2015-07-02:
@dreamkeep do you have position lock on your FCU (i.e. big status LED blinking green instead of blue)?

TSC21, 2015-07-22:
@dreamkeep what's the status on this?

closed 2015-07-24 by vooon