mavlink/mavros #328: [commad] Change CommandHome.msg

vooon, 2015-07-02:
Maybe use [geographic_msgs/GeoPoint.msg][1] instead of simple LLA?
I'm going to add home position publishing (PX4), and seems that this message well fit.

[1]: http://docs.ros.org/api/geographic_msgs/html/msg/GeoPoint.html

LorenzMeier, 2015-07-02:
Wait wait - we're in the process of a cleanup sprint for home handling. What was your plan here?

vooon, 2015-07-02:
Just change message format to:

```
bool current_gps
geographic_msgs/GeoPoint home
---
bool success
uint8 result
```

vooon, 2015-07-02:
As i seen in sources, home position streamed every 2 seconds. But if your going to change i can wait.
What PR/issue?

vooon, 2015-07-02:
With proposal https://github.com/mavlink/mavlink/pull/412 that change not make sense, closing.

closed 2015-07-02 by vooon