mavlink/mavros #330: undefined symbol: _ZN3ros8DurationC1ERKNS_4RateE

jabberwack, 2015-07-03:
Hello. I'm new to mavros and I'm getting this error whenever I launch the node.

/home/odroid/catkin_ws/devel/lib/mavros/mavros_node: symbol lookup error: /home/odroid/catkin_ws/devel/lib//libmavros_plugins.so: undefined symbol: _ZN3ros8DurationC1ERKNS_4RateE

can anyone help?

I installed mavros from source.

Thank you.

TSC21, 2015-07-03:
That's because mavros lib is not being linked. Please delete `devel` and `build` folders from your `catlin_ws`, delete CMakeList file on your src folder, do `catkin_init_workspace` on it again then do `catkin_make` on home folder of your workspace.

TSC21, 2015-07-03:
Also make sure you have all dependencies installed. Did you followed the instructions from https://github.com/mavlink/mavros/blob/master/mavros/README.md?

TSC21, 2015-07-03:
I advise to also use `catkin_tools` if you are using Jade. You can installed it on Indigo following http://catkin-tools.readthedocs.org/en/latest/installing.html. Much more versatile.

vooon, 2015-07-03:
What version of ROS do you use?

jabberwack, 2015-07-06:
Hello guys. Thnx for the help. Really appreciate it.

@TSC21 I tries deleteing devel and build folders as you said but launching the px4.launch just gives me the same error.
Something might have gone wrong when I removed the apt-get version of the MAVROS and re-compiled it from the source.
I'm thinking about just starting from scratch.

@vooon I'm using Indigo.

Thank you.

jabberwack, 2015-07-06:
Hey guys.

So I installed everything again and I'm still getting the exact same error.

/home/odroid/catkin_ws/devel/lib/mavros/mavros_node: symbol lookup error: /home/odroid/catkin_ws/devel/lib//libmavros_plugins.so: undefined symbol: _ZN3ros8DurationC1ERKNS_4RateE

The px4flow.launch of mavros_extras works fine.

Btw, I installed mavlink from source code that I got from mavlink-gbp-release.
The apt-get version was giving me error about 'MAV_SENSOR_ORIENTATION has not been declared.'

vooon, 2015-07-06:
Use mavlink from [shadow repo][1], main repo not updated yet (also note that next release tomorrow).

`_ZN3ros8DurationC1ERKNS_4RateE` it is not mavros or mavlink part, it is ROS part.
Strange that i can't find that definition in my Jade installation (while libmavros_plugins.so has it).

[1]: http://packages.ros.org/ros-shadow-fixed/ubuntu/pool/main/r/ros-indigo-mavlink/

vooon, 2015-07-06:
Just to point: i think it is constructor for `ros::Duration(ros::Rate)`.

vooon, 2015-07-06:
Hmm, related: https://github.com/ros/roscpp_core/issues/30
https://github.com/ros/roscpp_core/commit/26e74bcbc759f4c7a78b7f12a2f66068140fc063

vooon, 2015-07-06:
Ok, `/opt/ros` binaries stripped, so no symbols available.
I done `wstool set roscpp_core --git https://github.com/ros/roscpp_core.git -v indigo-devel` and then `catkin build`.

Then:
```
vovan@robert:~/ros/src$ objdump -x ../devel/lib/librostime.so | grep _ZN3ros8DurationC1ERKNS_4RateE
000000000002e1d0 g     F .text	00000000000000bc              _ZN3ros8DurationC1ERKNS_4RateE
```

jabberwack, 2015-07-07:
@vooon  Ok. Fixed the problem by reinstalling the roscpp_core.

Thank you so much.

closed 2015-07-07 by vooon

archenroot, 2017-06-14:
ufff, this also was causing issues with different package, but finally, I just used different term => environment.