mavlink/mavros #331: setpoint problem: Periodically jump

congleetea, 2015-07-04:
when I use mavros send some setpoint or local pose into fcu, I check some value in QGC, there are some problems:
1 as following picture, I use mocap or vision_pose plugin send relative information to fcu , LOCAL_POSITION_NED(x,y,z,vx,vy,vz) have a Periodically jump. and period time is two minutes.
[image]
2 when i use setpoint_attitude plugin to send some target attitude to fcu, it is strange to see that in the QGC , roll sp, pitch sp, yaw sp (or ATTITUDE_TARGET.q) will change to -2147483648 at the same time. as following picture. I set the target roll, pitch and yaw are (0.5, -0.4, -1.57)
[image]

[image]

someone know the reason and how to deal with it?

vooon, 2015-07-05:
-214... looks like integer overflow. Try also plot local position using rqt_plot.

congleetea, 2015-07-06:
@vooon  for the first question, I look local position in rqt_plot， it also shows as qgc. every 2 mins, local position x,y,z will jump. I think it is caused by position_estimator_inav app. 
for the second question of setpoint_attitude, I find the reason. I use setpoint_position and setpoint_attitude plugin at the same time. both plugins have orientation quaternion， in above picture, I don't set the orientation quaternion of setpoint_position, but set orientation quaternion of setpoint_attitude  as q=[w x y z]=[0.706477 0.035463 0.307553 0.636431] (conversing to euler is rpy=[0.5 0.4 1.57] as the picture), so the roll sp, pitch sp and yaw sp will sometimes are setpoint_position's target orientation, and sometimes are setpoint_attitude's target orientation. when the q=[0 0 0 0], qgc will show -214....  So , if we use setpoint_position and setpoint_attitude at the same time, both orientation should be set as same value.

however , i still don't know why the local position has two mins periodically jump.

TSC21, 2015-07-06:
@congleetea logs please

congleetea, 2015-07-06:
the above picture is captured when I just switch to offboard mode but not armd, it jumps larger. 
https://drive.google.com/file/d/0B6s1E1cM5KYrSzFLdFpwMk9jS2M/view?usp=sharing
I get this log when I switch to offboard mode and armed the mav, it jumps small, but also can find that it has a two mins periodically jump .

LorenzMeier, 2015-07-06:
Could you please share log file (.px4log from the microSD)? Its impossible to debug without.

congleetea, 2015-07-06:
@LorenzMeier  I have shared the .px4log file in this link, https://drive.google.com/file/d/0B6s1E1cM5KYrSzFLdFpwMk9jS2M/view?usp=sharing

congleetea, 2015-07-06:
btw, in these tests, I make the mav in the fixed position and not equip rotors.

Aperture-Laboratories, 2015-07-22:
http://dash.oznet.ch/view/WEkrRq7JjxQPVKtwj8a5j5  -  This is the log you uploaded. I put it through the log muncher.

When you want to upload logs, use http://dash.oznet.ch to upload and view your logs. Makes the data easier to understand.

TSC21, 2015-07-22:
@congleetea updates?

congleetea, 2015-07-26:
I am sorry for late, I have  not solved this problem,  I am very confused to something(what i want to do is trajectory tracking, so i will send target position xyz and target attitude rpy to fcu):
1) whether I can give setpoint_position and setpiont_attitude at the same time, because, in my test, if i send setpoint_attitude alone to FCU, the setpoint_attitude I see in QGC(roll sp, pitch sp, yaw sp) is correct , but if I send attitude and position at same time , roll sp and pitch sp will jump.like following picture.
[image]

2) @TSC21 I use your firmware (use mocap), in my test, I use optitrack, and remap mocap info to mocap/pose or vision_pose, i think if i keep the mav still, and move the marker , the local position and attitude of mav should change, however, the position and yaw will change corrrectly, roll and pitch can not change. so i think the firmware maybe not use roll and pitch information from topoic mocap/pose or vision_pose.
3) following log : i want the mav fly to point (0,0, 1.0m) with attitude rpy=(0,0,0). 
https://drive.google.com/file/d/0B6s1E1cM5KYra0c1ZlhXVHAwTkU/view?usp=sharing
in this test , i can not make the mav take off , i see the ATSP.ThrustSP like following picture:
[image]

someone can give me advice? thanks in advance.

LorenzMeier, 2015-07-26:
You cannot send the target attitude and position at the same time - if you write down the block diagram of your controller you will notice it makes no sense, and the transfer function from position to attitude is undefined.

You either need to send position, velocity and acceleration (and verify that the current control scheme is properly reading them in) or you need to run the complete controller on your computer and only send attitude.

Note that the argument I'm making is not one about technical limitations. The design of your controller is incomplete / incorrect, and you need to fix this fundamental issue first.

LorenzMeier, 2015-07-26:
I'm closing this since you confirmed that the root cause was an incorrect use of the interface. Please file new issues if you still face problems with the correct usage.

closed 2015-07-26 by LorenzMeier

congleetea, 2015-07-26:
OK, thank you very much. I think i understand it after review the controllor. thanks again!!