mavlink/mavros #334: Pixhawk Odroid UART connection

TiagoMMF, 2015-07-08:
Hi,

I've been trying to connect an Odroid XU3 to a Pixhawk, with PX4 firmware, using the UART port on the 30pin IO, using a level shifter between the two ports. I'm running Ubuntu 14.04 with mavros for ROS Jade. I'm getting a VER: request list timeout and a PR: request list timeout.  I've been able to get mavlink status and messages but cant get parameters. I've changes the SYS_COMPANION to 921600, and also added the extras.txt with the mavlink start command. On the /diagnostics it says it is connected. I searched but didn't find any solution. Any idea how to solve this?

Best regards 
Tiago

vooon, 2015-07-08:
1. You should use SYS_COMPANION xor extras.txt startup
2. If SYS_COMPANION == 921600, then url should be /dev/ttySAC0:921600
3. If there is no VER responses then old firmware or wrong target id's
4. Param and waypoint plugins fail if there target id's mismatch

TiagoMMF, 2015-07-09:
Thanks for the reply.

I'm thinking of using the serial 4/5 port for the UART comunication. Can the SYS_COMPANION parameter open that port as well or is it the command "mavlink start -d /dev/ttyS6 -b 921600 -p" on the /etc/extras.txt? Which would you advice using?

The PX4 version is the latest install from QGC, so I don't think it's old firmware.

The target id and component id are set to 1 and 50 respectively onQGC, so I believe it's correct.

vooon, 2015-07-09:
SYS_COMPANION now works only with TELEM2. So UART4 require custom startup in extras.txt.
Look [rsS L476](https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS#L476-L492).

TiagoMMF, 2015-07-14:
I have solved the issue. There was some problem with the level shifter which didn't enable transmission to the pixhawk.

closed 2015-07-14 by TiagoMMF

userpavanshetty, 2017-05-04:
Hello, Could you please how the connection is made between odroid and pixhawk using level shifter. Thanks

TSC21, 2017-05-04:
@userpavanshetty https://dev.px4.io/en/companion_computer/pixhawk_companion.html