mavlink/mavros #335: Mavros not Publishing

TempleWaterfOWLS, 2015-07-08:
I just put the erle-debian-flash-25-4-15.img.gz image on the BeagleBone Black. I'm looking to use APMrover2 so I made it and I got mavros to run. But when I go to $rostopic echo any of the mavros topics.

Here a list of warnings that I got before I killed the copter startup script:
[ERROR] [1436391593.335363009]: FCU: Initialising APM...
[ INFO] [1436391593.340575093]: CON: Got HEARTBEAT, connected.
[ WARN] [1436391604.347705136]: PR: request list timeout, retries left 2
[ WARN] [1436391604.418798427]: VER: request timeout, retries left 1
[ WARN] [1436391605.429735969]: PR: request list timeout, retries left 1
[ WARN] [1436391606.433376094]: PR: request list timeout, retries left 0
[ WARN] [1436391609.347714761]: WP: timeout, retries left 2
[ WARN] [1436391609.539454844]: VER: request timeout, retries left 0
[ WARN] [1436391610.360942761]: WP: timeout, retries left 1
[ WARN] [1436391611.367641220]: WP: timeout, retries left 0
[ERROR] [1436391612.434586345]: WP: timed out.
[ WARN] [1436391614.562060803]: VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities
[ WARN] [1436391623.347379429]: CON: Lost connection, HEARTBEAT timed out.

Here's the last output of mavros after I killed the copter startup script:
[ INFO] [1436391732.854899901]: Built-in mavlink dialect: ardupilotmega
[ INFO] [1436391732.857788109]: MAVROS started. MY ID [1, 240], TARGET ID [1, 1]

@vmayoral any ideas?

TempleWaterfOWLS, 2015-07-09:
Going to add @vooon to this as well

vooon, 2015-07-09:
I need:
1. `/diagnostics`
2. which launch script do you use, and parameters
3. full mavros log output

vooon, 2015-07-09:
Now only what i can say, that you have bad link.

TSC21, 2015-07-22:
@TempleWaterfOWLS any updates on this?

closed 2015-07-24 by vooon

CrazyPopLin, 2015-12-07:
Hi, I reinstalled **mavlink 11.25** and **mavros 0.16** from source in odroid xu4 which installed ros indigo and build successfully. I am using pixhawk with apm stack.
But when I launch the apm.launch, the **HEARTBEAT** connected first and time out. Like each time I reboot odroid board, it connect and time out. Restart pixhawk won't connect at all. Also I try to update the apm firmware to 3.3.2, that didn't change anything. 
Before I update the mavlink and mavros, everything works fine. 
Is that problem of installation of mavlink?

**PARAMETERS**
>
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 10.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/fcu_url: /dev/ttyUSB0:57600
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: fcu
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
 * /mavros/mocap/use_tf: False
 * /mavros/plugin_blacklist: ['actuator_contro...
 * /mavros/plugin_whitelist: []
 * /mavros/px4flow/frame_id: px4flow
 * /mavros/px4flow/ranger_fov: 0.0
 * /mavros/px4flow/ranger_max_range: 5.0
 * /mavros/px4flow/ranger_min_range: 0.3
 * /mavros/safety_area/p1/x: 1.0
 * /mavros/safety_area/p1/y: 1.0
 * /mavros/safety_area/p1/z: 1.0
 * /mavros/safety_area/p2/x: -1.0
 * /mavros/safety_area/p2/y: -1.0
 * /mavros/safety_area/p2/z: -1.0
 * /mavros/setpoint_accel/send_force: False
 * /mavros/setpoint_attitude/reverse_throttle: False
 * /mavros/setpoint_attitude/tf/child_frame_id: attitude
 * /mavros/setpoint_attitude/tf/frame_id: local_origin
 * /mavros/setpoint_attitude/tf/listen: False
 * /mavros/setpoint_attitude/tf/rate_limit: 10.0
 * /mavros/setpoint_position/tf/child_frame_id: setpoint
 * /mavros/setpoint_position/tf/frame_id: local_origin
 * /mavros/setpoint_position/tf/listen: False
 * /mavros/setpoint_position/tf/rate_limit: 50.0
 * /mavros/startup_px4_usb_quirk: False
 * /mavros/sys/disable_diag: False
 * /mavros/sys/min_voltage: 10.0
 * /mavros/target_component_id: 1
 * /mavros/target_system_id: 1
 * /mavros/tdr_radio/low_rssi: 40
 * /mavros/time/time_ref_source: fcu
 * /mavros/time/timesync_avg_alpha: 0.6
 * /mavros/vibration/frame_id: vibration
 * /mavros/vision_pose/tf/child_frame_id: vision
 * /mavros/vision_pose/tf/frame_id: local_origin
 * /mavros/vision_pose/tf/listen: False
 * /mavros/vision_pose/tf/rate_limit: 10.0
 * /mavros/vision_speed/listen_twist: False
 * /rosdistro: indigo
 * /rosversion: 1.11.16
[... 198 lines cut ...]
        value: -1.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 0.0
---
>

**/mavros/state**
>
header: 
  seq: 71
  stamp: 
    secs: 1449483000
    nsecs: 298424602
  frame_id: ''
connected: False
armed: False
guided: False
mode: ''
>

Thanks for your help.

vooon, 2015-12-07:
Try to increase connection timeout.

---
```
[ INFO] [1449482921.050486680]: CON: Got HEARTBEAT, connected.
...
[ WARN] [1449483000.296174701]: CON: Lost connection, HEARTBEAT timed out.
```

Strange: you got 79 second timeout. Default apm_config.yaml is 10.

---
Nope: something strange: we got 70 HEARTBEAT's (1 Hz), then APM stops talking, +10 sec -> disconnect.

CrazyPopLin, 2015-12-07:
Yes, the default is 1HZ, then I change to 10 Hz want to fix the problem.
What is **increase connection timeout** means?

vooon, 2015-12-07:
Timeout param from apm_config.yaml. But it is not hope. APM just shuts telemetry down - i do not see repeated *connected* messages.

You tried to use USB port? Just to ensure that problem is in TELEM2 operation.

CrazyPopLin, 2015-12-07:
yes, I am using USB FDTI cable connect pihxaw(TELEM2) to odroid, the same cable I used before which works fine.

vooon, 2015-12-07:
I mean pixhawk's side microUSB.

CrazyPopLin, 2015-12-07:
no

CrazyPopLin, 2015-12-07:
quick question, why `/mavros/local_position/pose ` didn't publish, do i need to setup something?

killerlayup, 2015-12-30:
@CrazyPopLin  I have the same issue, `/mavros/local_position/pose` didn't publish, have you solved this issue?