mavlink/mavros #336: Move visualization plugin to node

vooon, 2015-07-09:
As i see viz only wants local position, so why not move that code to node?
I think that was more usable because since extraction you not need to copy-paste local_position code
and may restart viz at any time.

Ping @mhkabir

mhkabir, 2015-07-10:
@vooon sounds good! There is a slight issue with the markers not displaying sometimes, which we need to find and fix as well.

vooon, 2015-07-10:
@mhkabir i'm done, now testing required.

mhkabir, 2015-07-10:
@vooon Awesome. Can you see the marker in RViz? Can you give it a try, because I don't have my copter around ATM.

TSC21, 2015-07-10:
@mhkabir why don't you try with ROS SITL?

vooon, 2015-10-21:
I checked RViz with some bag file.

For some time it works, but later stops, when node still produce new markers.
Also changing origin frame break viz.

So it works, but URDF+Mesh definitely better.

closed 2015-10-21 by vooon