mavlink/mavros #338: ROS SITL tests suggestions

TSC21, 2015-07-11:
I opened this issue so users/devs can give some suggestions of SITL tests to add to `test_mavros`. Right now there's one implemented, which tests offboard position control through position setpoints (issues a square shaped path). Will add some more options for the shape of the paths (circles, ellipses, ...) and also add option for velocity and accel setpoint tests.
But some more test to estimators, control maneuvers, data stream, etc, which results can be followed in real time on Gazebo, are also required. So leave your suggestions bellow so I an implement those.

NOTE: no auto-tests have been implemented yet, so all these test have to be launched by user first. Auto-test will be created later, which will work together with `gtest`.
Current architecture state can be follow in https://github.com/mavlink/mavros/pull/337.

@LorenzMeier ping.

LorenzMeier, 2015-07-11:
Sounds great! I need to install ROS on my new machine first before I can lend a hand here.

TSC21, 2015-07-12:
**UPDATE:** added support for a eight-shaped path for offboard position control tests.

TSC21, 2015-07-13:
@LorenzMeier do you have some suggestions on tests we can implement?

TSC21, 2015-07-13:
**UPDATE:** Finished `offboard_control` test-bench, only missing `acceleration` tests. Suggestions to improve the current tests or add some more are welcomed.

wangsen1312, 2015-07-13:
@TSC21  @LorenzMeier  Thank you for your contribute to MAVROS with SITL!!  I want  to know the difference between what you just published with https://github.com/PX4/Firmware/tree/master/launch
Because I use it before. If needed, maybe I can do something with the simulation part?

TSC21, 2015-07-13:
@wangsen1312 there's not much of a difference. The difference mainly is in the purpose: while those launches have some general purposes, besides some demo apps, `test_mavros` and this SITL test architecture has the purpose of testing the MAVROS<->PX4 ROS SITL bridge, implementing some default tests that, for now, can and have to be launched by the user so he can check the resulting behavior in the Gazebo simulation environment. In the other hand, there's also the purpose of implementing auto-tests, which will launch together with `rostest` environment so to validate changes on MAVROS or user custom apps/nodes that interact with MAVROS.

The thing I'm looking for is for suggestions of simulations/tests that users are looking for to be implemented in this SITL test architecture/API. So, for example, in the case of the project you are developing, what are the things you were looking to get tested in simulation before moving to the real deal? Like mocap/vision data stream testing, offboard testing (which I already implemented in #337), etc.

wangsen1312, 2015-07-13:
@TSC21  Thank you for your reply. For my part, maybe the mocap/vision data stream testing is the main purpose, if you can write a demo for this, it will be great. By the way, I have tested the square motion using my own structure, it works fine. maybe next time you can add it to master branch.

TSC21, 2015-07-13:
> For my part, maybe the mocap/vision data stream testing is the main purpose, if you can write a demo for this, it will be great.

Thanks for the input. I'll be adding some more sensor models so to simulate this.

> By the way, I have tested the square motion using my own structure, it works fine. maybe next time you can add it to master branch.

Currently this is a PR launched to MAVROS, waiting review so it can be merged.

wangsen1312, 2015-07-15:
@TSC21   I just download the SILT code you write these days. I think you'd better split them into pieces such as some simple node, if some want to change something themselves, it' s hard to change a big classes. I don't know what I said right or wrong, just a suggestion.

TSC21, 2015-07-15:
@wangsen1312 the tests I'm writing are... tests. So they are supposed to be used to test the behavior of MAVROS code / SITL API  / Firmware changes. As it is now, the user has to launch them manually but, in the future, they are supposed to be issued automatically in conjunction with `rostest`. This is not supposed to be sample code for users to use. If one wants sample nodes, they are already available on Firmware launch folder and in https://pixhawk.org/dev/ros/sitl.

andre-nguyen, 2015-07-18:
IIRC the point of having all the setpoints in the same offboard control message was to be able to give velocity and position targets at the same time. e.g. Go at that place with this speed. To my knowledge, no one has ever tested this, I don't even know if this is supported right in the px4 firmware/mavros, might be a useful test case instead of limiting ourselves to only one of attitude/velocity/position control at a time.

wangsen1312, 2015-07-19:
@TSC21  Can you add a multiple UAV SITL sample?  maybe you can just give a small sample to let me know how to subscribe and publish same topics for different UAVs?

vooon, 2015-07-20:
@wangsen1312 easiest way to run multiple mavros nodes is to place it in different `<group ns="uav1"/>`, but that require to adjust topic paths.

> the tests I'm writing are... tests.

@TSC21 good test is useful code sample. Also i think that you can remove unneeded classes and implement different tests in different nodes.
At least i don't understand why you want one `big sitl_test_node` with bunch of subcommands instead of different nodes, like `sitl_test_offboard_control`.

TSC21, 2015-07-20:
> @TSC21 good test is useful code sample.

Agree. Didn't say the contrary.

> Also i think that you can remove unneeded classes

One can't call it uneeded until they get propper usage. And I already said to you the purpose of them.

> At least i don't understand why you want one big `sitl_test_node` with bunch of subcommands

First thing: it's not a big node. It's pretty simple actually (https://github.com/mavlink/mavros/blob/master/test_mavros/sitl_test/sitl_test_node.cpp).
Second thing: For me it makes more sense to have a test API where I have a central node and which ever kind of test I want to issue, it calls uppon a specific header file which runs all the needed test functions. This works as a modular testbench. And I pretend to keep it this way.

TSC21, 2015-07-20:
Also @wangsen1312 as I said, as soon as I can, I can help you with that approach. But right now I'm allocating time to other stuff, so feel free to advance a bit on implementation if you manage to.

TSC21, 2015-07-20:
@andre-nguyen,
> To my knowledge, no one has ever tested this, I don't even know if this is supported right in the px4 firmware

From a quick look at the `mc_pos_control`, I think it only supports one mode at a time.

devbharat, 2015-07-22:
@TSC21 Here is how I integrated gazebo iris model into the posix SITL. I'll file in the PRs after cleaning up a few things in the corresponding repos soon. 

Briefly put, I used the 'mavros gcs_bridge' node to recieve HIL_control packets from the udp port, packed them up into mavlink ros topic with the help of mavros/utils.h. On the gazebo side, I appeneded the controller_interface plugin to recieve the ros mavlink packets, unpack them, convert the pwm values to RPMs and set that as the reference value for the motor_controller plugin. From there on the existing gazebo plugins in iris are sufficient to generate the required forces and moments.

For sending the sensor information back to posix sitl, I modified the gazebo imu plugin to also pack a mavlink hils_sensor packect with the imu data and push it to the ros mavlink node, which the mavros gcs_bridge promptly recieves, strips ros off mavlink, and sends it to the udp port posix sitl is waiting on, thus completing the loop. For the moment, a mag data model similar to that in jMAVSIM, and a simplified GPS data model (to work in and around Zurich) are also added to the imu plugin, but they deserve to be their own plugins (with something like geographiclib to handle projections from xyz to LLA). Also, a SENS_BOARD_ROT value of 8 (roll 180) is necessary for frame of reference conversion between gazebo model and SITL. You can set it in qgcs.

The standard posix SITL interface for qgcs and radio control works as is. There is no major change on the PX4 side, except for a new SITL rc file to handle new parameter values, and a new launch file to launch gazebo iris without all the other multiplatform ros nodes.

I am able to fly around the iris no problem (gazebo realtime factor is around 0.6 with the gui running, it seems to be much better running headless). The gains need a little more tuning.

gokoreas, 2016-01-20:
Is there a way to fly multiple UAVs(specificly iris models) in gazebo using SITL?

If not it would be nice to get suggestions on how to do so!

TSC21, 2017-04-04:
Closing this as it will be planned on "test_mavros"

closed 2017-04-04 by TSC21