mavlink/mavros #339: NSH terminal plugin

vooon, 2015-07-14:
APM now has feature: nsh terminal via mavlink.
Technically this implemented via serial control messages (serial device via mavlink).

Because it not implemented in PX4 that issue has very low priority, so don't expect that i implement that soon.

TODO:
- [ ] serial_control plugin, provide service to create a PTY and delete.
- [ ] mavserial create & delete - tool for PTY device manipulation
- [ ] mavserial nsh - simple terminal (does create + miniterm.py + delete on close)

vooon, 2015-08-06:
Planned for PX4: PX4/Firmware#2650

vooon, 2016-08-12:
Now it supported, but i'm not sure does it really need to be in mavros?

Perhaps [mavlink_shell.py][1] is good enough.

[1]: https://github.com/PX4/Firmware/blob/master/Tools/mavlink_shell.py

TSC21, 2017-04-04:
@vooon should we move on to this or just stick with [mavlink_shell.py](https://github.com/PX4/Firmware/blob/master/Tools/mavlink_shell.py)?

vooon, 2017-04-04:
It is interesting and maybe useful API, but really i don't see a reason to spend time on this.

Passing mavlink stream to `mavlink_shell.py` is good enough.

closed 2017-04-04 by vooon