mavlink/mavros #344: scaled rc values topic

v01d, 2015-07-18:
Hi,
Would it be possible to add support for scaled (-100% to 100%) RcIn values? I need to read the RC controls and with raw values is a bit difficult.

vooon, 2015-07-18:
Depends on FCU. Present FCU's provide only raw pulse widths.
PX4 may return normalized actuator outputs (but that not your question).

Or you may do normalization yourself. Just write small python node that will work as a glue.

v01d, 2015-07-18:
I would be using Pixhawk. This board does not send scaled RC inputs?
If not, how would I scale values? Would I need to perform a manual calibration to see which are the max/min values?

vooon, 2015-07-18:
FYI: Pixhawk can run APM or PX4.

I don't seen messages for scaled inputs, so yes. But calibration data may be requested from FCU.
Bunch of `RC%n_MIN`/`RC%n_MAX` and others. But you can't get this values at mavros startup, so you have wait.
Or copy values by your script into filtering node ns.

v01d, 2015-07-18:
I asked because of this: https://pixhawk.ethz.ch/mavlink/#RC_CHANNELS_SCALED

Sorry, I meant PX4 stack. So, they are not supported by PX4?

If I scale them myself, I can make a command which retrieves values and writes a YAML file to be loaded later by the "scaling" node, I guess.

vooon, 2015-07-18:
There only RC_CHANNELS: [mavlink_messages.cpp 1966](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_messages.cpp#L1966)

v01d, 2015-07-18:
Ok, thanks!

vooon, 2015-07-23:
Updates?

closed 2015-07-24 by vooon