mavlink/mavros #345: Release 0.13

vooon, 2015-07-21:
I think to release 0.13 at August first. But before we need complete with:

- [ ] TF cleaning, use TF2 only and eigen
- [ ] Tests for uas_sensor_orientation module
- [ ] Manual + video on how to use PX4 SITL tests written by @TSC21 

---
Changelog:

- #71 porting to TF2 done
- #325 gps merged in global_position
- #320 imu_pub cleanup
- #273 remove px4 bug workaround
- #341 uas_sensor_orientation reworked
- #337 **new package** test_mavros with sitl tests and manuals

TSC21, 2015-07-21:
I'm working on this @vooon. Will launch the necessary PR's as soon as I can.

vooon, 2015-07-29:
@TSC21 there 2 days before deadline.

TSC21, 2015-07-30:
> @TSC21 there 2 days before deadline.

Sorry @vooon had some other stuff to do till now. I'll be doing it this weekend.

vooon, 2015-07-31:
- https://github.com/ros/rosdistro/pull/9048
- https://github.com/ros/rosdistro/pull/9049

closed 2015-08-01 by vooon