mavlink/mavros #346: Error while launching and no Heartbeat.

rs1990, 2015-07-21:
Hello ,

I have been trying to get my x8 octocopter with the pixhawk flying for quite a while but i cannot seem to connect to it ever. 
Any help. I just cant seem to figure out anything . its quite confusing . I have connected my odroid U3 running ubuntu 14.04 and the TELEM2 port of the the pixhawk via the USB to serial converter   like so: 
[image]

Whenever i run "roslaunch mavros px4.launch"i have tried different baudrates as well. Could you tell me the correct baudrate for this type of connection?
 
i get the following error:

 
odroid@bat:~$ roslaunch mavros px4.launch 
... logging to /home/odroid/.ros/log/ef922eba-2fc4-11e5-a620-74da38065ef3/roslaunch-bat-1960.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://bat:49969/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: 8
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: 8
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: 8
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: 8
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_url: /dev/ttyUSB0:921600
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: fcu
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
 * /mavros/mocap/use_tf: False
 * /mavros/plugin_blacklist: ['safety_area', '...
 * /mavros/plugin_whitelist: []
 * /mavros/px4flow/frame_id: px4flow
 * /mavros/px4flow/ranger_fov: 0.0
 * /mavros/px4flow/ranger_max_range: 5.0
 * /mavros/px4flow/ranger_min_range: 0.3
 * /mavros/safety_area/p1/x: 1.0
 * /mavros/safety_area/p1/y: 1.0
 * /mavros/safety_area/p1/z: 1.0
 * /mavros/safety_area/p2/x: -1.0
 * /mavros/safety_area/p2/y: -1.0
 * /mavros/safety_area/p2/z: -1.0
 * /mavros/setpoint_accel/send_force: False
 * /mavros/setpoint_attitude/reverse_throttle: False
 * /mavros/setpoint_attitude/tf/child_frame_id: attitude
 * /mavros/setpoint_attitude/tf/frame_id: local_origin
 * /mavros/setpoint_attitude/tf/listen: False
 * /mavros/setpoint_attitude/tf/rate_limit: 10.0
 * /mavros/setpoint_position/tf/child_frame_id: setpoint
 * /mavros/setpoint_position/tf/frame_id: local_origin
 * /mavros/setpoint_position/tf/listen: False
 * /mavros/setpoint_position/tf/rate_limit: 50.0
 * /mavros/startup_px4_usb_quirk: True
 * /mavros/sys/disable_diag: False
 * /mavros/sys/min_voltage: 10.0
 * /mavros/target_component_id: 1
 * /mavros/target_system_id: 1
 * /mavros/tdr_radio/low_rssi: 40
 * /mavros/time/time_ref_source: fcu
 * /mavros/time/timesync_avg_alpha: 0.6
 * /mavros/vibration/frame_id: vibration
 * /mavros/vision_pose/tf/child_frame_id: vision
 * /mavros/vision_pose/tf/frame_id: local_origin
 * /mavros/vision_pose/tf/listen: False
 * /mavros/vision_pose/tf/rate_limit: 10.0
 * /mavros/vision_speed/listen_twist: False
 * /rosdistro: indigo
 * /rosversion: 1.11.13

NODES
  /
    mavros (mavros/mavros_node)

auto-starting new master
process[master]: started with pid [1971]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ef922eba-2fc4-11e5-a620-74da38065ef3
process[rosout-1]: started with pid [1984]
started core service [/rosout]
process[mavros-2]: started with pid [1988]
[ INFO] [1437495861.007422791]: FCU URL: /dev/ttyUSB0:921600
[ INFO] [1437495861.014099760]: serial0: device: /dev/ttyUSB0 @ 921600 bps
[ INFO] [1437495861.017697911]: GCS bridge disabled
[ INFO] [1437495861.550304048]: Plugin 3dr_radio loaded and initialized
[ INFO] [1437495861.564767524]: Plugin actuator_control loaded and initialized
[ INFO] [1437495861.592204523]: Plugin command loaded and initialized
[ INFO] [1437495861.592773603]: Plugin distance_sensor blacklisted
[ INFO] [1437495861.632774170]: Plugin ftp loaded and initialized
[ INFO] [1437495861.674008189]: Plugin global_position loaded and initialized
[ INFO] [1437495861.674707061]: Plugin image_pub blacklisted
[ INFO] [1437495861.709403193]: Plugin imu_pub loaded and initialized
[ INFO] [1437495861.727033362]: Plugin local_position loaded and initialized
[ INFO] [1437495862.631232919]: Plugin mocap_pose_estimate loaded and initialized
[ INFO] [1437495862.646063976]: Plugin param loaded and initialized
[ INFO] [1437495862.673441433]: Plugin px4flow loaded and initialized
[ INFO] [1437495862.691940098]: Plugin rc_io loaded and initialized
[ INFO] [1437495862.692543179]: Plugin safety_area blacklisted
[ INFO] [1437495862.712904169]: Plugin setpoint_accel loaded and initialized
[ INFO] [1437495862.768918787]: Plugin setpoint_attitude loaded and initialized
[ INFO] [1437495862.796322952]: Plugin setpoint_position loaded and initialized
[ INFO] [1437495862.811329175]: Plugin setpoint_velocity loaded and initialized
[ INFO] [1437495862.841003997]: Plugin sys_status loaded and initialized
[ INFO] [1437495862.856489051]: Plugin sys_time loaded and initialized
[ INFO] [1437495862.863996767]: Plugin vfr_hud loaded and initialized
[ INFO] [1437495862.864775471]: Plugin vibration blacklisted
[ INFO] [1437495862.894723542]: Plugin vision_pose_estimate loaded and initialized
[ INFO] [1437495862.912610002]: Plugin vision_speed_estimate loaded and initialized
[ INFO] [1437495862.939421545]: Plugin waypoint loaded and initialized
[ INFO] [1437495862.940119167]: Autostarting mavlink via USB on PX4
[ INFO] [1437495862.940525874]: Built-in mavlink dialect: ardupilotmega
[ INFO] [1437495862.940675373]: MAVROS started. MY ID [1, 240], TARGET ID [1, 1]

vooon, 2015-07-21:
What exactly firmware do you use? Try to connect via USB first (ttyACM0), then check /diagnostics.
If all ok, then perhaps you missed to configure `SYS_COMPANION` parameter on flight controller.

rs1990, 2015-07-21:
I updated to the latest px4 firmware using qgroundcontrol. I think that's
the right one isn't it??.

vooon, 2015-07-22:
Maybe better to use latest stable (it is default option in QGC).
Then perhaps you don't configured SYS_COMPANION.

vooon, 2015-07-24:
Updates?

rs1990, 2015-07-24:
Hello,

Sorry, I haven't been to my lab till now....will update asap.

closed 2015-09-10 by vooon