mavlink/mavros #349: Position not getting locked using vision pose in mavros

mmsarode, 2015-07-26:
*I am trying to use motion capture data by qualysis system for position estimate of the PX4 in hexacopter.
   1. Run mavros
   2. Push motion capture data to mavros/vision_pose/pose using a publisher (there is a message saying "VISION estimate valid" when I run the publisher.
    However, in the mavros terminal I keep getting the message after some time "GPS: no Fix message".
There is no solid green color (expected in case of position lock) on PX4 main LED. When I echo to vision_pose/pose and local_position/local in mavros, I get the correct motion capture data (x, y, z) which means that it should lock position. Am I missing something?

Here is the screenshot of mavros terminal messages-
[image]

* What should be the fame_ID for the vision_pose/pose topic? I wrote the following code which publishes motion capture data to pose topic. Is there anything wrong that I am doing?
CODE for publisher-

#include <ros/ros.h>
#include <std_msgs/String.h> 
#include <stdio.h>
#include "geometry_msgs/PoseStamped.h"
#include "geometry_msgs/Vector3Stamped.h"
#include </home/mahesh/catkin_ws/src/beginner_tutorials/src/Qualisys.h>

int main(int argc, char **argv)
{
   ros::init(argc, argv, "data_to_vision_3");
   ros::NodeHandle n;

   ros::Publisher chatter_pub = n.advertise<geometry_msgs::PoseStamped>("/mavros/vision_pose/pose",100);
   ros::Rate loop_rate(10);

   geometry_msgs::PoseStamped msg;
   int count = 1;
   	
   PositionReciever qp;
   Body some_object;
   qp.connect_to_server();
 
 	   while(ros::ok()){
	   some_object = qp.getStatus();
		// some_object.print();
		printf("%f\n",some_object.position_x);
       msg.header.stamp = ros::Time::now();
       msg.header.seq=count;
       msg.header.frame_id = 1;
       msg.pose.position.x = 0.001*some_object.position_x;
       msg.pose.position.y = 0.001*some_object.position_y;
       msg.pose.position.z = 0.001*some_object.position_z;
       msg.pose.orientation.x = 0.1;
       msg.pose.orientation.y = 0.2;
       msg.pose.orientation.z = 0.3;
       msg.pose.orientation.w = 0.4;

       chatter_pub.publish(msg);
       ros::spinOnce();
       count++;
       loop_rate.sleep();
   }
       
   return 0;
}

mhkabir, 2015-07-26:
@mmsarode The no GPS text is just a warning, nothing critical and doesn't affect your setup in any manner. 
This is a Firmware issue, if the system isn't flagging ```position_valid``` true when the vision estimate comes through (which it apparently does, since the estimator uses it.) We might've broken something recently in ```inav``` which causes this.
Might want to delve into Firmware to deal with the issue, or open an issue at PX4/Firmware and tag me please.

closed 2015-07-26 by mhkabir

mmsarode, 2015-07-26:
@mhkabir - I opened an issue at PX4/Firmware. Can you confirm what should I send as "Frame_id" while publishing to vision_pose/pose? Right now I am using " msg.header.frame_id = 1; "

TSC21, 2015-07-26:
`frame_id` doesn't make any difference, since only the stamp matters.