mavlink/mavros #353: Cannot Launch Node

Tomofoye, 2015-07-30:
Hey all I am running indigo and I installed mavros. I tried to access the launch file using

roslaunch mavros apm2.launch

but I always get the error

ERROR: cannot launch node of type [mavros/mavros_node]: can't locate node [mavros_node] in package [mavros]

Any ideas of what the problem is?

vooon, 2015-07-30:
Did you do `catkin build` (or `catkin_make`)? If done, did you sourced your devel setup?

```sh
source ~/ros_ws/devel/setup.bash
```

Tomofoye, 2015-07-30:
Yes I have sourced the devel/setup.bash and I used a catkin_make

vooon, 2015-07-30:
```
rospack list
ls ~/ros_ws/devel/lib/mavros
```

Tomofoye, 2015-07-30:
Here is the result from rospack list: 
actionlib /opt/ros/indigo/share/actionlib
actionlib_msgs /opt/ros/indigo/share/actionlib_msgs
angles /opt/ros/indigo/share/angles
bond /opt/ros/indigo/share/bond
bondcpp /opt/ros/indigo/share/bondcpp
bondpy /opt/ros/indigo/share/bondpy
catkin /opt/ros/indigo/share/catkin
class_loader /opt/ros/indigo/share/class_loader
cmake_modules /opt/ros/indigo/share/cmake_modules
cpp_common /opt/ros/indigo/share/cpp_common
cv_bridge /opt/ros/indigo/share/cv_bridge
diagnostic_msgs /opt/ros/indigo/share/diagnostic_msgs
diagnostic_updater /opt/ros/indigo/share/diagnostic_updater
dynamic_reconfigure /opt/ros/indigo/share/dynamic_reconfigure
gencpp /opt/ros/indigo/share/gencpp
genlisp /opt/ros/indigo/share/genlisp
genmsg /opt/ros/indigo/share/genmsg
genpy /opt/ros/indigo/share/genpy
geometry_msgs /opt/ros/indigo/share/geometry_msgs
image_transport /opt/ros/indigo/share/image_transport
libmavconn /opt/ros/indigo/share/libmavconn
mavlink /opt/ros/indigo/share/mavlink
mavros /opt/ros/indigo/share/mavros
mavros_extras /opt/ros/indigo/share/mavros_extras
message_filters /opt/ros/indigo/share/message_filters
message_generation /opt/ros/indigo/share/message_generation
message_runtime /opt/ros/indigo/share/message_runtime
mk /opt/ros/indigo/share/mk
nav_msgs /opt/ros/indigo/share/nav_msgs
nodelet /opt/ros/indigo/share/nodelet
nodelet_topic_tools /opt/ros/indigo/share/nodelet_topic_tools
pluginlib /opt/ros/indigo/share/pluginlib
rosbag /opt/ros/indigo/share/rosbag
rosbag_migration_rule /opt/ros/indigo/share/rosbag_migration_rule
rosbag_storage /opt/ros/indigo/share/rosbag_storage
rosbash /opt/ros/indigo/share/rosbash
rosboost_cfg /opt/ros/indigo/share/rosboost_cfg
rosbuild /opt/ros/indigo/share/rosbuild
rosclean /opt/ros/indigo/share/rosclean
rosconsole /opt/ros/indigo/share/rosconsole
rosconsole_bridge /opt/ros/indigo/share/rosconsole_bridge
roscopter /home/dukerama/catkin_ws/src/roscopter
roscpp /opt/ros/indigo/share/roscpp
roscpp_serialization /opt/ros/indigo/share/roscpp_serialization
roscpp_traits /opt/ros/indigo/share/roscpp_traits
roscreate /opt/ros/indigo/share/roscreate
rosgraph /opt/ros/indigo/share/rosgraph
rosgraph_msgs /opt/ros/indigo/share/rosgraph_msgs
roslang /opt/ros/indigo/share/roslang
roslaunch /opt/ros/indigo/share/roslaunch
roslib /opt/ros/indigo/share/roslib
roslisp /opt/ros/indigo/share/roslisp
roslz4 /opt/ros/indigo/share/roslz4
rosmake /opt/ros/indigo/share/rosmake
rosmaster /opt/ros/indigo/share/rosmaster
rosmsg /opt/ros/indigo/share/rosmsg
rosnode /opt/ros/indigo/share/rosnode
rosout /opt/ros/indigo/share/rosout
rospack /opt/ros/indigo/share/rospack
rosparam /opt/ros/indigo/share/rosparam
rospy /opt/ros/indigo/share/rospy
rosservice /opt/ros/indigo/share/rosservice
rostest /opt/ros/indigo/share/rostest
rostime /opt/ros/indigo/share/rostime
rostopic /opt/ros/indigo/share/rostopic
rosunit /opt/ros/indigo/share/rosunit
roswtf /opt/ros/indigo/share/roswtf
sensor_msgs /opt/ros/indigo/share/sensor_msgs
shape_msgs /opt/ros/indigo/share/shape_msgs
smclib /opt/ros/indigo/share/smclib
std_msgs /opt/ros/indigo/share/std_msgs
std_srvs /opt/ros/indigo/share/std_srvs
stereo_msgs /opt/ros/indigo/share/stereo_msgs
tf /opt/ros/indigo/share/tf
tf2 /opt/ros/indigo/share/tf2
tf2_msgs /opt/ros/indigo/share/tf2_msgs
tf2_py /opt/ros/indigo/share/tf2_py
tf2_ros /opt/ros/indigo/share/tf2_ros
topic_tools /opt/ros/indigo/share/topic_tools
trajectory_msgs /opt/ros/indigo/share/trajectory_msgs
visualization_msgs /opt/ros/indigo/share/visualization_msgs
xmlrpcpp /opt/ros/indigo/share/xmlrpcpp

And there is nothing in the devel/lib/mavros folder

vooon, 2015-07-30:
```
source /opt/ros/indigo/setup.bash
ls /opt/ros/indigo/lib/mavros
```

Tomofoye, 2015-07-30:
Thanks that worked. I didn't think I needed to do that because I already added that line to my .bashrc file. Do you know any reason why that wasn't working? And is there a way to make it so that I don't have to source /opt/ros/indigo/setup.bash every time i need to run mavros?

Tomofoye, 2015-07-30:
Never mind, it seems to be working well now, maybe I forgot to source /~.bashrc. Thanks again

vooon, 2015-07-30:
Part of my .zshenv (you may place that in .bashrc):
```sh
# ROS
if [[ -e "$HOME/ros/setup.zsh" ]]; then
	source $HOME/ros/setup.zsh
elif [[ -e "$HOME/ros/devel/setup.zsh" ]]; then
	source $HOME/ros/devel/setup.zsh
elif [[ -e "/opt/ros/jade/setup.zsh" ]]; then
	source /opt/ros/jade/setup.zsh
elif [[ -e "/opt/ros/indigo/setup.zsh" ]]; then
	source /opt/ros/indigo/setup.zsh
elif [[ -e "/opt/ros/hydro/setup.zsh" ]]; then
	source /opt/ros/hydro/setup.zsh
fi
```

First two checks: rosbuild place setup in ws, catkin in ws/devel.
Next is fallback. Also note that you should source distro setup before creation of ws.
Else remove `build`, `devel`, `install` folders, CMakeLists.txt from `src`. Source setup, then `catkin build`.

closed 2015-07-30 by vooon

davidrotor, 2016-04-13:
@vooon can you help me reinstall roscpp on my odroid I literally have no clue on how to do so.

vooon, 2016-04-13:
Try to upgrade using apt-get, then check shadow repo: http://packages.ros.org/ros-shadow-fixed/ubuntu/pool/main/r/

If that not helps you should follow ROS source installation instructions (and that will take a long time...).

Ankita-Singh-21, 2020-05-03:
I have Firmware(px4) and mavros located in same workspace , so i needed to build my ws using catkin_make_isolated. Si while running mavros i have to run :
source devel_isolated/setup.bash

manzurmurshid, 2020-08-05:
I had the same problem and keeping mavros in catkin_ws/src and to build I had to use the `catking_make_isolated` like @Ankita-Singh-21. This problem can be solved by `source ~/catkin_ws/devel_isolated/setup.bash` or inserting this line at in the `source ~/.bashrc`

zeobora, 2022-10-22:
> Thanks that worked!! I didn't think I needed to do that because I already added that line to my .bashrc file. Do you know any reason why that wasn't working? And is there a way to make it so that I don't have to source /opt/ros/indigo/setup.bash every time i need to run mavros?

Do you have any solutions for this. Do I have to do source every single time? Like you I sourced it in my .bashrc file but it doesn't work until I source it in my terminal