mavlink/mavros #356: IMU and setpoint/vision pose frames not aligned?

MarcGyongyosi, 2015-08-03:
Hi,

Looking at IMU output, I can see that the IMU coordinate system is:

x - forward
y - left
z - up

I was wondering why this does not match with the ENU frame used everywhere else in mavros (i.e. vision estimates, position setpoints). There we have:

x - right
y - forward 
z - up

The extra rotation is not a big deal but I think it would make things easier if these two were aligned...

Maybe I am wrong, but hopefully someone can explain why it's is this way anyways!

Thanks!

Marc

vooon, 2015-08-03:
There lot of talks about frame conversions. Look #49, [REP-103][1]. Also in 0.12 we redone conversions.

[1]: http://www.ros.org/reps/rep-0103.html#chirality

TSC21, 2015-08-03:
> I was wondering why this does not match with the ENU frame used everywhere else in mavros (i.e. vision estimates, position setpoints). There we have:

x - right
y - forward 
z - up

I believe this is not true. The conversion ENU<->NED means that we negate Z and exchange X and Y. ENU convention, as you can see in [REP-103][1] has X forward, Y left and Z up and that's the one we are using.

[1]: http://www.ros.org/reps/rep-0103.html#chirality

TSC21, 2015-08-03:
Also, you can checkout the rotations here: https://github.com/mavlink/mavros/blob/master/mavros/src/lib/uas_frame_conversions.cpp#L24-L28

MarcGyongyosi, 2015-08-03:
Ahh it's the same thing again with ENU where the vehicle is considered to be pointing east as forward ... Then it's obviously correct.

closed 2015-08-03 by MarcGyongyosi