mavlink/mavros #362: mavros/vision_pose/pose doesn't accept PoseWithCovarianceStamped anymore

MarcGyongyosi, 2015-08-05:
Hi,

I updated mavros today in the morning. Now I am getting an error because the vision plugin is expecting a pose stamped although I set pose_with_covariance to true.... 

[image]
[image]

What exactly do I need to change to make this work again?

Thanks,
Marc

MarcGyongyosi, 2015-08-05:
Just found this: https://github.com/mavlink/mavros/commit/f22ab3bfc9467965e9e9113a3ecc868e47d93fce why did you guys take out the covariance listener?

vooon, 2015-08-05:
Just to reduce support complexity. Anyway covariance not used.

mhkabir, 2015-08-05:
We will "upgrade" to PoseWithCovariance when we support the mavlink messages with covariance.

MarcGyongyosi, 2015-08-06:
That's two different answers... And tbh I don't see how removing one callback for a topic that's used in wikis removes support complexity. After all you will have people following the MSF tutorial and then end up running into the same issue. But anyways, I added the callback again in my mavros copy and now it works again.

closed 2015-08-06 by MarcGyongyosi