mavlink/mavros #370: Add support for requesting message interval

vooon, 2015-08-10:
Interface introduced in https://github.com/mavlink/mavlink/commit/25eccb1bfc5db3758d8336791641e06aa098cb3c

vooon, 2016-08-12:
Interface implemented in PX4. APM still uses per-stream messages.

Need to move message rate funcs to `sys_msgctl` plugin.

TSC21, 2017-04-04:
@vooon what's the state of this?

vooon, 2017-04-04:
Still not implemented on our side. But not so bad, because you still may configure PX4 via SD card.

rmackay9, 2018-11-29:
ArduPilot is just about to add support for the get/set-message-interval message.  It would certainly be nice to be able to use the mavros setrate command to be able to set a specific message's update rate.  I might try and do that myself although I certainly wouldn't complain if someone beat me to it.

vooon, 2018-11-29:
@rmackay9 i think best way to do it - just add `~set_message_interval` service to `sys_status` and don't change older set-rate (but mark it as deprecated).
Then add new subcommand to mavsys. Perhaps it would be great if new command can use message names or msgid.

But problem is that this info is hidden inside libmavconn.

vooon, 2018-11-29:
Oh, and another idea - add interface like param to set intervals on connection.