mavlink/mavros #371: Add support for HOME_POSITION

vooon, 2015-08-10:
Introduced: https://github.com/mavlink/mavlink/commit/0c16a3841e8865982800ef9e03e04f9f002db417

tstastny, 2016-12-12:
@vooon Was this ever put into a plugin? I did not find it. If not, I was planning to do this anyway and could make a pull request. Was thinking it could be its own plugin, as it's not really a waypoint, and not an aircraft position. Let me know if you disagree. Otherwise I'll start today.

vooon, 2016-12-13:
@tstastny write new plugin.

TSC21, 2017-04-04:
@tstastny have you implemented this already? If yes, issuing a PR wourld be awesome. Maybe integrate it on `global_position` plugin?

tstastny, 2017-04-05:
@TSC21 I actually did implement, but on my group's own fork of MAVROS. In December we were quite behind the current master. Now we have caught up. I could make a PR in the next couple of days. (Note I've implemented as a new plugin specifically for `home_position`, as I thought it best to separate a home point from vehicle specific positions in `global_position`).

TSC21, 2017-04-05:
Yeah makes sense as it also includes the local position. Will be waiting for your PR then. Thank you!

vooon, 2017-04-29:
Implemented in #695.

closed 2017-04-29 by vooon