mavlink/mavros #381: RcIn message is periodically empty

v01d, 2015-09-01:
The "channels" fields is most of the time populated with channel values, but about once or twice per second it is empty. I'm using turnigy 9x receiver with PPM encoder. On QGC it does not seem that RC is lost.

vooon, 2015-09-01:
Perhaps firmware send empty [RC_CHANNELS][1] message (chancount=0)

[1]: https://pixhawk.ethz.ch/mavlink/#RC_CHANNELS

v01d, 2015-09-01:
> But i got similar only if FCU lost some frames (when i connected SBUS RX
> to wrong port).

I was also wondering if it is a response to noise. Sounds reasonable.
Strange thing is that if I turn off transmitter the behavior does not
change (I read noise, of course).
But I guess that for another receiver which is able to inform of actual
link loss might not do this.

If you feel this is expected, then, feel free to close it.

vooon, 2015-09-01:
I'm not sure that is what expected, but that it works now. If rc subsystem lost link it changes channel count to 0.
Perhaps you have unreliable connections or high-amp wires near PPM.

v01d, 2015-09-01:
Ok, I will check it. Thanks

v01d, 2015-09-03:
This ended up being a hardware problem, not having sufficient current when just powering the autopilot + receiver using from USB.

closed 2015-09-03 by v01d