mavlink/mavros #383: Mavros crash on ARM

mhkabir, 2015-09-06:
Odroid C1 // 14.04 LTS // Mavros master

Mavros crashes right on startup.
```
/home/odroid/px4_ws/devel/lib/mavros/mavros_node: symbol lookup error: /home/odroid/px4_ws/devel/lib//libmavros_plugins.so: undefined symbol: _ZN3ros8DurationC1ERKNS_4RateE
================================================================================REQUIRED process [mavros-2] has died!
process has died [pid 2215, exit code 127, cmd /home/odroid/px4_ws/devel/lib/mavros/mavros_node __name:=mavros __log:=/home/odroid/.ros/log/6c86afa0-5490-11e5-80b2-e84e0611cb48/mavros-2.log].
log file: /home/odroid/.ros/log/6c86afa0-5490-11e5-80b2-e84e0611cb48/mavros-2*.log
Initiating shutdown!
================================================================================

```

vooon, 2015-09-06:
Does you tried to update indigo and recompile?

Also what version used? I think that issue in constructor for:
```cpp
ros::Duration(ros::Rate(whatever));
```

But recent versions of roscpp should have it.

mhkabir, 2015-09-06:
Hmm. Do you know how to update all the *existing* ```ros-indigo-*``` packages on my system to latest apt versions in one go?

vooon, 2015-09-06:
Nope, i usually do full-upgrade (aptitude).

vooon, 2015-09-06:
For full upgrade you may try:
```
sudo -i
apt-get update
apt-get upgrade
```

mhkabir, 2015-09-06:
That fixed it.
Thanks. Closing.

closed 2015-09-06 by mhkabir