mavlink/mavros #384: Is the velocity controller in this package and Pixhawk (stable version) same ?

weiweikong, 2015-09-08:
Hi, 
  I tested `iris_empty_world_offboard_ctl.launch` successfully with position control and velocity control under Gazebo in OFFBOARD mode. But it was failed when I try to test the **velocity control** under OFFBOARD mode with Pixhawk HIL setup. 

  My HIL setup is as follows
- jMAVsim
- Qgroundcontrol
- Pixhawk with current stable firmware

I modified the `base_node.launch` as follows
```
<launch>
	<!-- vim: set ft=xml noet : -->
	<!-- Base node launch file-->
    <arg name="args" value="offboard_control"/>    <!-- offboard ctl test -->
	<arg name="num_of_tests" default="1" />  <!-- number of repetitions of the same test -->
    <arg name="rate" default="10.0" />      <!-- loop rate of tests -->
    <arg name="use_pid" default="true" />     <!-- use default PID velocity control -->

	
	<arg name="mode" default="position" />     <!-- position ctl mode -->
    <arg name="shape" default="square" />     <!-- square shaped path -->

    <!-- if use_pid is set, PID values should be configured -->
    <arg name="linvel_p_gain" default="0.4" />     <!-- Proportional gain for linear velocity -->
    <arg name="linvel_i_gain" default="0.05" />     <!-- Integral gain for linear velocity -->
    <arg name="linvel_d_gain" default="0.12" />     <!-- Derivative gain for linear velocity -->
    <arg name="linvel_i_max" default="0.1" />     <!-- Max integral windup for linear velocity -->
    <arg name="linvel_i_min" default="-0.1" />     <!-- Min integral windup for linear velocity -->

    <arg name="yawrate_p_gain" default="0.011" />     <!-- Proportional gain for yaw rate -->
    <arg name="yawrate_i_gain" default="0.00058" />     <!-- Integral gain for yaw rate -->
    <arg name="yawrate_d_gain" default="0.12" />     <!-- Derivative gain for yaw rate -->
    <arg name="yawrate_i_max" default="0.005" />     <!-- Max integral windup for yaw rate -->
    <arg name="yawrate_i_min" default="-0.005" />     <!-- Min integral windup for yaw rate -->

	<node pkg="test_mavros" type="sitl_test_node" args="$(arg args)" name="sitl_test_offboard_ctl" required="true" clear_params="true" output="screen">
		<param name="num_of_tests" value="$(arg num_of_tests)" />
		<param name="rate" value="$(arg rate)" />
		<param name="use_pid" value="$(arg use_pid)" />

		<param name="mode" value="$(arg mode)" />
		<param name="shape" value="$(arg shape)" />

		<param name="linvel_p_gain" value="$(arg linvel_p_gain)" />
		<param name="linvel_i_gain" value="$(arg linvel_i_gain)" />
		<param name="linvel_d_gain" value="$(arg linvel_d_gain)" />
		<param name="linvel_i_max" value="$(arg linvel_i_max)" />
		<param name="linvel_i_min" value="$(arg linvel_i_min)" />

		<param name="yawrate_p_gain" value="$(arg yawrate_p_gain)" />
		<param name="yawrate_i_gain" value="$(arg yawrate_i_gain)" />
		<param name="yawrate_d_gain" value="$(arg yawrate_d_gain)" />
		<param name="yawrate_i_max" value="$(arg yawrate_i_max)" />
		<param name="yawrate_i_min" value="$(arg yawrate_i_min)" />
	</node>

</launch>
```
then I run
- `roslaunch base_node.launch mode:=position shape:=circle`, the quadrotor worked well.
- `roslaunch base_node.launch mode:=velocity shape:=circle`, the quadrotor flied randomly near the point before I switched the mode to OFFBOARD. But, the system send the OFFBOARD data steam correctly by checking `rostopic echo /mavros/setpoint_velocity/cmd_vel`.

So is the velocity controller in this package different from the one in Pixhawk?

vooon, 2015-09-08:
What node generate setpoints in iris_empty_world_offboard_ctl.launch?
Could you run it with HIL?

Mavros plugin does nothing than translating ros message to mavlink and stream it.
But as i know there some differences in PX4 between ROS SITL and HIL - not all modules are ported to multiplatform.
POSIX SITL should be nearest to HIL, but i'm still don't run it.

weiweikong, 2015-09-08:
- The setpoints could be generate through the node `sitl_test_node` due to the `iris_empty_world_offboard_ctl.launch` and `base_node.launch`.
- I could run HIL with jMAVsim and switch to OFFBOARD position control, but failed to OFFBOARD velocity control
- I am now checking the controller of  ROS SITL and HIL.
- I'd like to test POSIX SITL later and report the results.

vooon, 2015-09-08:
Hmm, i don't see entry for `mavros_node` in posted launch.
You should include `node.launch` from mavros package (or may be better `px4.launch`).

vooon, 2015-09-08:
Here that inclusion in iris launch: https://github.com/mavlink/mavros/blob/master/test_mavros/launch/px4/iris_empty_world_offboard_ctl.launch#L25-L39

weiweikong, 2015-09-08:
For HIL, I set up jMAVsim with `px4.launch`, which could access OFFBOARD mode and run position control well . This should be the `mavros_node` you mentioned ?

The reason that I use `base_node.launch` is try to publish the OFFBOARD velocity control data stream, which works fine with PX4 ROS SITL, but it was failed. So I guess, the difference of these two controllers was the reason.

vooon, 2015-09-15:
Updates?

weiweikong, 2015-09-16:
I still checking the details of the codes from the two packages because that I am not so familiar with the code details.

In the master branch of the PX4 firmware, it seems doing the things in
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L712-L714, which is
```
/* set position setpoint move rate */
_sp_move_rate(0) = _pos_sp_triplet.current.vx;
_sp_move_rate(1) = _pos_sp_triplet.current.vy;
...
_pos_sp += _sp_move_rate * dt;
```
In this `control_offboard` function, the system try to set the next position point based on the OFFBOARD expecting velocity using `_pos_sp`. I think the should be right, which convert the velocity command to position command.

But in 
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1284-L1343
which is
```
...
/* calculate euler angles, for logging only, must not be used for control */
_att_sp.roll_body = euler(0);
_att_sp.pitch_body = euler(1);
...
```
In this section, I am not totally understand the calculation of the roll/pitch setpoints. Also, as the comments mentioned, these two lines are only for logging. So I am still confused.

I will continue checking the codes and try to find the differences.

mhkabir, 2015-09-16:
The controller implementations are totally different, as far as I know.

weiweikong, 2015-09-16:
@mhkabir do you know some other version firmware of Pixhawk could implement velocity control ? Or based on the recent firmware, we could only set expecting roll/pitch/yaw angle to achieve velocity control ?

weiweikong, 2015-09-25:
Hi @vooon and @mhkabir , here are some updates.

# The Controller from PX4 ?

I check the `iris_empty_world_offboard_ctl.launch` file and draw the diagram as below.
[image]

The **Control Nodes** are loaded from PX4 and the two important part is `mc_pos_control` and `mc_att_control`. 
- I think this is the SAME position controller and attitude controller runing in PX4 firmware. 
- Also, it is the SAME as running `make sitl_quad` in [PX4 Standard SITL Setup]( https://pixhawk.org/dev/simulation/native_sitl#px4_standard_sitl_setup), according to the `rcS` file,
https://github.com/PX4/Firmware/blob/master/posix-configs/SITL/init/rcS#L45-#L46, which mentioned
```
...
mc_pos_control start
mc_att_control start
...
```
Are these correct ?

# The Velocity Setpoints
So I continued to test the OFFBOARD velocity control.
I running a keyboard node which published `cmd_vel` and the whole nodes connections are shown as below.
[image]

I tried to set the `/mavros/setpoint_velocity/cmd_vel` in one direction, such as `vx=1.0`, to test the controller.
```
header: 
  seq: 24925
  stamp: 
    secs: 6947
    nsecs: 50000000
  frame_id: velocity_cmd
twist: 
  linear: 
    x: 1.0
    y: 0.0
    z: 0.0
  angular: 
    x: 0.0
    y: 0.0
    z: 0.0
```
- In `iris_empty_world_offboard_ctl.launch` based environemnts
The quadrotor was running correctly while the x-axis velocity was publishing.

- In `make sitl_quad` and `jMAVSim HIL` based environments
The quadrotor was running for a while, HOWEVER, it stopped even though the velocity was still publishing.

Some suggestions ?

closed 2015-10-20 by vooon