mavlink/mavros #385: rosbag recorder triggerd by arming

vooon, 2015-09-08:
I think it might be useful to start log recording on FCU arming.

mhkabir, 2015-09-08:
I like it :dancer:
Ability to call custom rosbag record script would be nice as well. Till now, I had been manually doing this, but automation would be a plus.

vooon, 2015-09-08:
I'm actually think to make custom `record` script. As i know it is python-based, so i can base on it.
Alternatively we may to run program, in that case we may automate "do at arming event" anything.

meuchel, 2015-09-08:
Because rosbag is anyway a standalone program I like more the generic one to be able to run any program.
If already on track with a module for event triggered actions you should also think about other events which could trigger something. E.g. disarm or others.

vooon, 2015-09-08:
I checked rosbag and found that `play` and `record` is native binaries.
So generic only. Next thing is on what stack to write: python or c++.

Proposed config:
```yaml
some_node: # only used in log
  event: armed # not sure that it needed, for now i only want armning
  action: run, stop # required, start on run, stop on disarm
  name: rosbag # optional, define node name
  restart: true # optional, restart? not sure that it'll be implemented soon
  output: log # optional, same as <node output="" />
  rosrun: rosbag record -a # like rosrun
some_script_1:
  event: armed
  shell: turn_off_wifi.sh # any executable, 
some_script_2:
  event: disarmed
  shell: turn_on_wifi.sh
```

meuchel, 2015-09-08:
Looks good.
If there is rosrun there should be also roslaunch possible?

What about during same session if armed then disarmed and  armed again?
Then e.g. script will be started again and user must take care of double start? (afaik rosrun handles this itself)
In this cases event: disarmed would make sense.

vooon, 2015-09-08:
Perhaps roslaunch is too complex.

Yes, rosrun kills previous node with same name. That don't work for anonymous nodes.

Also first must be more verbose:
```yaml
event: armed, disarmed
action: run, stop
```

vooon, 2015-09-08:
http://wiki.ros.org/roslaunch/API%20Usage - helps a lot. Last thing - good name for that node.

eventlauncher?

meuchel, 2015-09-09:
of course also integration of roslaunch would be nice. 
Although all could be handled already only with the script option (including rosrun).
Regarding the name here some ideas:
event_handler_node
event_launcher_node
mavros_events_node

vooon, 2015-09-10:
Shell-launcher complete. Unfortunate stdout/stderr logging do not work (break execution).

Example config: [event_launcher.yaml](https://github.com/mavlink/mavros/blob/c368f38417c9b9156cb457a2124a4f2a21eb8555/mavros/launch/event_launcher.yaml)

As you can see there was ability to define custom trigger for event.
```yaml
event_name: # used in `event`
  service: service_path # std_srvs/Trigger
```

Next thing is rosrun.

vooon, 2015-09-10:
Hmm, implementing rosrun has problems.

I think that we may use shell nodes for all our needs.

vooon, 2015-09-10:
Closed by #386.

Shell entries just works fine.

closed 2015-09-10 by vooon