mavlink/mavros #389: Trying to run px4.launch on beaglebone black results in FCU permission denied error

anusna, 2015-09-19:
Hi All,

I am trying to setup mavros on Beaglebone Black which runs ubuntu 14.03 with ROS Indigo. I have installed the mavros package and connected to the pixhawk via FTDI to Telemetry 2 port in the pixhawk. The connection in the beaglebone black is /dev/ttyS2:921600. I have modified that in the px4.launch FCU url at the beginning.
However, when I am trying to do a roslaunch mavros px4.launch, after roscore starts I get a fatal error stating permission denied on the Serial port FCU. I have tried giving full permission by doing

sudo chmod 666 px4.launch

but that did not help. So I tried 

chmod +x px4.launch 

which turned the launch file into green but gives the same error.
I am totally at my wit's end and if anyone can help that will be really great.

PS: I have checked the FTDI - Pixhawk connection and that works fine. But in Mission Planner I could not find the SYS_COMPANION parameter to set it to 921600. Instead I have set the serial2 to 921600. I do not know if that is an error but any help will be greatly appreciated.

Thanks!

mhkabir, 2015-09-19:
Two things :
* Set your serial port permissions to 666, or alternatively add yourself to the ```dialout``` group.
```sudo usermod -a -G diaout <user_name>```
* You seem to be using APM. The correct launch file would be ```apm.launch```

vooon, 2015-09-19:
@anusna  launch is not an shell/has no shebang, so you should pass it to roslaunch. Executable flag does nothing.

Also you may look at [rarog_bringup][1] package and [rarog.conf][2] upstart job as example how to make OBC autostart ros.

[1]: https://github.com/vooon/rarog/tree/master/rarog_bringup
[2]: https://github.com/vooon/rarog/blob/master/rarog_configs/rpi2/rarog.conf

anusna, 2015-09-23:
Thanks all, adding to the dialout group and then logging out and logging in again!

closed 2015-09-23 by vooon

bigbellmercy, 2020-09-25:
As mentioned above,
`sudo usermod -a -G dialout $USER`

After rebooting, it works.