mavlink/mavros #397: Increasing stream rate APM

tuloski, 2015-10-09:
I'm a bit confused about this (I tried to search a lot before writing this):
how to increase the stream rate of a particular MAVlink message with APM? I know I can use mavsys rate --all frequency, but there are only groups of information. And I understood that a limit of 25Hz is imposed.
How can I set HIGRES_IMU at 100Hz for example? And is it possible with APM code? I know it is possible for PX4 but not sure about APM.
Thanks

vooon, 2015-10-09:
You can't get HIGRES_IMU from APM simply because it don't support that message.
And with APM you only may change rate by groups. Also "rate" units not documented, so it might be period in milliseconds, or other thing.

tuloski, 2015-10-09:
Is there any documentation about the supported messages?
Is there only raw IMU? And which is maximum frequency?
So the command is mavsys rate --raw_sensors (what here??)
And which message will be sent? This part of APM is very bad documented...

vooon, 2015-10-09:
I don't know does streams documented somewhere. I learned from source: [GCS_MAVLink.cpp](https://github.com/diydrones/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L756)

For imu you need RAW_IMU and ATTITUDE, so:
```
rosrun mavros mavsys rate --raw-sensors 10 --extra1 10
```

tuloski, 2015-10-11:
Ok thanks...but it seems that maximum frequency is 10Hz.
I think it is time to go back to my own autopilot based on PX4 :)

closed 2015-10-20 by vooon