mavlink/mavros #400: how to hold the MAV altitude by mavros

suimo, 2015-10-14:
now I use odroid xu3 to control pixhawk by mavros,I can arm the MAV and make it takeoff by giving the setpoint of throttle.but I can't hold the altitude ,when I change the flight mode from offboard to ALTCTL ,the failsafe mode is on,and the flight mode change to auto-land not altctl mode .I don't know why.
please who can help me with this issue? thanks a lot!

vooon, 2015-10-14:
What FCU do you use? What version of firmware? How OBC connected to FCU?

suimo, 2015-10-14:
@vooon hi,I use pixhawk with px4 firmware. and I use odroid to control pixhawk via the TELEM2 port,could you give me some help about this?

vooon, 2015-10-20:
What you trying to do?

If you want to use OFFBOARD for whole flight control, then simply send position setpoint with same altitude.
If you only needs automatic takeoff/land use `~cmd/takeoff` services (see documentation).

suimo, 2015-10-27:
Dear,vooon!  Thank you so much for your reply,I just want to use offboard to send position setpoint,and make the MAV takeoff and hold the position  automaticly, now I just want to use px4flow to implement this goal, but there are some issues during  parameter tuning ,could you give me some help?

mhkabir, 2015-10-27:
@suimo Please refer to the PX4 wiki/mailing list for more help. See README.

closed 2015-10-27 by mhkabir

chenshiyuhit, 2015-11-06:
@suimo  Hi,I also use Odroid XU3 to connect to the piahawk, but I can not set the mode to OFFBOARD mode, I don't know why?
My pixhawk firmware may be not the latest, because this?

suimo, 2015-12-18:
@chenshiyuhit hi，I think you can refer to this page  http://pixhawk.org/dev/ros/mavros_offboard