mavlink/mavros #403: Weird /tf timestamp/translation values

atech1380, 2015-10-19:
When I echo the /tf topic published by mavros running on my drone I get:

    - 
    header: 
    seq: 0
    stamp: 
    secs: 351641
    nsecs: 788089172
    frame_id: local_origin
    child_frame_id: fcu
    transform: 
    translation: 
    x: 0.0
    y: 166021.443179
    z: -1.0
    rotation: 
    x: 0.00538760551592
    y: 0.00619496139752
    z: 0.892766652461
    w: -0.450444558647
    ---

I'm not sure what to make of the massive 166km y translation reading when the drone is sitting still (may be an issue with my system), but I think the 350k timestamp is an issue with mavros.

It seems that the /tf timestamp is being set to whenever mavros was installed on the machine it is running. When I update the mavros installation from 0.11.0 to 0.13.0 (My drone comes with an older version of mavros preinstalled) the timestamp gets reset to ~1.5k:

    - 
    header: 
    seq: 0
    stamp: 
    secs: 1572
    nsecs: 192321700
    frame_id: local_origin
    child_frame_id: fcu_utm
    transform: 
    translation: 
    x: 166021.443179
    y: 0.0
    z: 0.84
    rotation: 
    x: -0.00606376771624
    y: -0.00740089278356
    z: -0.890106190627
    w: 0.455652748169
    ---

It is my understanding that the /tf timestamps are supposed to be related to the unix epoch time of the system [(Link)](http://docs.ros.org/jade/api/std_msgs/html/msg/Header.html), but the /tf time of ~350k seconds is not even close to the epoch time on my drone:

    :/# date
    Mon Oct 19 16:42:59 UTC 2015
    :/# date +"%s"
    1445272989

As an example of /tf messages usually look like for me, here is what it looks like if you echo the /tf topic when running the ardrone_autonomy ros driver for the Parrot AR.Drone 2:

    :~$ rostopic echo /tf
    ---
    transforms: 
    - 
    header: 
    seq: 0
    stamp: 
    secs: 1445033096
    nsecs: 388464635
    frame_id: ardrone_base_link
    child_frame_id: ardrone_base_bottomcam
    transform: 
    translation: 
    x: 0.0
    y: -0.02
    z: 0.0
    rotation: 
    x: 0.707106189653
    y: 0.707107372717
    z: -1.18306495402e-06
    w: -1.18306297463e-06
    ---
    transforms: 
    - 
    header: 
    seq: 0
    stamp: 
    secs: 1445033096
    nsecs: 388464675
    frame_id: odom
    child_frame_id: ardrone_base_link
    transform: 
    translation: 
    x: 0.0
    y: 0.0
    z: 0.0
    rotation: 
    x: -0.000229246524165
      y: -0.0186881185008
            z: 0.978807714478
            w: 0.203926849022
    ---

vooon, 2015-10-19:
Please update to 0.15 first. There bug in timesync on APM.

atech1380, 2015-10-20:
I'm now trying to update to 0.15 by building mavros from source, but I'm getting this error when trying to build:

    (ErleRobotics)root@localhost:~/catkin_ws# catkin_make --pkg mavros
    Base path: /root/catkin_ws
    Source space: /root/catkin_ws/src
    Build space: /root/catkin_ws/build
    Devel space: /root/catkin_ws/devel
    Install space: /root/catkin_ws/install
    ####
    #### Running command: "make cmake_check_build_system" in "/root/catkin_ws/build"
    ####
    ####
    #### Running command: "make -j1 -l1" in "/root/catkin_ws/build/mavros/mavros"
    ####
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_FileRead
    [  0%] Built target std_msgs_generate_messages_cpp
    [  0%] Built target geometry_msgs_generate_messages_cpp
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_FileClose
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_ParamPull
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_WaypointList
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_FileTruncate
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_BatteryStatus
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_FileRemoveDir
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_VFR_HUD
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_ParamPush
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_ParamSet
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_CommandTriggerControl
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_WaypointPull
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_FileWrite
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_FileMakeDir
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_RadioStatus
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_FileRename
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_CamIMUStamp
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_FileEntry
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_FileOpen
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_HomePosition
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_ActuatorControl
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_SetMode
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_WaypointGOTO
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_State
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_FileList
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_RCIn
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_FileChecksum
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_CommandInt
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_OpticalFlowRad
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_WaypointSetCurrent
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_WaypointPush
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_CommandTOL
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_FileRemove
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_ParamGet
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_CommandBool
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_RCOut
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_CommandLong
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_CommandHome
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_StreamRate
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_OverrideRCIn
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_Waypoint
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_Vibration
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_ExtendedState
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_Mavlink
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_ManualControl
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_WaypointClear
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_ParamValue
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_CommandCode
    [ 56%] Built target mavros_msgs_generate_messages_cpp
    [ 63%] Built target mavconn
    [ 65%] Building CXX object mavros/mavros/CMakeFiles/mavros.dir/src/lib/mavros.cpp.o
    c++: internal compiler error: Killed (program cc1plus)
    Please submit a full bug report,
    with preprocessed source if appropriate.
    See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
    make[2]: *** [mavros/mavros/CMakeFiles/mavros.dir/src/lib/mavros.cpp.o] Error 4
    make[1]: *** [mavros/mavros/CMakeFiles/mavros.dir/all] Error 2
    make: *** [all] Error 2
    Invoking "make -j1 -l1" failed
    (ErleRobotics)root@localhost:~/catkin_ws#

vooon, 2015-10-20:
Try to clean workspace first. Also i recommend to install `catkin-tools` package and use `catkin build`.

Version 0.15 also require fresh mavlink package. With catkin tools you can simply clone release repo to workspace.
```
git clone https://github.com/mavlink/mavlink-gbp-release.git mavlink -b release/jade/mavlink
```

meuchel, 2015-10-20:
for me it looks a little bit like running out of memory. Did you check your memory consumption during building, e. g. with top?

atech1380, 2015-10-20:
Yeah I checked the memory issues and it is running low on memory when it tries to compile. I'm looking into making a swap space to see if that solves the issue, then I'll try cleaning the workspace as vooon suggested.

vooon, 2015-10-20:
Try `-j1` flag. Or setup distcc.

atech1380, 2015-10-20:
So I added the swap space and that seemed to fix the issue because the compiler got further in the process, but now I'm getting a new error:

    (ErleRobotics)root@localhost:~/catkin_ws# catkin_make --pkg mavros
    Base path: /root/catkin_ws
    Source space: /root/catkin_ws/src
    Build space: /root/catkin_ws/build
    Devel space: /root/catkin_ws/devel
    Install space: /root/catkin_ws/install
    ####
    #### Running command: "cmake /root/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/root/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/root/catkin_ws/install -G Unix Makefiles" in "/root/catkin_ws/build"
    ####
    -- Using CATKIN_DEVEL_PREFIX: /root/catkin_ws/devel
    -- Using CMAKE_PREFIX_PATH: /root/catkin_ws/devel;/opt/ros/indigo
    -- This workspace overlays: /root/catkin_ws/devel;/opt/ros/indigo
    -- Using PYTHON_EXECUTABLE: /usr/bin/python
    -- Using Debian Python package layout
    -- Using empy: /usr/bin/empy
    -- Using CATKIN_ENABLE_TESTING: ON
    -- Call enable_testing()
    -- Using CATKIN_TEST_RESULTS_DIR: /root/catkin_ws/build/test_results
    -- Found gtest sources under '/usr/src/gtest': gtests will be built
    -- Using Python nosetests: /usr/bin/nosetests-2.7
    -- catkin 0.6.11
    -- BUILD_SHARED_LIBS is on
    -- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    -- ~~  traversing 5 packages in topological order:
    -- ~~  - libmavconn
    -- ~~  - mavros_msgs
    -- ~~  - mavros
    -- ~~  - mavros_extras
    -- ~~  - test_mavros
    -- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    -- +++ processing catkin package: 'libmavconn'
    -- ==> add_subdirectory(mavros/libmavconn)
    -- Boost version: 1.54.0
    -- Found the following Boost libraries:
    --   system
    -- Selected MAVLink dialect: ardupilotmega
    -- +++ processing catkin package: 'mavros_msgs'
    -- ==> add_subdirectory(mavros/mavros_msgs)
    -- Using these message generators: gencpp;genlisp;genpy
    -- mavros_msgs: 20 messages, 28 services
    -- +++ processing catkin package: 'mavros'
    -- ==> add_subdirectory(mavros/mavros)
    -- Using these message generators: gencpp;genlisp;genpy
    -- Boost version: 1.54.0
    -- Found the following Boost libraries:
    --   system
    -- Selected MAVLink dialect: ardupilotmega
    -- +++ processing catkin package: 'mavros_extras'
    -- ==> add_subdirectory(mavros/mavros_extras)
    -- Using these message generators: gencpp;genlisp;genpy
    -- Boost version: 1.54.0
    -- Found the following Boost libraries:
    --   system
    -- Selected MAVLink dialect: ardupilotmega
    -- +++ processing catkin package: 'test_mavros'
    -- ==> add_subdirectory(mavros/test_mavros)
    -- Using these message generators: gencpp;genlisp;genpy
    -- Boost version: 1.54.0
    -- Found the following Boost libraries:
    --   system
    -- Configuring done
    -- Generating done
    -- Build files have been written to: /root/catkin_ws/build
    ####
    #### Running command: "make -j1 -l1" in "/root/catkin_ws/build/mavros/mavros"
    ####
    [  0%] Built target _mavros_msgs_generate_messages_check_deps_FileRead
    [  0%] Built target geometry_msgs_generate_messages_cpp
[... 77 lines cut ...]
    /root/catkin_ws/src/mavros/mavros/src/plugins/sys_status.cpp: In member function ‘void mavplugin::SystemStatusPlugin::handle_extended_sys_state(const mavlink_message_t*, uint8_t, uint8_t)’:
    /root/catkin_ws/src/mavros/mavros/src/plugins/sys_status.cpp:622:3: error: ‘mavlink_extended_sys_state_t’ was not declared in this scope
       mavlink_extended_sys_state_t state;
       ^
    /root/catkin_ws/src/mavros/mavros/src/plugins/sys_status.cpp:622:32: error: expected ‘;’ before ‘state’
       mavlink_extended_sys_state_t state;
                                    ^
    /root/catkin_ws/src/mavros/mavros/src/plugins/sys_status.cpp:623:47: error: ‘state’ was not declared in this scope
       mavlink_msg_extended_sys_state_decode(msg, &state);
                                                   ^
    /root/catkin_ws/src/mavros/mavros/src/plugins/sys_status.cpp:623:47: note: suggested alternative:
    In file included from /usr/include/boost/thread.hpp:24:0,
                     from /opt/ros/indigo/include/diagnostic_updater/diagnostic_updater.h:48,
                     from /root/catkin_ws/src/mavros/mavros/include/mavros/mavros_plugin.h:21,
                     from /root/catkin_ws/src/mavros/mavros/src/plugins/sys_status.cpp:17:
    /usr/include/boost/thread/future.hpp:168:14: note:   ‘boost::future_state::state’
             enum state { uninitialized, waiting, ready, moved, deferred };
                  ^
    /root/catkin_ws/src/mavros/mavros/src/plugins/sys_status.cpp:623:52: error: ‘mavlink_msg_extended_sys_state_decode’ was not declared in this scope
       mavlink_msg_extended_sys_state_decode(msg, &state);
                                                        ^
    make[2]: *** [mavros/mavros/CMakeFiles/mavros_plugins.dir/src/plugins/sys_status.cpp.o] Error 1
    make[1]: *** [mavros/mavros/CMakeFiles/mavros_plugins.dir/all] Error 2
    make: *** [all] Error 2
    Invoking "make -j1 -l1" failed

vooon, 2015-10-21:
I already mentioned that you need fresh mavlink. Also you must rebuild all mavros packages, not only `mavros`.

atech1380, 2015-10-22:
Sorry, I must have missed the mavlink suggestion further up. I have since installed the latest mavlink from source. Updating mavlink and adding the swap space seemed to do the trick because all the mavros packages were able to build after the changes. The timestamp issue seems to be fixed because now I'm now getting correct timestamps:

    transforms: 
      - 
        header: 
          seq: 0
          stamp: 
            secs: 1445547498
            nsecs: 650907270
          frame_id: local_origin
        child_frame_id: fcu_utm
        transform: 
          translation: 
            x: 166021.443179
            y: 0.0
            z: 2.15
          rotation: 
            x: -0.00414518306689
            y: -0.00609408418961
            z: -0.895286144996
            w: 0.445430576154
    ---

Which looks like the correct timestamp. The weird translation is still present, but I'll look into that some more and maybe open a separate issue if I can't figure it out.

One more thing before this issue is totally resolved though...

When I run mavros I'm getting

    [ WARN] [1445547722.293151353]: TM: Wrong FCU time.

Is this something I should be concerned about?

vooon, 2015-10-22:
Nope. For first time that is ok.

Weird translation - check what FCU sends in LOCAL_POSITION_NED.

atech1380, 2015-10-22:
It reapeats the "Wrong FCU Time" warning every 10 seconds or so.

vooon, 2015-10-22:
Did you connected GPS module? Did it got fix?

atech1380, 2015-10-22:
I'm not using a GPS module. The application is meant for indoors.

vooon, 2015-10-22:
So nothing strange in wrong time and strange position data.

You can disable timesync and SYSTEM_TIME message. But you must feed FCU's position estimator with some data.
It might be optical flow position sensor (px4flow) or some ROS estimators (visual_position).

atech1380, 2015-10-26:
Well the timestamp issue is figured out, which was the biggest problem. I'll see what I can do about the translation stuff, but for now it isn't causing any headaches. Thanks for the help @vooon and @meuchel.

closed 2015-10-26 by vooon