mavlink/mavros #404: HELP! Unable to run setpoint_position_demo.py on native PX4 Flight stack with Offboard control

anusna, 2015-10-21:
Hi,
Here is my setup:
1. IRIS+ with PX4 flight stack flashed on the pixhawk onboard
2. The pixhawk is connected via TELEM2 to a Beagle Bone Black (running ROS indigo with mavros installed) using an FTDI board
I have successfully run the px4.launch without any error and do receive a connection heartbeat. The pixhawk can be armed and disarmed using the command 'rosrun mavros mavsafety arm(or disarm)' with very high delay of 7-8 seconds. I am however unable to run the 'setpoint_position_demo' file as described in https://pixhawk.org/dev/offboard_control/testing. The error being encountered is 'ImportError: No module named tf.transformations'. 

Here are my questions...
1. Is there a way to decrease the 7-8 seconds of delay in sending commands? Is this being caused because of the beaglebone's processing limitations?
2. Where would I be able to find the tf.transformations module and how to get it installed in the right place?
3. I ultimately want to send commands which send Thrust, Pitch, Roll, and Yaw angles to the Pixhawk from an algorithm that would run on the BeagleBone Black. Could somebody please guide me in this regard?

Thanks!

vooon, 2015-10-22:
1. I not sure why. What baud rate used?
2. Install `tf` package (`ros-jade-tf` or `ros-indigo-tf` depending on your distro)
3. setpoint_attitude

vooon, 2015-10-29:
Ping.

closed 2015-10-31 by vooon