mavlink/mavros #405: Errors are reported duing my installation, anyone could help? Thanks a lot

peilibuaa, 2015-10-22:
Some problems come out when I try to build and install this tool with the following codes
"""
wstool init src # (if not already initialized)
wstool set -t src mavros --git https://github.com/mavlink/mavros.git
wstool update -t src
rosdep install --from-paths src --ignore-src --rosdistro indigo -y
catkin_make
"""
[image]

vooon, 2015-10-22:
Update mavlink package from shadow-fixed repo:
- http://packages.ros.org/ros-shadow-fixed/ubuntu/pool/main/r/ros-indigo-mavlink/
- http://packages.ros.org/ros-shadow-fixed/ubuntu/pool/main/r/ros-jade-mavlink/

vooon, 2015-10-29:
Ping.

closed 2015-10-31 by vooon