mavlink/mavros #409: installing ros-*-mavlink from source -> $ROSINSTALL does not exist

ChristophTobler, 2015-10-28:
https://github.com/mavlink/mavros/blob/master/mavros/README.md

catkin_make_isolated --install-space $ROSINSTALL --install -DCMAKE_BUILD_TYPE=Release

-> $ROSINSTALL should be replaced by (the beginning of?) $ROS_PACKAGE_PATH

mhkabir, 2015-10-28:
> -> $ROSINSTALL should be replaced by (the beginning of?) $ROS_PACKAGE_PATH
Yep, that's correct.

As-such I'd recommend that you switch to the new catkin build tools, which is much more simpler to use, and you can directly build mavlink in the same workspace.

vooon, 2015-10-28:
Not exactly `$ROS_PACKAGE_PATH` provide several paths: workspace, installation (+ possible another workspace).

`$ROSINSTALL` should be manually set to place where you want to install package.

But new catkin tool is much easier, simply place mavlink-gpb package and build.
```
$ rosinstall_generator --rosdistro jade mavlink
```
```yaml
- git:
    local-name: mavlink
    uri: https://github.com/mavlink/mavlink-gbp-release.git
    version: release/jade/mavlink/2015.10.10-0
```

LorenzMeier, 2015-10-28:
Could you please fix the README accordingly? It's too verbose and outdated. It should contain less instructions, but those should be accurate.

vooon, 2015-10-28:
Ok. But need to decide what tool to use: old catkin_make or catkin.

Catkin tools in my opinion more user friendly, also it is available for Indigo+ (even Hydro - Travis use it now).

vooon, 2015-10-31:
@ChristophTobler updates?

ChristophTobler, 2015-11-03:
when i follow these new source  installation instructions I get (after wstool merge -t src /tmp/mavros.rosinstall):
ERROR in config: /home/christophtobler/catkin_ws_mavros/src has no workspace configuration file '.rosinstall'

ChristophTobler, 2015-11-03:
And if I try to install it like this: http://answers.ros.org/answers/115386/revisions/
with a file named mavros_catkin.rosinstall with:
- git:
    uri: https://github.com/mavlink/mavros.git
    local-name: src/mavros
    version: master

as soon as I run "catkin_make" I get the error:
/home/christophtobler/mavros_test/src/mavros/mavros/src/plugins/sys_status.cpp: In member function ‘void mavplugin::SystemStatusPlugin::handle_extended_sys_state(const mavlink_message_t*, uint8_t, uint8_t)’:
/home/christophtobler/mavros_test/src/mavros/mavros/src/plugins/sys_status.cpp:594:3: error: ‘mavlink_extended_sys_state_t’ was not declared in this scope
   mavlink_extended_sys_state_t state;
   ^
/home/christophtobler/mavros_test/src/mavros/mavros/src/plugins/sys_status.cpp:594:32: error: expected ‘;’ before ‘state’
   mavlink_extended_sys_state_t state;
                                ^
/home/christophtobler/mavros_test/src/mavros/mavros/src/plugins/sys_status.cpp:595:47: error: ‘state’ was not declared in this scope
   mavlink_msg_extended_sys_state_decode(msg, &state);
                                               ^
/home/christophtobler/mavros_test/src/mavros/mavros/src/plugins/sys_status.cpp:595:47: note: suggested alternative:
In file included from /usr/include/boost/thread.hpp:24:0,
                 from /opt/ros/jade/include/diagnostic_updater/diagnostic_updater.h:48,
                 from /home/christophtobler/mavros_test/src/mavros/mavros/include/mavros/mavros_plugin.h:21,
                 from /home/christophtobler/mavros_test/src/mavros/mavros/src/plugins/sys_status.cpp:17:
/usr/include/boost/thread/future.hpp:168:14: note:   ‘boost::future_state::state’
         enum state { uninitialized, waiting, ready, moved, deferred };
              ^
/home/christophtobler/mavros_test/src/mavros/mavros/src/plugins/sys_status.cpp:595:52: error: ‘mavlink_msg_extended_sys_state_decode’ was not declared in this scope
   mavlink_msg_extended_sys_state_decode(msg, &state);
                                                    ^

vooon, 2015-11-03:
https://github.com/mavlink/mavros/tree/master/mavros#source-installation - *Build error* note.

If you has uninitialized workspace - use init. Just read `wstool init --help`!

ChristophTobler, 2015-11-03:
I already have a fresh mavlink package and already tried with wstool init.

order of what I have done:

sudo apt-get install python-catkin-tools

# unneded if you already has workspace
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws
catkin init / or wstool init

# get source (upstream - released)
rosinstall_generator --upstream mavros | tee /tmp/mavros.rosinstall
# alternative: latest source
rosinstall_generator --upstream-development mavros | tee /tmp/mavros.rosinstall

# workspace & deps
wstool merge -t src /tmp/mavros.rosinstall
-> here I get the error: ERROR in config: /home/christophtobler/catkin_ws_test/src has no workspace configuration file '.rosinstall'

ChristophTobler, 2015-11-03:
ok. the init was my bad...
but now when I run "catkin build" I get:
make[1]: *** [CMakeFiles/mavros_plugins.dir/all] Error 2
make: *** [all] Error 2
[mavros] <== '/home/christophtobler/new_catkin_ws/build/mavros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' failed with return code '2'
                                                                                 
Failed   <== mavros        [ 36.5 seconds ]                                     
[build] There were '1' errors:                                                  

Failed to build package 'mavros' because the following command:

   Command to reproduce:
cd /home/christophtobler/new_catkin_ws/build/mavros && /home/christophtobler/new_catkin_ws/build/mavros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j; cd -

    Path to log:
cat /home/christophtobler/new_catkin_ws/build/build_logs/mavros.log

Exited with return code: 2 
                                                                                
[build] Runtime: 48.6 seconds 

any suggestions?

ChristophTobler, 2015-11-03:
works, thanks!

LorenzMeier, 2015-11-03:
@vooon could we update the docs accordingly? It seems quite a few people are challenged with building mavros.

LorenzMeier, 2015-11-03:
Will close this once we are confident it's foolproof.

vooon, 2015-11-03:
@LorenzMeier perhaps good idea is to merge mavlink and mavros source installation.
Because with `catkin` we may simply drop sources and that's it.

With that will be guaranteed that mavlink is newest (which usually required for master branch).

closed 2015-11-04 by vooon