mavlink/mavros #412: can't start HIL,need help

killerlayup, 2015-10-29:
Hi, all

I follow the setup on http://pixhawk.org/dev/ros/hil, and can start jMAVSim and verify that it is communicating with the Pixhawk as it prints:

    joel@joel-X450VC:~/jMAVSim$ java -Djava.ext.dirs= -cp lib/*:out/production/jmavsim.jar   
    me.drton.jmavsim.Simulator -serial /dev/ttyACM0 921600 -qgc
    Success!
    3D [dev] 1.6.0-pre12-daily-experimental daily

    Init MAVLink
    MSG: Switched to ON hil state
    MSG: [cmd] DISARMED by set mode command

but when I run :

    cd ~/ros_catkin_ws  (or cd ~/catkin_ws) 
    roslaunch mavros px4.launch

It seems that it is not communicating with pixhawk, no heartbeat. when I run:

    rosrun mavros mavsys mode -c OFFBOARD

it shows:

    [ INFO] [1446107174.611400916]: Autostarting mavlink via USB on PX4
    [ INFO] [1446107174.611560334]: MAVROS started. MY ID [1, 240], TARGET ID [1, 1]
    [ERROR] [1446107208.174644376]: MODE: Unsupported FCU

by the way, I use ubuntu 12.04,and px4 firmware, really need help~

vooon, 2015-10-29:
Dump here `rostopic echo -n1 /diagnostics`.

What version of mavros, px4 and ROS used?
You do not have connection, do not expect that set_mode will work.

killerlayup, 2015-10-29:
I use ros hydro and the latest px4, and hydro level mavros.

I think jMAVSim is connecting with the Pixhawk, as it shows:

    Init MAVLink
    MSG: Switched to ON hil state
    MSG: [cmd] DISARMED by set mode command

in http://pixhawk.org/dev/ros/hil it says: 

         setpoint_demo.py
                          /
    QGC <-> ROS/Mavros <-> jMAVSim <-> Pixhawk

so I think it may be something wrong between jMAVSim and Mavros.

killerlayup, 2015-10-29:
i run rostopic echo -n1 /diagnostics:

header: 
  seq: 1745
  stamp: 
    secs: 1446108917
    nsecs: 55892033
  frame_id: ''
status: 
  - 
    level: 1
    name: mavros: FCU connection
    message: not connected
    hardware_id: Mavlink
    values: 
      - 
        key: Received packets:
        value: 0
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 0
      - 
        key: Tx sequence number:
        value: 0
  - 
    level: 0
    name: mavros: GCS bridge
    message: connected
    hardware_id: Mavlink
    values: 
      - 
        key: Received packets:
        value: 0
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 0
      - 
        key: Tx sequence number:
        value: 0
  - 
    level: 2
    name: mavros: GPS
    message: No satellites
    hardware_id: Mavlink
    values: 
      - 
        key: Satellites visible
        value: -1
      - 
        key: Fix type
        value: 0
      - 
        key: EPH (m)
        value: Unknown
      - 
        key: EPV (m)
        value: Unknown
  - 
    level: 2
    name: mavros: Heartbeat
    message: No events recorded.
    hardware_id: Mavlink
    values: 
      - 
        key: Events in window
        value: 0
      - 
        key: Events since startup
        value: 0
      - 
        key: Duration of window (s)
        value: 10.008362
      - 
        key: Actual frequency (Hz)
        value: 0.000000
      - 
        key: MAV Type
        value: 0
      - 
        key: Autopilot type
        value: 0
      - 
        key: Autopilot base mode
        value: 0x00
      - 
        key: Autopilot custom mode
        value: 0x00000000
      - 
        key: Autopilot system status
        value: 0
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: Mavlink
    values: 
      - 
        key: Sensor present
        value: 0x00000000
      - 
        key: Sensor enabled
        value: 0x00000000
      - 
        key: Sensor helth
        value: 0x00000000
      - 
        key: CPU Load (%)
        value: 0.0
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 2
    name: mavros: Battery
    message: No data
    hardware_id: Mavlink
    values: 
      - 
        key: Voltage
        value: -1.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 0.0
  - 
    level: 2
    name: mavros: APM Memory
    message: No data
    hardware_id: Mavlink
    values: 
      - 
        key: Free memory (B)
        value: -1
      - 
        key: Heap top
        value: 0x0000
  - 
    level: 2
    name: mavros: APM Hardware
    message: No data
    hardware_id: Mavlink
    values: 
      - 
        key: Core voltage
        value: -1.000000
      - 
        key: I2C errors
        value: 0
  - 
    level: 2
    name: mavros: Time Sync
    message: No events recorded.
    hardware_id: Mavlink
    values: 
      - 
        key: Events in window
        value: 0
      - 
        key: Events since startup
        value: 0
      - 
        key: Duration of window (s)
        value: 10.008364
      - 
        key: Actual frequency (Hz)
        value: 0.000000
      - 
        key: Last dt (ms)
        value: 0.000
      - 
        key: Mean dt (ms)
        value: 0.000
      - 
        key: Last system time (s)
        value: 590726812732.479492
---

vooon, 2015-10-29:
I think you should use that launch: https://github.com/PX4/Firmware/blob/master/launch/mavros_sitl.launch

And your mavros too old.

vooon, 2015-10-29:
```xml
	<arg name="fcu_url" default="udp://:14554@192.168.1.5:14555" />
	<arg name="gcs_url" default="udp://:14556@192.168.1.5:14550" />
```

So at least you should provide this parameters to px4.launch.

killerlayup, 2015-10-29:
@vooon 
why 192.168.1.5?

killerlayup, 2015-10-29:
here is my launch file:
```xml
<launch>
	<!-- vim: ft=xml -->
	<!-- example launch script for PX4 based FCU's -->
	<arg name="gcs_url" default="udp://:14556@192.168.1.100:14550" /> 
	<arg  name="fcu_url" default="udp://:14554@192.168.1.100:14555" />
	<arg name="tgt_system" default="1" />
	<arg name="tgt_component" default="1" />

	<include file="$(find mavros)/launch/node.launch">
		<arg name="blacklist_yaml" value="$(find mavros)/launch/px4_blacklist.yaml" />
		<arg name="config_yaml" value="$(find mavros)/launch/px4_config.yaml" />

		<arg name="fcu_url" value="$(arg fcu_url)" />
		<arg name="gcs_url" value="$(arg gcs_url)" />
		<arg name="tgt_system" value="$(arg tgt_system)" />
		<arg name="tgt_component" value="$(arg tgt_component)" />
	</include>
</launch>
```

Is there something wrong?

killerlayup, 2015-10-29:
@v01d 
@liamstask 
@claymation 
@dagar 
I really need help, anyone help me :(

vooon, 2015-10-29:
Perhaps there may be `localhost`, not sure on what IP jMAVSim binds.
I recommend to `use px4.launch fcu_url:=<url> gcs_url:=<url>`, script from manual little outdated (before i added whitelist).
Or you can do your own:
```xml
<launch>
<include file="$(find mavros)/launch/px4.launch">
<arg name="fcu_url" value="<url>" />
<arg name="gcs_url" value="<url>" />
</include>
</launch>
```

Anyway your mavros looks outdated (something about version 0.8-0.9), i recommend to upgrade
because there lots of bugs fixed and specially with setpoint plugins.

Next check with `netstat` does simulator listening on fcu port (:14550).
Note that you can connect with QGC directly to that port - check that problem in mavros.

killerlayup, 2015-10-31:
@

killerlayup, 2015-10-31:
@vooon
hi, I update my ubuntu to 14.04 and ros to indigo,and update the mavros too. 
but I still can't fix this issue.

Meanwhile, when I connect pixhawk via serial port and run: 

    roslaunch mavros px4.launch 

it can be connected as it shows:

[ INFO] [1446334866.819164422]: MAVROS started. MY ID [1, 240], TARGET ID [1, 1]
[ WARN] [1446334866.909027438]: TM: Clock skew detected (-9223372036.854730606 s). Hard syncing clocks.
[ INFO] [1446334867.796013881]: CON: Got HEARTBEAT, connected.
[ WARN] [1446334869.797809052]: VER: broadcast request timeout, retries left 4
[ WARN] [1446334870.797978701]: VER: broadcast request timeout, retries left 3
[ WARN] [1446334871.797738623]: VER: unicast request timeout, retries left 2
[ WARN] [1446334872.797911530]: VER: unicast request timeout, retries left 1
[ WARN] [1446334873.797571743]: VER: unicast request timeout, retries left 0
[ WARN] [1446334874.798431048]: VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities
[ WARN] [1446334878.208535680]: TM: Clock skew detected (9223372036.854656219 s). Hard syncing clocks.

BUT when I run 

    rosrun mavros mavsys mode -c OFFBOARD

it shows：

    [ERROR] [1446334878.497378022]: MODE: Unsupported FCU

why does this happened? Can you help me?

Thanks a lot

vooon, 2015-10-31:
Because:
```
      - 
        key: Autopilot type
        value: 0
```

0 - Generic - modes unknown. We support only APM (ArduPilotMega) and PX4 firmwares.
Also check that with recent version "Autopilot type" should be string.

killerlayup, 2015-11-01:
hi, @vooon 

here is log when I run   'rostopic echo -n1 /diagnostics'

header: 
  seq: 25
  stamp: 
    secs: 1446401269
    nsecs: 108194133
  frame_id: ''
status: 
  - 
    level: 1
    name: mavros: FCU connection
    message: not connected
    hardware_id: /dev/ttyACM0:57600
    values: 
      - 
        key: Received packets:
        value: 0
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 0
      - 
        key: Tx sequence number:
        value: 27
      - 
        key: Rx total bytes:
        value: 4
      - 
        key: Tx total bytes:
        value: 459
      - 
        key: Rx speed:
        value: 0.000000
      - 
        key: Tx speed:
        value: 17.000000
  - 
    level: 2
    name: mavros: GPS
    message: No satellites
    hardware_id: /dev/ttyACM0:57600
    values: 
      - 
        key: Satellites visible
        value: 0
      - 
        key: Fix type
        value: 0
      - 
        key: EPH (m)
        value: Unknown
      - 
        key: EPV (m)
        value: Unknown
  - 
    level: 2
    name: mavros: Heartbeat
    message: No events recorded.
    hardware_id: /dev/ttyACM0:57600
    values: 
      - 
        key: Heartbeats since startup
        value: 0
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Vehicle type
        value: Generic
      - 
        key: Autopilot type
        value: Generic
      - 
        key: Mode
        value: ''
      - 
        key: System status
        value: Uninit
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: /dev/ttyACM0:57600
    values: 
      - 
        key: Sensor present
        value: 0x00000000
      - 
        key: Sensor enabled
        value: 0x00000000
      - 
        key: Sensor helth
        value: 0x00000000
      - 
        key: CPU Load (%)
        value: 0.0
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 2
    name: mavros: Battery
    message: No data
    hardware_id: /dev/ttyACM0:57600
    values: 
      - 
        key: Voltage
        value: -1.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 0.0
---
joel@joel-X450VC:~/catkin_ws$ rostopic echo -n1 /diagnostics
header: 
  seq: 23
  stamp: 
    secs: 1446401626
    nsecs: 611840330
  frame_id: ''
status: 
  - 
    level: 1
    name: mavros: FCU connection
    message: not connected
    hardware_id: udp://:14554@192.168.1.126:14555
    values: 
      - 
        key: Received packets:
        value: 0
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 0
      - 
        key: Tx sequence number:
        value: 25
      - 
        key: Rx total bytes:
        value: 0
      - 
        key: Tx total bytes:
        value: 425
      - 
        key: Rx speed:
        value: 0.000000
      - 
        key: Tx speed:
        value: 17.000000
  - 
    level: 2
    name: mavros: GPS
    message: No satellites
    hardware_id: udp://:14554@192.168.1.126:14555
    values: 
      - 
        key: Satellites visible
        value: 0
      - 
        key: Fix type
        value: 0
      - 
        key: EPH (m)
        value: Unknown
      - 
        key: EPV (m)
        value: Unknown
  - 
    level: 2
    name: mavros: Heartbeat
    message: No events recorded.
    hardware_id: udp://:14554@192.168.1.126:14555
    values: 
      - 
        key: Heartbeats since startup
        value: 0
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Vehicle type
        value: Generic
      - 
        key: Autopilot type
        value: Generic
      - 
        key: Mode
        value: ''
      - 
        key: System status
        value: Uninit
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: udp://:14554@192.168.1.126:14555
    values: 
      - 
        key: Sensor present
        value: 0x00000000
      - 
        key: Sensor enabled
        value: 0x00000000
      - 
        key: Sensor helth
        value: 0x00000000
      - 
        key: CPU Load (%)
        value: 0.0
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 2
    name: mavros: Battery
    message: No data
    hardware_id: udp://:14554@192.168.1.126:14555
    values: 
      - 
        key: Voltage
        value: -1.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 0.0
---

So, it shows:

    - 
            key: Vehicle type
            value: Generic
          - 
            key: Autopilot type
            value: Generic

But it still shows：

    [ERROR] [1446334878.497378022]: MODE: Unsupported FCU

when I run 

    rosrun mavros mavsys mode -c OFFBOARD

I really confused.. :{

killerlayup, 2015-11-02:
anyone help???

vooon, 2015-11-02:
Autopilot is still `Generic`. It is issue source.

vooon, 2015-11-02:
And then it it not mavros but firmware side.

killerlayup, 2015-11-02:
do you mean that something is wrong with my pixhawk Firmware?
my pixhawk firmware is Firmware/stable/px4fmu-v2_default.px4 ,is that right?

vooon, 2015-11-02:
Oh, i overlooked this:

```
  - 
    level: 1
    name: mavros: FCU connection
    message: not connected
    hardware_id: Mavlink
```

**not connected** - check your connection URL!

killerlayup, 2015-11-02:
hi @vooon 
sorry to bother you again, now I can connect px4 with jMAVSim,  meanwhile I can connect px4 with MAVROS,but when I start jMAVSim and MAVROS like :

                     setpoint_demo.py
                       /
    QGC <-> ROS/Mavros <-> jMAVSim <-> Pixhawk

jMAVSim can't connect with Mavros, as it shows:

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_url: udp://:14554@loca...
 * /mavros/gcs_url: udp://:14556@loca...
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: fcu
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
 * /mavros/mocap/use_tf: False
 * /mavros/plugin_blacklist: ['safety_area', '...
 * /mavros/plugin_whitelist: []
 * /mavros/px4flow/frame_id: px4flow
 * /mavros/px4flow/ranger_fov: 0.0
 * /mavros/px4flow/ranger_max_range: 5.0
 * /mavros/px4flow/ranger_min_range: 0.3
 * /mavros/safety_area/p1/x: 1.0
 * /mavros/safety_area/p1/y: 1.0
 * /mavros/safety_area/p1/z: 1.0
 * /mavros/safety_area/p2/x: -1.0
[... 73 lines cut ...]

no heartbeat! here is my launch file :

'''xml
<launch>
	
	

	<arg name="fcu_url" default="udp://:14554@localhost:14555" />
	<arg name="gcs_url" default="udp://:14556@localhost:14550" />
	<arg name="tgt_system" default="1" />
	<arg name="tgt_component" default="50" />

	<include file="$(find mavros)/launch/node.launch">
		<arg name="config_yaml" value="$(find mavros)/launch/px4_config.yaml" />
		<arg name="pluginlists_yaml" value="$(find mavros)/launch/px4_pluginlists.yaml" />
		<arg name="fcu_url" value="$(arg fcu_url)" />
		<arg name="gcs_url" value="$(arg gcs_url)" />
		<arg name="tgt_system" value="$(arg tgt_system)" />
		<arg name="tgt_component" value="$(arg tgt_component)" />
	</include>
</launch>
'''

it really confused me..

killerlayup, 2015-11-02:
well, I have another question, in jMAVSim, when I start jMAVSim, it can communicate with px4, but the log shows:

joel@joel-X450VC:~/jMAVSim$ java -Djava.ext.dirs= -cp lib/*:out/production/jmavsim.jar me.drton.jmavsim.Simulator -serial /dev/ttyACM0 921600 -qgc
Success!
3D [dev] 1.6.0-pre12-daily-experimental daily

WARNING: Got heartbeat from system #2 but configured to only accept messages from system #1. Please change the system ID parameter to match in order to use HITL/SITL.
Init MAVLink
WARNING: Got heartbeat from system #2 but configured to only accept messages from system #1. Please change the system ID parameter to match in order to use HITL/SITL.
WARNING: Got heartbeat from system #2 but configured to only accept messages from system #1. Please change the system ID parameter to match in order to use HITL/SITL.
WARNING: Got heartbeat from system #2 but configured to only accept messages from system #1. Please change the system ID parameter to match in order to use HITL/SITL.
WARNING: Got heartbeat from system #2 but configured to only accept messages from system #1. Please change the system ID parameter to match in order to use HITL/SITL.
WARNING: Got heartbeat from system #2 but configured to only accept messages from system #1. Please change the system ID parameter to match in order to use HITL/SITL.

how to fix this problem? Is that related to MAV_SYS_ID？

vooon, 2015-11-02:
It is related to network connection. Until you got:
```
[INFO] CON: Got HEARTBEAT, connected.
```

There no connection. Why? Who knows... But first check what ports actually jMAVSim actually listens.
Because looks like that mavros url configured for wrong ports and because UDP is connectionless you can't check that you trying wrong address/port.

I'm not sure, perhaps there some changes in jMAVSim, check sources.

killerlayup, 2015-11-03:
hi, @vooon 
now I can start jMAVSim and mavros together and mavros can get heartbeat. But I can't set mode to OFFMODE as the log says: 

    MSG: REJECT OFFBOARD

I read the setup at https://pixhawk.org/dev/ros/mavros_offboard:

**Make sure the flight controller gets a stream of setpoints (0.5s timeout) before you switch to offboard mode. The system will reject offboard otherwise**

so I follow above:

first, arm the multirotor, done
second, run ./setpoint_demo.py , done
third, use my transmitter switch to set mode OFFBOARD

but it still says:

    MSG: REJECT OFFBOARD

I am really confused.

BTW, another question, when I use QGC via UDP,  QGC popup window says:

    Warning: A second system is using the same system id (255)

then the connection will be lost.
why did this happen?  Do you have any idea?

Thanks

killerlayup, 2015-11-03:
I forget one thing, when I use the command:

    rosrun mavros mavsys mode -c OFFBOARD

instead of using my transmitter switch, it shows the log on mavros terminal:

    [ERROR] [1446559320.758279126]: MODE: Unsupported FCU

Do you have any idea about this?

Thanks

vooon, 2015-11-03:
Please show diagnostics again.

killerlayup, 2015-11-04:
hi, @vooon 

here is the diagnostics log:

header: 
  seq: 42
  stamp: 
    secs: 1446599813
    nsecs: 416201253
  frame_id: ''
status: 
  - 
    level: 0
    name: mavros: FCU connection
    message: connected
    hardware_id: udp://:36895@localhost:36895
    values: 
      - 
        key: Received packets:
        value: 1055
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 22
      - 
        key: Tx sequence number:
        value: 23
      - 
        key: Rx total bytes:
        value: 24797
      - 
        key: Tx total bytes:
        value: 24797
      - 
        key: Rx speed:
        value: 517.000000
      - 
        key: Tx speed:
        value: 517.000000
  - 
    level: 0
    name: mavros: GCS bridge
    message: connected
    hardware_id: udp://:36895@localhost:36895
    values: 
      - 
        key: Received packets:
        value: 6
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 74
      - 
        key: Tx sequence number:
        value: 0
      - 
        key: Rx total bytes:
        value: 102
      - 
        key: Tx total bytes:
        value: 24768
      - 
        key: Rx speed:
        value: 0.000000
      - 
        key: Tx speed:
        value: 517.000000
  - 
    level: 2
    name: mavros: GPS
    message: No satellites
    hardware_id: udp://:36895@localhost:36895
    values: 
      - 
        key: Satellites visible
        value: 0
      - 
        key: Fix type
        value: 0
      - 
        key: EPH (m)
        value: Unknown
      - 
        key: EPV (m)
        value: Unknown
  - 
    level: 0
    name: mavros: Heartbeat
    message: Normal
    hardware_id: udp://:36895@localhost:36895
    values: 
      - 
        key: Heartbeats since startup
        value: 47
      - 
        key: Frequency (Hz)
        value: 0.999961
      - 
        key: Vehicle type
        value: Onboard-Controller
      - 
        key: Autopilot type
        value: INVALID
      - 
        key: Mode
        value: CMODE(0)
      - 
        key: System status
        value: Active
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: udp://:36895@localhost:36895
    values: 
      - 
        key: Sensor present
        value: 0x00000000
      - 
        key: Sensor enabled
        value: 0x00000000
      - 
        key: Sensor helth
        value: 0x00000000
      - 
        key: CPU Load (%)
        value: 0.0
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 2
    name: mavros: Battery
    message: No data
    hardware_id: udp://:36895@localhost:36895
    values: 
      - 
        key: Voltage
        value: -1.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 0.0
  - 
    level: 0
    name: mavros: Time Sync
    message: Normal
    hardware_id: udp://:36895@localhost:36895
    values: 
      - 
        key: Timesyncs since startup
        value: 270
      - 
        key: Frequency (Hz)
        value: 9.999617
      - 
        key: Last dt (ms)
        value: 0.004174
      - 
        key: Mean dt (ms)
        value: -0.000067
      - 
        key: Last system time (s)
        value: 1446599813.367889881
      - 
        key: Time offset (s)
        value: 9223372036.854730606
---

killerlayup, 2015-11-04:
do you have any free time?  could I invite you to gitter?
if that bothers you, tell me :)

vooon, 2015-11-04:
You may ask at mavros gitter channel.

You setted wrong UDP URL `udp://:36895@localhost:36895` - nope!
MAVROS listens to itself because you setted same port for bind and remote.

So it got self heartbeat:
```yaml
      - 
        key: Vehicle type
        value: Onboard-Controller
      - 
        key: Autopilot type
        value: INVALID
```

vooon, 2015-11-04:
@DrTon there still some problem with connection. We trying to setup following scheme:

[image]

vooon, 2015-11-04:
```dot
strict graph {
	subgraph ros {
		graph [
			label="ROS",
			compound=True,
			lheight=0.21,
			rank=same,
			style=bold]
		node [style=filled]
		mavros;
		gcs_bridge;
	}

	"Pixhawk" -- "mavros" [label="/dev/ttyACM0:921600"]
	"mavros" -- "jMAVSim" [label="udp://:14560@localhost:14565"]
	"mavros" -- "gcs_bridge" [label="ROSUDP"]
	"gcs_bridge" -- "QGroundControl" [label="udp://:14575@localhost:14570"]
}
```

killerlayup, 2015-11-05:
Hi, @vooon 

are you online?
last night I try it again,and only find can't set the 'OFFBOARD' mode

in mavros it shows:
```
[ERROR] [1446648948.318435175]: FCU: Not switching off HIL (safety)
[ERROR] [1446648948.368477778]: FCU: command temporarily rejected: 176
```
when I run rosrun mavros mavsys `mode -c OFFBOARD`

vooon, 2015-11-07:
@killerlayup seems that your error with arming related to #423.

vooon, 2015-11-08:
@killerlayup i fixed #423. Can you update and check?

killerlayup, 2015-11-09:
hi @vooon  
yes, I will checked it, thanks

iceicehyhy, 2015-11-11:
Hi @killerlayup,
i'm facing the same problem when establishing this framework, mavros could't get a heartbeat line when jMAVSim is used as a bridge to communicate with PX4. How did you solve the problem?

killerlayup, 2015-11-11:
Hi @iceicehyhy 
follow this https://github.com/DrTon/jMAVSim/issues/18

iceicehyhy, 2015-11-11:
Hi @killerlayup  , I followed the steps posted by @weiweikong, but the communication between QGC--->Mavros is not stable as I could always see " Connection Lost" . Any idea to solve this issue?
Btw, any updates for offboard control?
Thanks

killerlayup, 2015-11-12:
@iceicehyhy  
hi， I have the same issue with you about  communication between QGC--->Mavros, still not fixed.
But I can set OFFBOARD mode by following the step posted by @weiweikong

killerlayup, 2015-11-16:
@iceicehyhy 
Hi, have you solved your issue?

vooon, 2015-11-16:
@killerlayup #427 - silent.

You may try to use TCP connection method for GCS. Simply change gcs_url to `tcp-l://`.
Then use TCP connection in QGC to `127.0.0.1:5760`.

killerlayup, 2015-11-17:
yes！ it works! thanks man @vooon

vooon, 2015-11-17:
Finally :) i think may close issue now.

closed 2015-11-17 by vooon

weiweikong, 2015-11-17:
Why we here use TCP connection not UDP ? @vooon

vooon, 2015-11-17:
Just to eliminate possible interference with other mavlink talker programs that uses UDP.
By default all those programs uses ports 14550 or 14555 and because of connectionless nature of UDP both ends latch remote address from sender. So it is not to hard to break connection with transmission from some component.

I wanted to make multicast UDP transport, so port problem will gone. But really nobody support that now.
And most mavros plugins note ready because they won't check sender identifications.

iceicehyhy, 2015-11-17:
@vooon  
Hi, thanks so much for the solution.
@killerlayup 
Just chance gcs_url to  <arg name="gcs_url" default="tcp-l://" /> will make it work?

killerlayup, 2015-11-17:
@iceicehyhy  yep

iceicehyhy, 2015-11-17:
@killerlayup   yes, it works. thank you!

killerlayup, 2015-11-26:
@vooon  @iceicehyhy 
Hi guys!
Recently I have an issue about TCP connecttion which was OK a few days ago. 

when I connect via TCP, QGC popups window:

`Error on link TCP. Error on socket: Connection refused.`

killerlayup, 2015-11-26:
need help~~