mavlink/mavros #413: use setpoint_location to control APM seems fail

CrazyPopLin, 2015-10-29:
I can **arm** the APM successfully. But when I run ` rosrun mavros mavsys mode -c GUIDED` it said **Timed out !** after I launch the setpoint rosnode.
Here is the setpoint rosnode cpp file:

> #include <ros/ros.h>
#include <stdio.h>
#include "geometry_msgs/PoseStamped.h"
>
int main(int argc, char **argv)
{
   ros::init(argc, argv, "setpoint");
   ros::NodeHandle n;
   ros::Publisher position_pub = n.advertise<geometry_msgs::PoseStamped>("/mavros/setpoint_position/local",10);
   ros::Rate loop_rate(10);
>
   geometry_msgs::PoseStamped msg;
   while(ros::ok()){
       msg.header.stamp = ros::Time::now();
       msg.header.seq=1;
       msg.header.frame_id="local_origin";
       msg.pose.position.x = 0;
       msg.pose.position.y = 0;
       msg.pose.position.z = 2;
       msg.pose.orientation.x=0;
       msg.pose.orientation.y=0;
       msg.pose.orientation.z=0;
       msg.pose.orientation.w=1;
       position_pub.publish(msg);
       ros::spinOnce();
       loop_rate.sleep();
   }    
   return 0;
>}

Any suggestions will be thankful.

vooon, 2015-10-29:
Little strange. Do you able to switch to other modes?

CrazyPopLin, 2015-10-30:
Thanks @vooon , i fixed this issue when i put quad out of lab to GPS enabled place. But the motors seems didn't rotate when the setpoint rosnode running. Any suggestions?

killerlayup, 2015-11-01:
Hi, @CrazyPopLin 

Sorry to bother you, But I have some issues about connect pixhawk via MAVROS and set up HIL enviroment. 

here is my issue address

https://github.com/mavlink/mavros/issues/412

I test APM and px4 both, but all of  them have this issue, can you help me?

Thanks a lot!

CrazyPopLin, 2015-11-02:
I am sorry @killerlayup , I didn't use SILT before. Usually you need to arm the quad and the enable the OFFBOARD mode.

vooon, 2015-11-02:
@CrazyPopLin try other modes, all of them do not turn motors or not?

Then try different Z values (similar testing program `mavsetp` from scripts).
Also check that node is connected (`rosnode list`, `rosnode info <node>`, `rqt_graph`).

vooon, 2015-11-02:
Also please post console output (mavros log) from apm.launch.
Especially `FCU: *` messages.

jlian, 2015-11-02:
You should try your code with the SITL simulator first, which gives you mavproxy access as well, so you can use that to do A/B testing for things like arming and takeoff. 

In my experience with APM, arming the motors doesn't actually do anything than allow you the throttle up, which is what actually will turn spin the rotors.

To do autonomous flights you should follow `set GUIDED -> arm motors -> takeoff or throttle up -> setpoint`. This is for autonomous flights regardless of your off board control system (i.e. DroneKit will need to be set up in a similar way). In order to set GUIDED you need what's called "3D Fix" (GPS lock). After you arm the motors you need to rev up the motors within about 10 seconds to prevent automatic disarm. With `mavros` you could simply use `mavcmd` to auto takeoff right after you arm. I don't believe you could use setpoint unless you've taken off because it has literally never worked for me in any set up.

In your code I don't see your copter taking off. Try adding that in and see if it works then.

Also I'm under the impression that yaw control is faulty with set_point position from APM's side (see [here](https://github.com/diydrones/ardupilot/pull/2959)). I would use CONDITION_YAW instead.

CrazyPopLin, 2015-11-11:
Hi @jlian , do you have any ideas to throttle up.  i know **mavros_msgs/CommandTOL.srv** seems can be used for take off. But I am in the indoor environment without GPS. Thanks.

closed 2015-11-22 by vooon