mavlink/mavros #419: Guided mode equivalent in Pixhawk for indoor navigation

aitjellal, 2015-11-02:
Hi,  I am a PhD student and currently working with Pixhawk+Mavros
@vooon   @TSC21 
My set-up is :  Pixhawk + PixIO+ Optitrack + Mavros + Onboard Mini Computer (connected via USB to Serial FTDI) +  Indoor  No GPS
I have remapped the "optitrack, mocap" topic to "vision" and actived "vision" in QgroundControl.
I checked the topic "local_position/local" and it seems that the visual position is fused with IMU data. That's fine.
Now, I want to make indoor navigation (hovering , takeoff. land, Square flight )

In which flight mode should I set the Pixhawk.  I read that for  the "APM Software Stack" one could use the "Guided mode"     http://dev.ardupilot.com/wiki/copter-commands-in-guided-mode/ 
and set the desired position via :  SET_POSITION_TARGET_LOCAL_NED

is there an equivalent for this  flight mode  for "Pixhawk Stack" ?

Is it possible to takeoff (indoor) to a desired position autonomously after "arming" and switching from the manual mode?  
thanks.

jlian, 2015-11-02:
I use APM with GUIDED exclusively and am not too familiar with PX4 flight stack. In mavros the SET_POSITION_TARGET_LOCAL_NED is achieved with the `setpoint_position` plugin (see here http://wiki.ros.org/mavros#mavros.2BAC8-Plugins.). In PX4 I was under the impression that it's called OFFBOARD mode and the same set and the `setpoint_position` plugin should work the same way. Have you read this article https://pixhawk.org/dev/ros/mavros_offboard ?

For arming and takeoff autonomously it should be possible in PX4 as it is possible with APM. You need to set up service clients for [`~/cmd/arming` and `~/cmd/takeoff`](http://wiki.ros.org/mavros#mavros.2BAC8-Plugins.command) and also [`~set_mode](http://wiki.ros.org/mavros#mavros.2BAC8-Plugins.Services-3). This would be inside your own ROS program. From the command line it's even easier as you can just use [`mavcmd`](http://wiki.ros.org/mavros#mavcmd)

aitjellal, 2015-11-02:
Thank's @jlian 

can anybody confirm that the flight mode  "OFFBOARD in PX4"  is equivalent to "Guided in APM" ?

Would it work with the Mode "Assist -- POSCTL (position controlled) "  ?  https://pixhawk.org/users/system_modes  

" Have you read this article https://pixhawk.org/dev/ros/mavros_offboard ?"         Yes, but I did not understand what is does.  I wondered that OFFBOARD is used when the user wants to implement his own controller. 

I am confused, I do not know which flight mode to use.
best regards.

mhkabir, 2015-11-02:
> confirm that the flight mode "OFFBOARD in PX4" is equivalent to "Guided in APM"

Yes

>Would it work with the Mode "Assist -- POSCTL (position controlled) " ?

No. Use offboard.

aitjellal, 2015-11-02:
thanks @mhkabir  Mohammed .  
so I proceed just as @jlian  has described. 
does  the command ~cmd/takeoff also work in  indoor with local coordinate system NED (no gps , no global position estimate)?

aitjellal, 2015-11-03:
I checked 
~cmd/takeoff (mavros_msgs/CommandTOL)
    Send takeoff command. 
The ROS topic type is as follow : http://docs.ros.org/api/mavros_msgs/html/srv/CommandTOL.html

# Common type for Take Off and Landing http://docs.ros.org/api/mavros_msgs/html/srv/CommandTOL.html
float32 min_pitch # used by takeoff
float32 yaw
float32 latitude
float32 longitude
float32 altitude
---
bool success
uint8 result
------------------------------------------
so, I think the command "~cmd/takeoff "   does only work if GPS is available.  
for my indoor experiment using Optitrack, I think I can not use  "~cmd/takeoff ".
can any body confirm this ? @mhkabir   @jlian  @LorenzMeier 

thanks.

jlian, 2015-11-03:
I would really try to do all of these things in the simulator first before the real aircraft. You can turn off GPS in the simulator too, in pretty sure.

vooon, 2015-11-17:
Updates?

aitjellal, 2015-11-17:
I can flight autonomously using Optitrack. Thank you guys.

The mocap topic is remapped to "vision".
I have adapted the program 
 https://pixhawk.org/dev/offboard_control/testing for indoor use. (fly just  0.8m instead of 8m).

This program send desired position via  "mavros/setpoint_position/local "
the programm also subscribe to " mavros/local_position/local"  to get the current position In order to check if the desired position is reached or not.

I have enabled the "offboard" flight mode using QGC and a assigned an RC switch for it.
Flight procedure :
- make sure everything is OK:
     - Simulate the flight without prop. 
     - and move the quad to the defined positions and make sure that the programm will set the next  
       "waypoint" as desired position once the current desired position is reached.  

- Set the switch to "manuel mode "  in the RC.
- Switch on the RC.
- arm the Quadcopter
- switch to "Offboard mode".   success flight.

I still have an issue, which I think is related to "parameter tuning". I did no parameter tuning. I used the default values.

The Quadcopter fly slowly most of the time to a position above the desired position. and keep going up. 
I this a parameters tuning problem ? or did I miss something ? Any link to good tutorials where I can learn how to solve this problem.
thank you.

vooon, 2015-11-17:
Did you tried to fly it manually (via RC)? If it's response is same then it is definitely PID tuning problem.
My knowledge about copters is minimal, so please check manuals in px4 wiki and ask questions at px4-users.

closed 2015-11-17 by vooon

shiohigari, 2016-01-21:
Hey aitjellal,
I am working on trying to get a similar configuration working properly but have been having some troubles. When you start publishing to  "mavros/setpoint_position/local " does the light on the pixhawk go green like if you had a gps lock and stay green?

aitjellal, 2016-01-21:
hi, does the light on the pixhawk go green like if you had a gps lock and stay green?  no.  it does not. light  remains blue.

Date: Thu, 21 Jan 2016 01:47:30 -0800