mavlink/mavros #421: Need clarification on coordinate frame convention

jlian, 2015-11-02:
Hi, I'm trying to implement a simple velocity controller with mavros and APM. I've written a simple method to send velocity setpoints in what I thought would be the ENU conventions [as describe in the docs](https://github.com/mavlink/mavros/blob/master/mavros/README.md#coordinate-frames). Here's the code snippet:

```c++
...

// control the velocity of the copter
string vel_ = nh.resolveName("/mavros/setpoint_velocity/cmd_vel");
vel = nh.advertise<geometry_msgs::TwistStamped>(vel_,1);

...

void setEastNorthVelocity(double eastwardVelocity, double northwardVelocity)
{
    geometry_msgs::TwistStamped control_msg;

    control_msg.twist.linear.x = eastwardVelocity;
    control_msg.twist.linear.y = northwardVelocity;

    vel.publish(control_msg);    
}
```

Now when I call `setEastNorthVelocity(5, 10)` I expected the copter to fly towards the northeast at 5 m/s eastward and 10 m/s northward. However, I actually see that it goes **westward** at 10 m/s and northward at 5 m/s (see attached image), suggesting to me that the coordinate frame is actually *WNU*. Obviously it's an easy fix but I can't help but wonder if it's a bug or if I have the wrong understanding. Help would be appreciated. @ssoRL

[image]

vooon, 2015-11-02:
If i remember correctly that command is body related. So X - forward, Y - left, Z -up.

http://www.ros.org/reps/rep-0103.html#axis-orientation

jlian, 2015-11-03:
I thought this was the case too but I found this:

https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/setpoint_velocity.cpp#L76

In my understanding MAV_FRAME_LOCAL_NED means vehicle frame and MAV_FRAME_BODY_NED means vehicle-1 (body) frame. 

Also the `vel_enu` comes from the ROS topic right? So everything I see here is in the vehicle frame which should be ENU. 

I haven't implemented heading control yet so I report back when I can change the heading, which should confirm the frame.

jlian, 2015-11-03:
Hi @vooon,

So I was able to change the heading of the vehicle and I set the same `setEastNorthVelocity(5, 10)` command again. This time it's very clear that the command was being interpreted by APM as a vehicle frame command, *not* vehicle-1/body frame: 

[image]

Would you agree that this is a bug? What is the intended behaviour?

closed 2015-11-22 by vooon