mavlink/mavros #422: Mavros has old message definition for Odroid XU4

ilia-sheremet, 2015-11-03:
Hi,

I have mavros installed on Odroid XU4 (Samsung Exynos5422 Cortex™-A15 2Ghz and Cortex™-A7 Octa core CPUs). But when I install it using apt-get install ros-jade-mavros* (for jade or indigo ros the same problem) the version is outdated to compare with PC version, which make impossible to exchange /mavros/rc messages. The one on Odroid has mavros/RCIn - Out, but the PC one mavros_msgs/RCIn - Out

The packages being installed on Odroid through apt-get install  ros-jade-mavros* are:
ros-jade-mavros (0.11.2-0trusty-20150612-150936-0700_armhf.deb)
ros-jade-mavros-extras (0.11.2-0trusty-20150612-164450-0700_armhf.deb)

While PC version has also mavros_msgs package.

I tried to compile the source code following this https://github.com/mavlink/mavros/blob/master/mavros/README.md#source-installation , but on the last step the build fails.

Is there a way to get the binaries through apt-get? Or another way to fix the problem?

vooon, 2015-11-03:
Unfortunately ROS buildfarm for armhf seems stall. So source installation is best choice.
You need both mavros and mavlink packages.

ilia-sheremet, 2015-11-03:
I am not sure I understand properly the buildfarm concept, even after reading the page about that .. Is there a source code for Mavros armhf package? Or what exactly buildfarm armhf imply, the process of test? 
I am also having troubles building Mavros source code - the build just freezes at some point. If I just manually copy mavros/, mavros_msgs/ .. etc.  .. to /opt/ros/dist/ share\ includu\ bin\ .. etc. will the package work?
Thank you in advance!

vooon, 2015-11-04:
C++ it is compilable language, so you need building step. That what buildfarm do. Also it may run automatic tests (Continuous Integration) and make binary debian packages (.deb).

Copying do not work at all. Read [source installation](https://github.com/mavlink/mavros/blob/master/mavros/README.md#source-installation) again. It only prerequest that you already [installed ROS distribution](http://wiki.ros.org/jade/Installation/UbuntuARM).

ilia-sheremet, 2015-11-04:
Thank you, @vooon , once again for fast replies.

By "If I just manually copy mavros/, mavros_msgs/ .. etc. .. to /opt/ros/dist/ share\ includu\ bin\ .. etc. will the package work?" I meant folders that appears in devel folder after build. 

I manage to build on PC without problems. A feedback to [source installation](https://github.com/mavlink/mavros/blob/master/mavros/README.md#source-installation) that you might find useful - apart from "wstool", I had also to install "rosinstall_generator". And could you please help me to find the way to properly install the package after the build (it might be also a good point in the end of Source Installation). Running "make install" from catkin_workspace/build/mavros installs only mavros package and at the same level as build folder.  

According to instruction on [this page]  (http://wiki.ros.org/catkin/commands/catkin_make), I tried to run this command from catkin_workspace/:
cmake src/mavros/mavros/ -DCMAKE_INSTALL_PREFIX=/my_install -DCATKIN_DEVEL_PREFIX=/my_devel
where I would change /my_install to /opt/ros/jade/ if it works.
But I got this error: 

    CMake Error at /opt/ros/jade/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
    Could not find a package configuration file provided by "mavros_msgs" with
    any of the following names:

    mavros_msgsConfig.cmake
    mavros_msgs-config.cmake

and only my_devel folder was created.

And the last question - if I copy the /build folder after build on PC to Odroid and then execute make install to /opt/ros/jade will it work out? Because the compilation on Odroid for some reason stucks.

LorenzMeier, 2015-11-04:
@vooon It would be fantastic if we could get this in short and concise into the readme. A lot of people seem to be running into this, and we need to get better at making it accessible. In particular since mavros is the first ROS package people build from source in many instances, and a lot of people need to modify it.

LorenzMeier, 2015-11-04:
@vooon We have mavros install instructions now here:
http://dev.px4.io/ros-mavros.html

I'm happy to let them point to the official mavros docs, but we need to make sure they are short, concise and tested.

vooon, 2015-11-04:
@ilia-sheremet install targets are defined, so manual copying is a bad idea.
Read `catkin config --help`, what you want was done by `--install` and `--install-space <dir>` flags.
So after `catkin build` packages will be automatically installed.

But really you don't need to install, you may already use devel space. Just `source devel/setup.bash` and then rosrun/roslaunch may find and use mavros nodes.
Anyway you will make custom controller package, so just use same workspace.

Another topic is building debian packages, but it is possible only for released (stable) versions.
And i think that source installation is easier. Let that work to ros buildfarm.

@LorenzMeier i combined mavlink and mavros installation to one step because in last few days there bunch of questions related to old mavlink.
With current script user should not face that problem again.

LorenzMeier, 2015-11-04:
The readme is still way too complex: https://github.com/mavlink/mavros

I have no idea how to install "mavros". Try to rewrite into a simple installation section and only describe the sub-packages in a separate architecture chapter.

vooon, 2015-11-04:
@LorenzMeier You mean move main instructions to root readme? For me little hard write less verbose or easier script.

LorenzMeier, 2015-11-04:
Structure the main README better. You are hiding the relevant info and highlight the irrelevant.

mhkabir, 2015-11-04:
@vooon I suggest removing the wstool system build. Simple clone and build is best for most people. 
It is faster, less complex and easier to maintain even if one is a novice. 

If you agree, I'll send in a PR with tested instructions to put in the main Readme, where it'll be visible. Trust me, most users don't like to click a link to get to the instructions.

vooon, 2015-11-04:
@LorenzMeier ok, i try.
@mhkabir but with wstool there no worry about different branches needed for Indigo and Jade (and K-Turtle). Also used barches is updated with rosdistro updates (rosinstall_generate uses distribution.yaml).

closed 2015-12-07 by vooon