mavlink/mavros #423: MAVROS sends DO_SET_MODE with armed flag not based on current arming state

LorenzMeier, 2015-11-07:
As described here this can lead to disarming the system in-air:
https://github.com/PX4/Firmware/issues/3142#issuecomment-154612992

vooon, 2015-11-07:
But i thikned that with custom mode flag system only accept that value and not parse base_mode.

vooon, 2015-11-07:
Also obvious listen to SYSTEM_STATUS and save arm flag give race that may happen between COMMAND_LONG-ARMING and receiving next SYSTEM_STATUS (<=1 sec window).

vooon, 2015-11-07:
So i thinking that better react only to arming command and not base_mode...

vooon, 2015-11-08:
Fixed, but testing required.

vooon, 2015-11-09:
Checked with SITL.

closed 2015-11-09 by vooon

mhkabir, 2015-11-09:
What about the race conditions? This could cause weird issues or corner cases in the future.

LorenzMeier, 2015-11-09:
The protocol is getting support to address transaction support. In the meantime you could just disallow mode switches if not armed.