mavlink/mavros #429: HELP! I am unable to do a catkin_make for the mavros and mavlink packages on my Odroid XU4

iamveloci, 2015-11-16:
Hi,
I am trying to get mavros and mavlink built on my Odroid XU4 and had an error during the catkin_make process. This version of mavlink and mavros did not give me any trouble while building on my Beagle Bone Black. I am trying to switch to the Odroid XU4 since it is a faster computing platform in comparison to the BBB. I have attached the screen shot for your reference and the error is visible on the left terminal. Thanks in advance!
[image]

saihv, 2015-11-16:
Did you try "sudo apt-get install ros-indigo-tf"?

iamveloci, 2015-11-16:
It worked! Thanks a ton!

closed 2015-11-16 by iamveloci

vooon, 2015-11-16:
Just follow installetion instructions: https://github.com/mavlink/mavros/tree/master/mavros#source-installation